riscv-collab/riscv-openocd #914: Custom registers only work with a single range.

timsifive, 2023-09-13:
I haven't observed this bug. It came out of a code review of #913.

Probably https://github.com/riscv/riscv-openocd/blob/67c2835997c206ddeba58cfc0fad76789db7474b/src/target/riscv/riscv.c#L6151 should iterate through the list to find the one that actually contains the current register, instead of simply taking the first range from the expose_custom list.

JanMatCodasip, 2023-09-15:
Just a small correction: The original review that mentions this is https://github.com/riscv/riscv-openocd/pull/896

Link to the respective thread: https://github.com/riscv/riscv-openocd/pull/896#discussion_r1322655110

en-sc, 2023-10-04:
This is not a bug. The code does iterate over the list via the call to [list_rotate_left()](https://github.com/riscv/riscv-openocd/blob/67c2835997c206ddeba58cfc0fad76789db7474b/src/target/riscv/riscv.c#L6175) after the range at the list head is processed.

timsifive, 2023-10-11:
@en-sc is right. The code is correct.

closed 2023-10-11 by timsifive

JanMatCodasip, 2023-10-12:
Thanks for looking into it.