riscv-collab/riscv-openocd #919: CRC algorithm fails on verify_image

jimaandro, 2023-09-19:
Hi
I have a CVA6 design implemented to FPGA, I have connected the OpenOCD and after writing the image to RAM, I tried to verify it.
But the verification fails and I get this message
`
> verify_image uImage 0x80200000
Algorithm timed out after 180001 ms.
ra = 0x0
sp = 0x0
gp = 0x0
tp = 0x0
t0 = 0x0
t1 = 0x0
t2 = 0x0
s0 = 0x0
s1 = 0x0
a0 = 0x80200000
a1 = 0x82cb3e
a2 = 0x0
a3 = 0x0
a4 = 0x0
a5 = 0x0
a6 = 0x0
a7 = 0x0
s2 = 0x0
s3 = 0x0
s4 = 0x0
s5 = 0x0
s6 = 0x0
s7 = 0x0
s8 = 0x0
s9 = 0x0
s10 = 0x0
s11 = 0x0
t3 = 0x0
t4 = 0x0
t5 = 0x0
t6 = 0x0
pc = 0x80200040
mstatus = 0xa00001800
mepc = 0x80200040
mcause = 0xc
error executing RISC-V CRC algorithm
verified 8571710 bytes in 283.027466s (29.576 KiB/s)

The PC stucks at 0x80200040 address, the entry point and load address are 0x80200000
Why does this happen? Also I dump the 0x80200000 area and use objdump to see if there is some valid code there and I got this
[uImage.zip](https://github.com/riscv/riscv-openocd/files/12662824/uImage.zip)

TommyMurphyTM1234, 2023-09-19:
It'll probably help if you capture, compress, and attach an `openocd -d3` verbose log.

jimaandro, 2023-09-19:
Thank you for the reply @TommyMurphyTM1234 . [Here](https://drive.google.com/file/d/1V5sZ5K5QzTwrTvoDpfH3cnObFcsb7hzI/view?usp=sharing) is the compressed debug log file. (the compresses file was too big to upload here)

I don't understand why it gets a page fault, I have an ILA chipscope connected to the memory interface of the CVA6 and it shows that the core is reading data from that address.

jimaandro, 2023-09-19:
Well, this is what telnet prints :

> mdw 0x80200000 16
0x80200000: 56190527 2c3baaeb 13900965 feca8200 00002080 00002080 2c789e67 01021a05 
0x80200020: 34365643 694c3641 0078756e 00000000 00000000 00000000 00000000 00000000 
> mdw 0x80200040   
0x80200040: 08088b1f 

The **08088b1f** is the same in uImage dump:
      2a:	0078                	addi	a4,sp,12
	...
      40:	**8b1f 0808** 5738      	0x5738**08088b1f**
      46:	03026507          	0x3026507

The SDRAM address space is `0x80000000-0xFFFFFFFF`, so I don't know if the `0x80200040` could have any meaning in the core interface logic.

TommyMurphyTM1234, 2023-09-19:
> Well, this is what telnet prints :
> 
> > mdw 0x80200000 16
> > 0x80200000: 56190527 2c3baaeb 13900965 feca8200 00002080 00002080 2c789e67 01021a05
> > 0x80200020: 34365643 694c3641 0078756e 00000000 00000000 00000000 00000000 00000000
> > mdw 0x80200040
> > 0x80200040: 08088b1f

OK - I was just curious if such a read of that address might trigger the same problem (page fault).
Obviously not.

I presume that the original operations were done via GDB?
If so can you try to do the same (connect, load image, verify image) via the OpenOCD telnet interface just to see if that makes any difference and just to remove one item (GDB) from the setup?
You can find documentation on the relevant commands in the OpenOCD documentation:

* https://openocd.org/pages/documentation.html

TommyMurphyTM1234, 2023-09-19:
The verbose log is huge and difficult to open and analyse.
Usually they're big but not 2GB big!
But I see many instances of `riscv_run_algorithm()` failures before the final one which fails due to the page fault.
```
Search "Error:" (37 hits in 1 file of 1 searched)
  C:\Users\Tommy Murphy\Downloads\open_deb.log\open_deb.log (37 hits)
	Line 29540504: Error: 29540503 488769 riscv.c:1915 riscv_run_algorithm(): Algorithm timed out after 180001 ms.
	Line 29540722: Error: 29540721 488770 riscv.c:1935 riscv_run_algorithm(): ra = 0x0
	Line 29540794: Error: 29540793 488771 riscv.c:1935 riscv_run_algorithm(): sp = 0x0
	Line 29540866: Error: 29540865 488771 riscv.c:1935 riscv_run_algorithm(): gp = 0x0
	Line 29540938: Error: 29540937 488772 riscv.c:1935 riscv_run_algorithm(): tp = 0x0
	Line 29541010: Error: 29541009 488772 riscv.c:1935 riscv_run_algorithm(): t0 = 0x0
	Line 29541082: Error: 29541081 488773 riscv.c:1935 riscv_run_algorithm(): t1 = 0x0
	Line 29541154: Error: 29541153 488773 riscv.c:1935 riscv_run_algorithm(): t2 = 0x0
	Line 29541226: Error: 29541225 488774 riscv.c:1935 riscv_run_algorithm(): s0 = 0x0
	Line 29541298: Error: 29541297 488774 riscv.c:1935 riscv_run_algorithm(): s1 = 0x0
	Line 29541370: Error: 29541369 488775 riscv.c:1935 riscv_run_algorithm(): a0 = 0x80200000
	Line 29541442: Error: 29541441 488775 riscv.c:1935 riscv_run_algorithm(): a1 = 0x82cb3e
	Line 29541514: Error: 29541513 488776 riscv.c:1935 riscv_run_algorithm(): a2 = 0x0
	Line 29541586: Error: 29541585 488776 riscv.c:1935 riscv_run_algorithm(): a3 = 0x0
	Line 29541658: Error: 29541657 488777 riscv.c:1935 riscv_run_algorithm(): a4 = 0x0
	Line 29541730: Error: 29541729 488777 riscv.c:1935 riscv_run_algorithm(): a5 = 0x0
	Line 29541802: Error: 29541801 488778 riscv.c:1935 riscv_run_algorithm(): a6 = 0x0
	Line 29541874: Error: 29541873 488778 riscv.c:1935 riscv_run_algorithm(): a7 = 0x0
	Line 29541946: Error: 29541945 488779 riscv.c:1935 riscv_run_algorithm(): s2 = 0x0
	Line 29542018: Error: 29542017 488779 riscv.c:1935 riscv_run_algorithm(): s3 = 0x0
	Line 29542090: Error: 29542089 488780 riscv.c:1935 riscv_run_algorithm(): s4 = 0x0
	Line 29542162: Error: 29542161 488780 riscv.c:1935 riscv_run_algorithm(): s5 = 0x0
	Line 29542234: Error: 29542233 488781 riscv.c:1935 riscv_run_algorithm(): s6 = 0x0
	Line 29542306: Error: 29542305 488781 riscv.c:1935 riscv_run_algorithm(): s7 = 0x0
	Line 29542378: Error: 29542377 488782 riscv.c:1935 riscv_run_algorithm(): s8 = 0x0
	Line 29542450: Error: 29542449 488782 riscv.c:1935 riscv_run_algorithm(): s9 = 0x0
	Line 29542522: Error: 29542521 488783 riscv.c:1935 riscv_run_algorithm(): s10 = 0x0
	Line 29542594: Error: 29542593 488783 riscv.c:1935 riscv_run_algorithm(): s11 = 0x0
	Line 29542666: Error: 29542665 488784 riscv.c:1935 riscv_run_algorithm(): t3 = 0x0
	Line 29542738: Error: 29542737 488784 riscv.c:1935 riscv_run_algorithm(): t4 = 0x0
	Line 29542810: Error: 29542809 488785 riscv.c:1935 riscv_run_algorithm(): t5 = 0x0
	Line 29542882: Error: 29542881 488785 riscv.c:1935 riscv_run_algorithm(): t6 = 0x0
	Line 29542955: Error: 29542954 488786 riscv.c:1935 riscv_run_algorithm(): pc = 0x80200040
	Line 29543027: Error: 29543026 488786 riscv.c:1935 riscv_run_algorithm(): mstatus = 0xa00001800
	Line 29543099: Error: 29543098 488787 riscv.c:1935 riscv_run_algorithm(): mepc = 0x80200040
	Line 29543171: Error: 29543170 488787 riscv.c:1935 riscv_run_algorithm(): mcause = 0xc
	Line 29543172: Error: 29543171 488787 riscv.c:2065 riscv_checksum_memory(): error executing RISC-V CRC algorithm
```

TommyMurphyTM1234, 2023-09-19:
What OpenOCD script(s?) are you using to connect to your target?
What work area does it define?

* https://openocd.org/doc/html/CPU-Configuration.html#:~:text=You%20should%20specify%20a%20working%20area%20if%20you%20can%3B%20typically%20it%20uses%20some%20on%2Dchip%20SRAM.%20Such%20a%20working%20area%20can%20speed%20up%20many%20things%2C%20including%20bulk%20writes%20to%20target%20memory%3B%20flash%20operations%20like%20checking%20to%20see%20if%20memory%20needs%20to%20be%20erased%3B%20GDB%20memory%20checksumming%3B%20and%20more.

jimaandro, 2023-09-19:
> OK - I was just curious if such a read of that address might trigger the same problem (page fault). Obviously not.
>
> I presume that the original operations were done via GDB? If so can you try to do [...]

No I was using telnet from the beginning, but I will try GDB too.

> What OpenOCD script(s?) are you using to connect to your target? What work area does it define?
> 
> * https://openocd.org/doc/html/CPU-Configuration.html#:~:text=You%20should%20specify%20a%20working%20area%20if%20you%20can%3B%20typically%20it%20uses%20some%20on%2Dchip%20SRAM.%20Such%20a%20working%20area%20can%20speed%20up%20many%20things%2C%20including%20bulk%20writes%20to%20target%20memory%3B%20flash%20operations%20like%20checking%20to%20see%20if%20memory%20needs%20to%20be%20erased%3B%20GDB%20memory%20checksumming%3B%20and%20more.
> 
> Edit: sorry - I see the work area definition here:
> 
> ```
> Debug: 27 3 command.c:152 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu -coreid 0 -work-area-phys 0x8000000000 -work-area-size 1073741824
> ```
> 
> Are these details correct and is there RAM at that address range (`0x80_0000_0000-0x83_ffff_ffff`?!) that can be used as a work area by stuff such as image verification?
> 
> At a glance I cannot find the OpenOCD scripts mentioned in the CVA6 readme here:
> 
> * https://github.com/openhwgroup/cva6#debugging

Oooh, yes you are right I missed that point. The RAM starts from `0x8000_0000` not `0x80_0000_0000`! But, after fixing this I tried again I got the same error.

This is the cfg script I am using, it is similar to cva6's [script](https://github.com/openhwgroup/cva6/blob/master/corev_apu/fpga/ariane.cfg), But instead of FTDI I am using PCIe to connect to the board.
`
```
adapter speed     1000

adapter driver xlnx_pcie_xvc
xlnx_pcie_xvc config 0000:41:00.0

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0 -work-area-phys 0x80000000 -work-area-size 1073741824

gdb_report_data_abort enable
gdb_report_register_access_error enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

if { [catch {riscv set_enable_virtual on} ] } {
    echo "Warning: This version of OpenOCD does not support address translation. To debug on virtual addresses, please update to the latest version." }

init
halt
echo "Ready for Remote Connections"
```
`

jimaandro, 2023-09-19:
> The verbose log is huge and difficult to open and analyse. Usually they're big but not 2GB big! But I see many instances of `riscv_run_algorithm()` failures before the final one which fails due to [...]

I could open it only with Visual Studio Code

TommyMurphyTM1234, 2023-09-19:
> Oooh, yes you are right I missed that point. The RAM starts from 0x8000_0000 not 0x80_0000_0000! But, after fixing this I tried again I got the same error.

I presume that this makes no difference?
```
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0 -work-area-phys 0x80000000 -work-area-size 0x1000 -work-area-backup 1
```

jimaandro, 2023-09-19:
> I presume that this makes no difference?
> 
> ```
> target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0 -work-area-phys 0x80000000 -work-area-size 0x1000 -work-area-backup 1
> ```

No, same error

TommyMurphyTM1234, 2023-09-20:
It seems as if something goes wrong while running the checksum code from the work area such that repeated polling never detects that it has completed and the operation eventually times out.
* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2671
* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2684
* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2737
* https://github.com/riscv/riscv-openocd/blob/riscv/contrib/loaders/checksum/riscv64_crc.inc

Maybe it's possible to compile the checksum code as a standalone program just to check that it works like that?

* https://github.com/riscv/riscv-openocd/blob/riscv/contrib/loaders/checksum/riscv_crc.c
* https://github.com/riscv/riscv-openocd/blob/riscv/contrib/loaders/checksum/Makefile#L41

TommyMurphyTM1234, 2023-09-20:
I'm just guessing here but I wonder if the verify image code may be running OK but there's a problem on CVA6 when it tries to execute the `ebreak` instruction

* https://github.com/riscv/riscv-openocd/blob/riscv/contrib/loaders/checksum/riscv_crc.c#L134

that should terminate the helper code, halt the target and yield to the poll in `riscv_run_algorithm()` thus allowing the debugger to continue normally?

* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2503
* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2606

But instead it gets "stuck" and ends up timing out?

* https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L2583

jimaandro, 2023-09-20:
Well I made some changes on my design. I added memory address space at `0x80_0000_0000`. I used an address transformer to do this. It convetrs 64 bit addresses to 32bit, it actually cuts the 32 first bits. For example the `0x80_0000_0000` will access the physical memory at` 0x0000_0000`. 
So I used this address to write the image and I managed to verify it without any errors. Also I changed the workarea base address to` 0x80_0000_0000`. The problem is that the image generation scripts don't accept 64bit addresses for uImage entry point and load address. So I have to use 32 bit addresses like `0x80000000` (which fails on verification).

TommyMurphyTM1234, 2023-09-20:
> Well I made some changes on my design. I added memory address space at `0x80_0000_0000`. I used an address transformer to do this. It converts 64 bit addresses to 32bit, it actually cuts the 32 first bits. For example the `0x80_0000_0000` will access the physical memory at` 0x0000_0000`. So I used this address to write the image and I managed to verify it without any errors. Also I changed the workarea base address to` 0x80_0000_0000`. The problem is that the image generation scripts don't accept 64bit addresses for uImage entry point and load address. So I have to use 32 bit addresses like `0x80000000` (which fails on verification).

I don't really understand most of that or the rationale for such changes. Previously said that you had RAM at 0x8000_0000:

> Oooh, yes you are right I missed that point. The RAM starts from 0x8000_0000 not 0x80_0000_0000! But, after fixing this I tried again I got the same error.

TommyMurphyTM1234, 2023-09-20:
Ok, I don't really understand the memory configuration of your system and if/how it might be contributing to the ostensible failure of the checksum helper code running in the target from the work area. Sorry.

jimaandro, 2023-09-20:
> Ok, I don't really understand the memory configuration of your system and if/how it might be contributing to the ostensible failure of the checksum helper code running in the target from the work area. Sorry. But, as I said previously, it really sounds like this failure is down to the target characteristics/configuration and not OpenOCD's RISC-V support per se. In general it's easier to diagnose the precise cause of an anomaly if lots of things are not being changed at the same time.

Okay I understand, I think my configurations to the system maybe are wrong.
Thank you very much.

jimaandro, 2023-09-22:
Eventually, in my design the kernel cannot execute anything that is not written in 0x8000_0000_0000. So I am clossing this issue

closed 2023-09-22 by jimaandro