riscv-collab/riscv-openocd #923: erase/write flash error when flash-addr-width > 32bit on windows

Alan-19950616, 2023-09-26:
The error log is as follows, of course I added some print.
```
...
Info : Valid NUSPI on device Nuclei SoC SPI Flash at address 0xfa0000000 with spictrl regbase at 0xf8b120000
Info : Nuclei SPI controller version 0x00010207
Info : Found flash device 'win w25q128fv/jv' (ID 0x001840ef)
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Error: xxxx sizeof:4
Error: xxxx packet:vFlashErase:0000000fa0000000,00010000
Error: xxxx addr:ffffffff
Error: No flash at address 0xffffffff
Error: flash_erase returned -4
```

The unsigned long type on windows is defined as 4 bytes
https://learn.microsoft.com/en-us/cpp/cpp/data-type-ranges?view=msvc-170

I made the following changes and it works fine with flash, I'm not sure if a similar problem exists within openocd.
```
diff --git a/src/server/gdb_server.c b/src/server/gdb_server.c
index 8360ee1b9..b893ede4a 100644
--- a/src/server/gdb_server.c
+++ b/src/server/gdb_server.c
@@ -3332,8 +3332,8 @@ static int gdb_v_packet(struct connection *connection,
        }

        if (strncmp(packet, "vFlashErase:", 12) == 0) {
-               unsigned long addr;
-               unsigned long length;
+               unsigned long long addr;
+               unsigned long long length;

                char const *parse = packet + 12;
                if (*parse == '\0') {
@@ -3341,14 +3341,14 @@ static int gdb_v_packet(struct connection *connection,
                        return ERROR_SERVER_REMOTE_CLOSED;
                }

-               addr = strtoul(parse, (char **)&parse, 16);
+               addr = strtoull(parse, (char **)&parse, 16);

                if (*(parse++) != ',' || *parse == '\0') {
                        LOG_ERROR("incomplete vFlashErase packet received, dropping connection");
                        return ERROR_SERVER_REMOTE_CLOSED;
                }

-               length = strtoul(parse, (char **)&parse, 16);
+               length = strtoull(parse, (char **)&parse, 16);

                if (*parse != '\0') {
                        LOG_ERROR("incomplete vFlashErase packet received, dropping connection");
@@ -3389,15 +3389,16 @@ static int gdb_v_packet(struct connection *connection,

        if (strncmp(packet, "vFlashWrite:", 12) == 0) {
                int retval;
-               unsigned long addr;
-               unsigned long length;
+               unsigned long long addr;
+               unsigned long long length;
                char const *parse = packet + 12;

                if (*parse == '\0') {
                        LOG_ERROR("incomplete vFlashErase packet received, dropping connection");
                        return ERROR_SERVER_REMOTE_CLOSED;
                }
-               addr = strtoul(parse, (char **)&parse, 16);
+
+               addr = strtoull(parse, (char **)&parse, 16);
                if (*(parse++) != ':') {
                        LOG_ERROR("incomplete vFlashErase packet received, dropping connection");
                        return ERROR_SERVER_REMOTE_CLOSED;
```

timsifive, 2023-09-29:
This looks like it's not RISC-V related. Can you submit it upstream to be fixed there? (See https://openocd.org/doc-release/doxygen/HACKING_source.html for instructions.) The types should probably be uint64_t instead of `unsigned long long` which is more in line with current OpenOCD code. (I bet the code you're looking at is at least a decade old.)

en-sc, 2024-01-22:
@Alan-19950616, If you have filed an issue/PR to mainline OpenOCD, can you please link it here?

JanMatCodasip, 2024-01-24:
This patch looks to be it: https://review.openocd.org/c/openocd/+/8095

There have been few review suggestions posted but not yet addressed.

@Alan-19950616, would you like to fix the upstream review suggestions, so that your patch can be merged to the upstream? Thank you!

I am closing this thread since the work should continue on the upstream patch, and there is no more to do in riscv-openocd.

closed 2024-01-24 by JanMatCodasip