riscv-collab/riscv-openocd #932: Test suite failure - what am I doing wrong?

TommyMurphyTM1234, 2023-10-03:
Am I doing something obviously wrong here?

When I try to run the[ basic tests against Spike](https://github.com/riscv-software-src/riscv-tests/tree/master/debug#usage) it ends with this:
```
...
csr2288        Could not fetch register "csr2288"; remote failure reply 'E0E'
scountovf      Could not fetch register "scountovf"; remote failure reply 'E0E'
custom12345    0x3039   12345
custom12346    Could not fetch register "custom12346"; remote failure reply 'E0E'
custom12347    0x303b   12347
custom12348    Could not fetch register "custom12348"; remote failure reply 'E0E'
custom1        0x1      1
(gdb) maintenance flush register-cache
maintenance flush register-cache
Register cache flushed.
(gdb) info threads
info threads
  Id   Target Id         Frame
* 1    Remote target     loop_forever () at programs/entry.S:205
(gdb)
-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20231003-224433-spike64-UnavailableCycleTest.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py UnavailableCycleTest
Time elapsed: 64.36s
::::::::::::::::::::::::::::[ ran 1 tests in 64s ]::::::::::::::::::::::::::::
1 tests returned exception
   UnavailableCycleTest > logs/20231003-224433-spike64-UnavailableCycleTest.log
make: *** [Makefile:19: run.spike64.UnavailableCycleTest] Error 1
```
This is the overall summary log file (zipped): [tests.zip](https://github.com/riscv/riscv-openocd/files/12797192/tests.zip)
This is the specific test log file (zipped): [20231003-224433-spike64-UnavailableCycleTest.zip](https://github.com/riscv/riscv-openocd/files/12797212/20231003-224433-spike64-UnavailableCycleTest.zip)

TommyMurphyTM1234, 2023-10-04:
I tried the same thing on a different platform (Zorin OS Lite 16.3 on a VirtualBox VM) but it failed on a different test:
```
(gdb) maintenance flush register-cache
maintenance flush register-cache
Register cache flushed.
(gdb) info threads
info threads
  Id   Target Id         Frame 
* 1    Remote target     main () at programs/debug.c:62
(gdb) 
-------------------------------[ End of logs ]--------------------------------
Result: fail
Logfile: logs/20231004-182412-spike64-MemorySampleSingle.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py MemorySampleSingle
Time elapsed: 11.37s
::::::::::::::::::::::::::::[ ran 1 tests in 11s ]::::::::::::::::::::::::::::
1 tests returned fail
   MemorySampleSingle > logs/20231004-182412-spike64-MemorySampleSingle.log
make: *** [Makefile:19: run.spike64.MemorySampleSingle] Error 1
```

TommyMurphyTM1234, 2023-10-04:
A subsequent run (on Zorin OS) fails on a different test.
```
(gdb) info threads
info threads
  Id   Target Id         Frame 
* 1    Remote target     loop_forever () at programs/entry.S:205
(gdb) 
-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20231004-200902-spike64-UnavailableCycleTest.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py UnavailableCycleTest
Time elapsed: 68.72s
::::::::::::::::::::::::::::[ ran 1 tests in 69s ]::::::::::::::::::::::::::::
1 tests returned exception
   UnavailableCycleTest > logs/20231004-200902-spike64-UnavailableCycleTest.log
make: *** [Makefile:19: run.spike64.UnavailableCycleTest] Error 1
```
Is the Spike test scenario simply unreliable or something?

timsifive, 2023-10-04:
Are you using latest OpenOCD, spike, and tests? That test is relatively new, and certainly depends on recent OpenOCD and recent spike.

TommyMurphyTM1234, 2023-10-04:
> Are you using latest OpenOCD, spike, and tests? That test is relatively new, and certainly depends on recent OpenOCD and recent spike.

Hi @timsifive - yes - I've cloned everything from the respective master/heads and built afresh before running the tests on both platforms (WSL2/Ubuntu and VirtualBox VM/Zorin OS Lite 16.3):
```
# Toolchain and Spike

cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=$HOME/Downloads/riscv-tools --enable-multilib --with-sim=spike
make
make build-sim

# OpenOCD

# I've installed libusb-1.0-0-dev a priori so that FTDI support gets enabled by configure
cd ~/Downloads
git clone https://github.com/riscv/riscv-openocd
cd riscv-openocd
./bootstrap
./configure --prefix=$HOME/Downloads/riscv-tools
make install

# Test suite

cd ~/Downloads
git clone --recursive https://github.com/riscv-software-src/riscv-tests
cd riscv-tests

export PATH=$HOME/Downloads/riscv-tools/bin
make

...
(gdb) maintenance flush register-cache
maintenance flush register-cache
Register cache flushed.
(gdb) info threads
info threads
  Id   Target Id         Frame 
* 1    Remote target     loop_forever () at programs/entry.S:205
(gdb) 
-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20231004-202627-spike64-UnavailableCycleTest.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py UnavailableCycleTest
Time elapsed: 66.79s
::::::::::::::::::::::::::::[ ran 1 tests in 67s ]::::::::::::::::::::::::::::
1 tests returned exception
   UnavailableCycleTest > logs/20231004-202627-spike64-UnavailableCycleTest.log
make: *** [Makefile:19: run.spike64.UnavailableCycleTest] Error 1
```

TommyMurphyTM1234, 2023-10-04:
I'm assuming that the Spike referenced by `riscv-gnu-toolchain` is appropriate?
Maybe that's not the case and that's my problem?

* https://github.com/riscv-collab/riscv-gnu-toolchain
  * https://github.com/riscv-software-src/riscv-isa-sim/tree/71f5a8fd1b1185b7c15151f116aa3fef058bf210

timsifive, 2023-10-04:
That's not recent enough. You need at least https://github.com/riscv-software-src/riscv-isa-sim/pull/1394 in your spike.

TommyMurphyTM1234, 2023-10-04:
Ah - OK. 
I'll update my Spike and try again.
Could/should `riscv-gnu-toolchain`'s Spike submodule be updated to a more recent commit?

TommyMurphyTM1234, 2023-10-05:
Hi @timsifive - I rebuilt Spike from the latest sources (latest commit on `master` branch) but still get the failure:
```
# Toolchain and Spike

cd ~/Downloads
git clone --recursive https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain/spike
git pull https://github.com/riscv-software-src/riscv-isa-sim master
git log
...
commit e9848ed3056eba91a5f0d15539358e5a03c66011
Merge: 09a3c644 ea674035
Author: Andrew Waterman <andrew@sifive.com>
Date:   Wed Jul 5 15:44:11 2023 -0400

    Merge pull request #1394 from riscv-software-src/unavailable
    
    Let debugger control hart availability
...

cd ..
./configure --prefix=$HOME/Downloads/riscv-tools --enable-multilib --with-sim=spike
make && make build-sim

# OpenOCD

# I've installed libusb-1.0-0-dev a priori so that FTDI support gets enabled by configure
cd ~/Downloads
git clone https://github.com/riscv/riscv-openocd
cd riscv-openocd
./bootstrap
./configure --prefix=$HOME/Downloads/riscv-tools
make install

# Test suite

cd ~/Downloads
git clone --recursive https://github.com/riscv-software-src/riscv-tests
cd riscv-tests

export PATH=$HOME/Downloads/riscv-tools/bin
make

...

csr2288        Could not fetch register "csr2288"; remote failure reply 'E0E'
scountovf      Could not fetch register "scountovf"; remote failure reply 'E0E'
custom12345    0x3039	12345
custom12346    Could not fetch register "custom12346"; remote failure reply 'E0E'
custom12347    0x303b	12347
custom12348    Could not fetch register "custom12348"; remote failure reply 'E0E'
custom1        0x1	1
(gdb) maintenance flush register-cache
maintenance flush register-cache
Register cache flushed.
(gdb) info threads
info threads
  Id   Target Id         Frame 
* 1    Remote target     loop_forever () at programs/entry.S:205
(gdb) 
-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20231005-100156-spike64-UnavailableCycleTest.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py UnavailableCycleTest
Time elapsed: 62.90s
::::::::::::::::::::::::::::[ ran 1 tests in 63s ]::::::::::::::::::::::::::::
1 tests returned exception
   UnavailableCycleTest > logs/20231005-100156-spike64-UnavailableCycleTest.log
make: *** [Makefile:19: run.spike64.UnavailableCycleTest] Error 1
```
This is the log: [20231005-100156-spike64-UnavailableCycleTest.zip](https://github.com/riscv/riscv-openocd/files/12816535/20231005-100156-spike64-UnavailableCycleTest.zip)

TommyMurphyTM1234, 2023-10-05:
In case I made any mistake (even though I don't think so), I rebuilt everything from scratch and tried again.

First run - MemTest8 failed:
```
./gdbserver.py \
	./targets/RISC-V/spike64.py \
	MemTest8 \
	--isolate \
	--print-failures \
	
Using $misa from hart definition: 0x8000000000141125
[MemTest8] Starting > logs/20231005-140826-spike64-MemTest8.log
[MemTest8] exception in 67.27s
Test: MemTest8
Target: spike64
---------------------------------[ Compile ]----------------------------------
+ riscv64-unknown-elf-gcc -g programs/checksum.c programs/tiny-malloc.c programs/infinite_loop.S -DDEFINE_MALLOC -DDEFINE_FREE programs/entry.S programs/init.c -DNHARTS=1 -I ../env -T targets/RISC-V/spike64.lds -nostartfiles -mcmodel=medany -DXLEN=64 -o /tmp/spike64_checksum-8000000000141125_ypfzahf5 -march=rv64imafc_zicsr -mabi=lp64
--------------------------------[ Traceback ]---------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 109, in expect_loop
    return self.timeout()
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect.pty_spawn.spawn object at 0x7f3d8d984dc0>
command: /home/user/Downloads/riscv-tools/bin/riscv64-unknown-elf-gdb
args: ['/home/user/Downloads/riscv-tools/bin/riscv64-unknown-elf-gdb']
buffer (last 100 chars): b'tation/>.\r\n\r\nFor help, type "help".\r\nType "apropos word" to search for commands related to "word".\r\n'
before (last 100 chars): b'tation/>.\r\n\r\nFor help, type "help".\r\nType "apropos word" to search for commands related to "word".\r\n'
after: <class 'pexpect.exceptions.TIMEOUT'>
match: None
match_index: None
exitstatus: None
flag_eof: False
pid: 520169
child_fd: 15
closed: False
timeout: 30
delimiter: <class 'pexpect.exceptions.EOF'>
logfile: <tempfile._TemporaryFileWrapper object at 0x7f3d8d984fa0>
logfile_read: None
logfile_send: None
maxread: 2000
ignorecase: False
searchwindowsize: None
delaybeforesend: 0.05
delayafterclose: 0.1
delayafterterminate: 0.1
searcher: searcher_re:
    0: re.compile(b'\\(gdb\\)')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 1331, in run
    self.classSetup()
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 1388, in classSetup
    self.gdb = Gdb(self.target, self.server.gdb_ports, cmd=gdb_cmd,
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 729, in __init__
    self.wait()
  File "/home/user/Downloads/riscv-tests/debug/testlib.py", line 807, in wait
    self.active_child.expect(r"\(gdb\)")
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/lib/python3/dist-packages/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 119, in expect_loop
    return self.timeout(e)
  File "/usr/lib/python3/dist-packages/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect.pty_spawn.spawn object at 0x7f3d8d984dc0>
[... 306 lines cut ...]
custom12346    Could not fetch register "custom12346"; remote failure reply 'E0E'
custom12347    0x303b	12347
custom12348    Could not fetch register "custom12348"; remote failure reply 'E0E'
custom1        0x1	1
(gdb) maintenance flush register-cache
maintenance flush register-cache
Register cache flushed.
(gdb) info threads
info threads
  Id   Target Id         Frame 
* 1    Remote target     0x0000001212340342 in main () at programs/debug.c:63
(gdb) 
-------------------------------[ End of logs ]--------------------------------
Result: fail
Logfile: logs/20231005-144604-spike64-MemorySampleSingle.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64.py MemorySampleSingle
Time elapsed: 51.57s
::::::::::::::::::::::::::::[ ran 1 tests in 52s ]::::::::::::::::::::::::::::
1 tests returned fail
   MemorySampleSingle > logs/20231005-144604-spike64-MemorySampleSingle.log
make: *** [Makefile:19: run.spike64.MemorySampleSingle] Error 1
```
So - so far I have been unable to successfully run a full set of tests against Spike.
Any ideas if I am doing something wrong?
Thanks

TommyMurphyTM1234, 2023-10-05:
OK - reverting to the WSL2/Ubuntu 20.04 host that I originally used, and having built everything (including the latest Spike) from scratch, all tests run OK. I can only assume that the previous failures are something to do with the VirtualBox VM/Zorin OS Lite 16.3 host and not the test suite itself - so I am closing the issue. Thanks for the input @timsifive.

closed 2023-10-05 by TommyMurphyTM1234