riscv-collab/riscv-openocd #933: execute_fence does not handle errors

aap-sc, 2023-10-10:
Currently execute_fence (from riscv-013.c) is structured as follows:

```
if (has_sufficient_progbuf(target, 3)) {                                                        
   riscv_program_init(&program, target);                                                         
   riscv_program_fence_rw_rw(&program);                                                          
   riscv_program_fence_i(&program);                                                              
   result = riscv_program_exec(&program, target);                                                
   if (result != ERROR_OK)                                                                       
       LOG_TARGET_DEBUG(target, "Unable to execute pre-fence");                                    
  return ERROR_OK;
} 

if (has_sufficient_progbuf(target, 2)) {
 ...
  return ERROR_OK;
}
return ERROR_FAIL
```

As you can see - this function always return `ERROR_OK` if there is enough space in progbuf. I've checked the codebase and it seems that it was this way for a while now (even before  https://github.com/riscv/riscv-openocd/pull/903 fixup).

Now, I don't quite understand why is the error ignored? @timsifive do you have any ideas/would you kindly clarify. Is it for compatibility with targets that do not have `fence` - like instructions or is it a bug?

NOTE: the reason I'm asking the question is that recently I've encountered a HW that under certain conditions returned an error during execution of `fence` - I was thinking about adding a more strict diagnostics for this case.

timsifive, 2023-10-10:
`fence.i` is not required to be implemented, and if it isn't, then there really isn't anything we can do about it. Certainly we don't want to report that to the user every time we attempt to executed it. That's just annoying. We could report the first failure and be silent from then on.

TommyMurphyTM1234, 2023-10-10:
This issue might also be relevant here?

* https://github.com/riscv/riscv-openocd/issues/415

aap-sc, 2023-10-11:
@TommyMurphyTM1234 yes, #415 has a direct relevance to this. Thank you! 

@timsifive , regarding this:

> fence.i is not required to be implemented, and if it isn't, then there really isn't anything we can do about it. 

Ok, got it.

> We could report the first failure and be silent from then on.

Well... I think that `LOG_TARGET_DEBUG` is not that annoying so it can be left as is IMHO. However, in my original message I state that:

>  I was thinking about adding a more strict diagnostics for this case.

By this I mean I'd like to modify `execute_fence` function to return a proper error, so callers (like `write_memory`) could identify potential errors. The fact that we don't have a simple way to figure out  if `i.fence` is supported is disappointing. I have few ideas how this can be done, however and would like to discuss these are reasonable:

- Option "a".

Modify `riscv_program_exec` to return (as an output parameter) an additional status indicating the reason for the failure, so we can differentiate between an exception during execution of program buffer and some other errors. Then, `execute_fence` could query this status and return `ERROR_FAIL` if `riscv_program_exec` fails for some other reason (other than encountering an exception).

- Option "b".

Introduce an additional target configuration parameter, something like `riscv_require_ifence` (which  should be true by default) and just query these options before deciding if we even need to issue the fence.

- Option "c" (I don't like it).

We could try to issue the fence during `examine` . And analyze the status of the operation (it requires Option "a" to be implemented). However, I don't like this approach renders our `examine` procedure even more invasive - and In think we should minimize the impact of the `examine` on the system.

My preference would be to implement "Option b" since the implementation is rather straightforward without any juggling with error codes. Targets that don't support fence operation will have to byte the bullet and modify their configuration files. We can add a big ugly/pretty error message to help with that.  

If anyone has an opinion on the matter, please do share.

MarekVCodasip, 2023-10-12:
In our opinion (consulted with @JanMatCodasip): 

- We don't like the option "c" either due to the invasivity.
- For option "a": We believe having different return codes for riscv_program_exec() is a good general improvement (ERROR_PROGBUF_TOO_SMALL, ERROR_PROGBUF_EXCEPTION, etc.). We would not prefer to add additional output parameter for this. Return code seems sufficient and is simpler to use.
- Option "b" seems OK to us.
- We have one more option "d" for consideration. It is described below.

Option "d":
Keep track if fence_i is supported on the target (three state value YES/NO/MAYBE). Upon the first attempt to execute fence_i, detect if its supported and store the outcome. If it's not supported, log a warning (which will be shown only once).
This option can go together with option "a".

aap-sc, 2023-10-12:
> We would not prefer to add additional output parameter for this. Return code seems sufficient and is simpler to use.

Regarding the error code. We have a major problem with that, since most of the functions that return errors use `int` values that are defined mostly in target-independent code. This may create a problem with maintaining these: 

1.  our riscv-(progbuf) specific codes may overlap with some target-independent value
2.  the rest of the codebase will operate on target-independent `errors` and figuring out what is what may be problematic.

Thus I feel that if we want to introduce special handling for progbuf-specific errors, these errors should be a separate type (likely a structure) to avoid confusion. Returning the error code as an output parameter may be beneficial for code uniformity, however I can see that having such output parameters may affect readability (so if we end up with choosing option "d", we should consider both implementations).

en-sc, 2023-10-12:
Regarding the option "a":
There is the `cmderr`  field in `riscv013_info_t` which contains the most recent value of `abstractcs.cmderr` (set in `execute_abstract_command`). This allows one to check if an exception has occurred without introducing additional return codes. However, it should be noted, that if such an approach is used, the value of `info->cmderr` should be reset prior to call to `riscv_program_exec`.

JanMatCodasip, 2023-10-12:
> Regarding the option "a": There is the cmderr field in riscv013_info_t ...

Communicating function call data / results through per-target global variables is not a good pattern. If possible, I would like to avoid it.

JanMatCodasip, 2023-10-12:
> our riscv-(progbuf) specific codes may overlap with some target-independent value

I assume that if we claim certain unique error code, no other part of OpenOCD (no other target and also no target independent) should use it.

In another words, there should not be any overlap, provided that other developers check the uniqueness of their newly introduced error codes.

aap-sc, 2023-10-12:
> In another words, there should not be any overlap, provided that other OpenOCD developers do check the uniqueness of their newly introduced error codes.

The problem is that upstream OpenOCD changes are not synchronized. I doubt that OpenOCD developers from https://openocd.org/ will keep these in-sync. So this is a recipe for potential issue in future. Alternative solution will be to return a structure with the necessary statuses from progbuf-related functions. A little cumbersome, but no pointers are involved.

 In any case - the finding @en-sc may be useful to address this.

aap-sc, 2023-10-12:
> When submitting code from riscv-openocd to vanilla upstream, we just need to check our new error codes (and in the unlikely event of a collision re-number them).

This implies that we will remember about this. Spoiler: we won't  :)

timsifive, 2023-10-13:
My point wasn't that it's hard to detect whether `fence.i` succeeds or not. The point is that it doesn't matter, because the debugger has no alternative. Either it works, which is good for the user. Or it doesn't work because the hardware doesn't support it, and then it sucks for the user (potentially, if there are multiple harts, and caches, and no cache snooping).

As a general principle it's good to have the code be aware of when errors happen, and I would be OK with a change that makes it so. But it's going to have no effect on the user experience at all.

aap-sc, 2023-10-13:
> But it's going to have no effect on the user experience at all.

In our specific case the problem is that (due to some bug in HW) the `ifence` hangs unexpectedly, but `load_image` function still reports success. I want the debugger to report an error in such cases (thus the ticket and discussion)

timsifive, 2023-10-16:
Hmm... But how can the debugger distinguish between `fence.i` not being implemented, and it leading to an unexpected hang? I wouldn't want to report an error simply because it isn't implemented. Is there a mechanism to determine whether `fence.i` is implemented without trying to execute the instruction?

aap-sc, 2023-10-16:
> Hmm... But how can the debugger distinguish between fence.i not being implemented, and it leading to an unexpected hang? I wouldn't want to report an error simply because it isn't implemented. Is [...]

Probably, I've formulated the issue incorrectly.

To detect that a situation that the fence.i is not implemented it is enough (I think) to identify a situation when the result of a corresponding abstract command indicates and exception.

By "hang" I mean a situation when execution of a corresponding abstract command results in timeout.

TommyMurphyTM1234, 2023-10-16:
I would have sworn that somebody else (elsewhere?) suggested adding a RISC-V specific boolean attribute and command with which OpenOCD could be told that the target did not support `fence.i` and then not try to use it. Obviously this attribute would be "true" by default for the de facto behaviour. Wouldn't that be a happy medium given that it seems like there's no simple way to dynamically detect target support for `fencei.i`?

aap-sc, 2023-10-16:
> with which OpenOCD could be told that the target did not support fence.i and then not try to use

this is "option b" :).

TommyMurphyTM1234, 2023-10-16:
> > with which OpenOCD could be told that the target did not support fence.i and then not try to use
> 
> this is "option b" :).

Thanks! My bad for not scrolling back far enough! :-D

timsifive, 2023-10-16:
> To detect that a situation that the fence.i is not implemented it is enough (I think) to identify a situation when the result of a corresponding abstract command indicates and exception.

Ah. That makes sense. If the command regularly fails that probably means `fence.i` is not implemented and there is no reason to bother the user. But a timeout error definitely indicates there is a problem that the user should be notified about.

aap-sc, 2024-01-09:
https://github.com/riscv/riscv-openocd/pull/964 addressed this issue. Decided to go with option "a" for now (see https://github.com/riscv/riscv-openocd/issues/933#issuecomment-1757465562 )

closed 2024-01-09 by aap-sc