riscv-collab/riscv-openocd #960: Why can't the thread command switch the current target?

Alan-19950616, 2023-11-10:
```
(gdb) info threads 
  Id   Target Id                                                     Frame 
* 1    Thread 1 "riscv.cpu.0" (Name: riscv.cpu.0, state: breakpoint) _start () at ../../../SoC/demosoc/Common/Source/GCC/startup_demosoc.S:154
  2    Thread 2 "riscv.cpu.1" (Name: riscv.cpu.1, state: breakpoint) _start () at ../../../SoC/demosoc/Common/Source/GCC/startup_demosoc.S:154
(gdb) monitor target current
riscv.cpu.0
(gdb) thread 2     
[Switching to thread 2 (Thread 2)]
#0  _start () at ../../../SoC/demosoc/Common/Source/GCC/startup_demosoc.S:154
154         csrc CSR_MSTATUS, MSTATUS_MIE
(gdb) monitor target current
riscv.cpu.0
(gdb)
```

timsifive, 2023-11-10:
They are completely separate. gdb has no idea what happens with any `monitor` commands. It just knows it's passing strings back and forth. Depending on what you're debugging, they are also different things. If you're debugging a FreeRTOS application on a single target, gdb can still see each FreeRTOS thread as a thread.

Alan-19950616, 2023-11-21:
So I can only use `monitor targets riscv0.cpu.1` to switch the current target, is there any other way?
```
(gdb) monitor targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* riscv.cpu.0        riscv      little riscv.cpu          halted
 1  riscv.cpu.1        riscv      little riscv.cpu          halted
(gdb) monitor targets riscv.cpu.1
(gdb) monitor targets
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0  riscv.cpu.0        riscv      little riscv.cpu          halted
 1* riscv.cpu.1        riscv      little riscv.cpu          halted
(gdb)
```

timsifive, 2023-11-21:
That is the only way I know of.

aap-sc, 2023-11-21:
> Why can't the thread command switch the current target? #960

Because it does not have to. Gdb operate on "inferrior threads", while openocd telnet/tcl server operate on openocd targets. Internally (depending on the specified -rtos) openocd maintains a mapping between gdb inferiors  that may represent a real hart (if -rtos hwthread is specified), but these can represent something else (if some other kind of -rtos is specified). 

When you switch current **target** in tcl openocd server - it means nothing for gdb and vice versa.

fanghuaqi, 2023-11-22:
Hi @timsifive & @aap-sc ，if I bind the riscv cores using `hwthread`, I can see threads and targets are binded like below:

~~~
(gdb) info threads
  Id   Target Id                                                        Frame
  1    Thread 1 "riscv.cpu.0" (Name: riscv.cpu.0, state: debug-request) 0x00000000fdfa8b72 in ?? ()
* 2    Thread 2 "riscv.cpu.1" (Name: riscv.cpu.1, state: debug-request) 0x000000008000b3be in ?? ()
  3    Thread 3 "riscv.cpu.2" (Name: riscv.cpu.2, state: debug-request) 0x000000008000b3be in ?? ()
  4    Thread 4 "riscv.cpu.3" (Name: riscv.cpu.3, state: debug-request) 0x000000008000b3be in ?? ()
  5    Thread 5 "riscv.cpu.4" (Name: riscv.cpu.4, state: debug-request) 0x000000008000b3be in ?? ()
  6    Thread 6 "riscv.cpu.5" (Name: riscv.cpu.5, state: debug-request) 0x000000008000b3be in ?? ()
  7    Thread 7 "riscv.cpu.6" (Name: riscv.cpu.6, state: debug-request) 0x000000008000b3be in ?? ()
  8    Thread 8 "riscv.cpu.7" (Name: riscv.cpu.7, state: debug-request) 0x000000008000b3be in ?? ()
(gdb) monitor targets
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* riscv.cpu.0        riscv      little riscv.cpu          halted
 1  riscv.cpu.1        riscv      little riscv.cpu          halted
 2  riscv.cpu.2        riscv      little riscv.cpu          halted
 3  riscv.cpu.3        riscv      little riscv.cpu          halted
 4  riscv.cpu.4        riscv      little riscv.cpu          halted
 5  riscv.cpu.5        riscv      little riscv.cpu          halted
 6  riscv.cpu.6        riscv      little riscv.cpu          halted
 7  riscv.cpu.7        riscv      little riscv.cpu          halted
~~~

If I switch `thread`, although the `target` is not switched, is there an API in openocd to figure out current target bind with the thread, if you know it, could you point it out?

Thanks

aap-sc, 2023-11-22:
@fanghuaqi  I'm not aware of such API too (I don't believe it exists, actually). Why do you need such API?

fanghuaqi, 2023-11-22:
Like code in https://github.com/riscv/riscv-openocd/blob/riscv/src/target/etm.c#L1191C26-L1192

I want to know the current target bind to the thread, so I can operate on the correct target using openocd without using `monitor targets riscv0.cpu.1` to switch target.

aap-sc, 2023-11-22:
Note: I think that in the current implementation the name of OpenOCD target goes directly to the name of inferior thread. So with some scripting and regexp matching you could do that. But I'm not sure if this is a future-proof method

timsifive, 2023-11-22:
It might be worth asking about this on the OpenOCD mailing list. None of this is RISC-V specific.

fanghuaqi, 2023-11-23:
Okay, thanks @timsifive @aap-sc

en-sc, 2024-01-22:
@Alan-19950616, @fanghuaqi, seems like the issue is resolved. I would like to close it.

fanghuaqi, 2024-01-24:
Okay, thanks, you can close it for now.

closed 2024-01-26 by en-sc