riscv-collab/riscv-openocd #965: 'riscv set_pfrefer_sba on' can't be executed.

hyperion009, 2023-11-20:
Hi 
I am trying to connect SimJTAG in Rockchip to OpenOCD by the following command(https://github.com/chipsalliance/rocket-chip/blob/master/scripts/RocketSim.cfg):
setenv JTAG_VPI_PORT 36054; setenv JTAG_DTM_ENABLE_SBA on; openocd -f RocketSim.cfg -d -l log
However, Got a issue:
[image]
Is there a real issue? How to fix/workaround this one?

thanks

TommyMurphyTM1234, 2023-11-20:
`riscv set_prefer_sba` was removed by this PR:

* https://github.com/riscv/riscv-openocd/pull/884#issuecomment-1655651193

Your `Rocketsim.cfg` script is using a command that is no longer supported.

hyperion009, 2023-11-20:
@TommyMurphyTM1234  thanks Tommy for your update.

closed 2023-11-20 by hyperion009