riscv-collab/riscv-openocd #966: Debug Module did not become active. dmcontrol=0x0

wizkad, 2023-11-21:
Hello, i am trying to program a zedboard using openocd to a ibex core. The output error is this one:
```
./util/load_demo_system2.sh halt ./sw/c/build/demo/hello_world/demo
Open On-Chip Debugger 0.12.0+dev-03370-g6f028846a (2023-11-10-09:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 3 options.c:52 configuration_output_handler(): debug_level: 3User : 4 3 options.c:52 configuration_output_handler(): 
Debug: 5 3 options.c:233 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 3 options.c:234 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 7 3 options.c:235 add_default_dirs(): exepath=/usr/local/bin
Debug: 8 3 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 3 configuration.c:33 add_script_search_dir(): adding /home/jorgelarman/.config/openocd
Debug: 10 3 configuration.c:33 add_script_search_dir(): adding /home/jorgelarman/.openocd
Debug: 11 3 configuration.c:33 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 12 3 configuration.c:33 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 13 3 command.c:152 script_debug(): command - ocd_find /home/jorgelarman/work/ibex-demo-system-zedboard/util/zedboard-openocd-cfg.tcl
Debug: 14 3 configuration.c:88 find_file(): found /home/jorgelarman/work/ibex-demo-system-zedboard/util/zedboard-openocd-cfg.tcl
Debug: 15 4 command.c:152 script_debug(): command - adapter driver ftdi
Debug: 16 4 command.c:152 script_debug(): command - transport select jtag
Debug: 17 4 command.c:152 script_debug(): command - ftdi channel 0
Debug: 18 4 command.c:152 script_debug(): command - ftdi device_desc Digilent USB Device
Debug: 19 4 command.c:152 script_debug(): command - ftdi vid_pid 0x0403 0x6014
Debug: 20 4 command.c:152 script_debug(): command - ftdi layout_init 0x2088 0x3f8b
Debug: 21 4 command.c:152 script_debug(): command - reset_config srst_only srst_push_pull
Debug: 22 4 command.c:152 script_debug(): command - ftdi tdo_sample_edge falling
Debug: 23 4 command.c:152 script_debug(): command - ftdi layout_init 0x2088 0x3f8b
Debug: 24 4 command.c:152 script_debug(): command - ftdi layout_signal nSRST -data 0x2000
Debug: 25 4 command.c:152 script_debug(): command - ftdi layout_signal GPIO2 -data 0x2000
Debug: 26 4 command.c:152 script_debug(): command - ftdi layout_signal GPIO1 -data 0x0200
Debug: 27 4 command.c:152 script_debug(): command - ftdi layout_signal GPIO0 -data 0x0100
Debug: 28 4 command.c:152 script_debug(): command - jtag newtap riscv cpu -irlen 6 -expected-id 0x03727093
Debug: 29 4 tcl.c:418 handle_jtag_newtap_args(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 30 4 core.c:1474 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 6, capture: 0x1 mask: 0x3
Debug: 31 4 command.c:152 script_debug(): command - jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477
Debug: 32 4 tcl.c:418 handle_jtag_newtap_args(): Creating New Tap, Chip: arm_unused, Tap: tap, Dotted: arm_unused.tap, 4 params
Debug: 33 4 core.c:1474 jtag_tap_init(): Created Tap: arm_unused.tap @ abs position 1, irlen 4, capture: 0x1 mask: 0x3
Debug: 34 4 command.c:152 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu
Debug: 35 4 target.c:2171 target_free_all_working_areas_restore(): freeing all working areas
Debug: 36 4 riscv.c:429 riscv_create_target(): [riscv.cpu] riscv_create_target()
Debug: 37 4 command.c:152 script_debug(): command - riscv set_ir idcode 0x09
Debug: 38 4 command.c:152 script_debug(): command - riscv set_ir dtmcs 0x22
Debug: 39 4 command.c:152 script_debug(): command - riscv set_ir dmi 0x23
Debug: 40 4 command.c:152 script_debug(): command - adapter speed 100
Debug: 41 4 adapter.c:249 adapter_config_khz(): handle adapter khz
Debug: 42 4 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 43 4 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 44 4 command.c:152 script_debug(): command - gdb_report_data_abort enable
Debug: 45 4 command.c:152 script_debug(): command - gdb_report_register_access_error enable
Debug: 46 4 command.c:152 script_debug(): command - gdb_breakpoint_override hard
User : 47 4 gdb_server.c:4041 handle_gdb_breakpoint_override_command(): force hard breakpoints
Debug: 48 4 command.c:152 script_debug(): command - reset_config none
Debug: 49 4 command.c:152 script_debug(): command - init
Debug: 50 4 command.c:152 script_debug(): command - target init
Debug: 51 4 command.c:152 script_debug(): command - target names
Debug: 52 4 command.c:152 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 53 4 command.c:152 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 54 4 command.c:152 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 55 4 command.c:152 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 56 4 command.c:152 script_debug(): command - riscv.cpu cget -event gdb-attach
Debug: 57 4 command.c:152 script_debug(): command - riscv.cpu configure -event gdb-attach halt 1000
Debug: 58 4 target.c:1629 handle_target_init_command(): Initializing targets...
Debug: 59 4 riscv.c:442 riscv_init_target(): [riscv.cpu] riscv_init_target()
Debug: 60 4 semihosting_common.c:107 semihosting_common_init():  
Debug: 61 4 ftdi.c:732 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Debug: 62 10 mpsse.c:406 mpsse_purge(): -
Debug: 63 10 mpsse.c:687 mpsse_loopback_config(): off
Debug: 64 10 mpsse.c:732 mpsse_set_frequency(): target 100000 Hz
Debug: 65 10 mpsse.c:724 mpsse_rtck_config(): off
Debug: 66 10 mpsse.c:713 mpsse_divide_by_5_config(): off
[... 123 lines cut ...]
jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id $_EXPECTED_ID
jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME 

riscv set_ir idcode 0x09
riscv set_ir dtmcs 0x22
riscv set_ir dmi 0x23

adapter speed 100

gdb_report_data_abort enable
gdb_report_register_access_error enable
gdb_breakpoint_override hard
reset_config none

init
halt
```
I created some udev rules like this:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6014", MODE="0666"
``` 
I don't know what i'm doing wrong yesterday i managed to load a program to the core and now it doesn't work. any help or suggestion on a solution or some guidelines in order to understand what i'm doing would be helpfull.

TommyMurphyTM1234, 2023-11-21:
Why do you think that this is an OpenOCD problem as opposed to, say, an issue with the implementation of the soft IP RISC-V core on FPGA (e.g. clocks, resets, timing issues etc.)?

Perhaps the comments here might help even if they refer to a different soft IP RISC-V core albeit what seems like the same error?

* https://github.com/riscv/riscv-openocd/issues/552

TommyMurphyTM1234, 2023-12-01:
> One day i managed to load a program to my implementation and i didn't change the debug unit so i think the error is either in how i configure openocd 

What do you mean by "how I configure OpenOCD"?
You mean when building it?
Or do you mean the script used at runtime?
Regarding the latter you should probably check with the lowRISC project/community on what sort of OpenOCD script to use. And also how to investigate FPGA integration/implementation issues today might cause debug interface problems/instability.

> or in how i create the udev rules 

Udev rules are almost certainly not an issue here and can be discounted if you test by running as root on which case the udev rules are irrelevant. (Running as root/sudo is not recommended generally but I'm just suggesting it here to eliminate udev rules as suspects).

closed 2023-12-01 by wizkad

newfrogg, 2024-02-24:
Hi @wizkad, have you fixed that issue. I am porting ibex to pynqz2 and get the same problem as you. The dmcontrol=0x00 and debug module is not active. Some thing like below, openocd version is 0.11.0 . If you fixed that issue, It would like to know your solution. Thank you.
In Addition, I port this design on arty z7 and everything is OK, but when move to pynq. These are both zynq7000. So I believe this should be openocd issues, not issue performing soft IP in FPGA as @TommyMurphyTM1234 metioned. 
`User : 13 2 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 2 options.c:63 configuration_output_handler(): 
Debug: 15 2 options.c:244 add_default_dirs(): bindir=/usr/bin
Debug: 16 3 options.c:245 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 17 3 options.c:246 add_default_dirs(): exepath=/usr/bin
Debug: 18 3 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 3 configuration.c:42 add_script_search_dir(): adding /home/newfrogg/.config/openocd
Debug: 20 3 configuration.c:42 add_script_search_dir(): adding /home/newfrogg/.openocd
Debug: 21 3 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 22 3 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 23 3 configuration.c:97 find_file(): found util/pynq-z2-openocd-cfg.tcl
Debug: 24 3 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 26 3 command.c:146 script_debug(): command - transport select jtag
Debug: 27 3 command.c:146 script_debug(): command - ftdi_tdo_sample_edge falling
Debug: 29 3 command.c:146 script_debug(): command - ftdi_device_desc TUL
Debug: 31 3 command.c:146 script_debug(): command - ftdi_vid_pid 0x0403 0x6010
Debug: 33 3 command.c:146 script_debug(): command - ftdi_channel 0
Debug: 35 3 command.c:146 script_debug(): command - ftdi_layout_init 0x0088 0x008b
Debug: 37 3 command.c:146 script_debug(): command - reset_config none
Debug: 39 3 command.c:146 script_debug(): command - jtag newtap riscv cpu -irlen 6 -expected-id 0x23727093 -ignore-version
Debug: 40 3 tcl.c:571 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 5 params
Debug: 41 3 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 42 3 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 43 3 tcl.c:596 jim_newtap_cmd(): Processing option: -ignore-version
Debug: 44 3 core.c:1484 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 6, capture: 0x1 mask: 0x3
Debug: 45 3 command.c:146 script_debug(): command - jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477
Debug: 46 3 tcl.c:571 jim_newtap_cmd(): Creating New Tap, Chip: arm_unused, Tap: tap, Dotted: arm_unused.tap, 4 params
Debug: 47 3 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 48 3 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
Debug: 49 3 core.c:1484 jtag_tap_init(): Created Tap: arm_unused.tap @ abs position 1, irlen 4, capture: 0x1 mask: 0x3
Debug: 50 3 command.c:146 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu
Debug: 51 3 target.c:2172 target_free_all_working_areas_restore(): freeing all working areas
Debug: 52 3 command.c:146 script_debug(): command - riscv set_ir idcode 0x09
Debug: 54 3 command.c:146 script_debug(): command - riscv set_ir dtmcs 0x22
Debug: 56 3 command.c:146 script_debug(): command - riscv set_ir dmi 0x23
Debug: 58 4 command.c:146 script_debug(): command - adapter speed 1000
Debug: 60 4 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 61 4 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 62 4 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 63 4 command.c:146 script_debug(): command - riscv set_prefer_sba on
Debug: 65 4 command.c:146 script_debug(): command - gdb_report_data_abort enable
Debug: 67 4 command.c:146 script_debug(): command - gdb_report_register_access_error enable
Debug: 69 4 command.c:146 script_debug(): command - gdb_breakpoint_override hard
User : 71 4 gdb_server.c:3679 handle_gdb_breakpoint_override_command(): force hard breakpoints
Debug: 72 4 command.c:146 script_debug(): command - reset_config none
Debug: 74 4 command.c:146 script_debug(): command - init
Debug: 76 4 command.c:146 script_debug(): command - target init
Debug: 78 4 command.c:146 script_debug(): command - target names
Debug: 79 4 command.c:146 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 80 4 command.c:146 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 81 4 command.c:146 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 82 4 command.c:146 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 83 4 command.c:146 script_debug(): command - riscv.cpu cget -event gdb-attach
Debug: 84 4 command.c:146 script_debug(): command - riscv.cpu configure -event gdb-attach halt 1000
Debug: 85 4 target.c:1639 handle_target_init_command(): Initializing targets...
Debug: 86 4 riscv.c:427 riscv_init_target(): riscv_init_target()
Debug: 87 4 semihosting_common.c:99 semihosting_common_init():  
Debug: 88 4 ftdi.c:650 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Debug: 89 14 mpsse.c:422 mpsse_purge(): -
Debug: 90 14 mpsse.c:703 mpsse_loopback_config(): off
Debug: 91 14 mpsse.c:748 mpsse_set_frequency(): target 1000000 Hz
[... cut ...]
# just to avoid a warning about the auto-detected arm core
# see: https://github.com/pulp-platform/riscv-dbg/blob/master/doc/debug-system.md
jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

riscv set_ir idcode 0x09
riscv set_ir dtmcs 0x22
riscv set_ir dmi 0x23

adapter speed 1000

riscv set_prefer_sba on
gdb_report_data_abort enable
gdb_report_register_access_error enable
gdb_breakpoint_override hard

reset_config none

init
halt
`

TommyMurphyTM1234, 2024-02-24:
> openocd version is 0.11.0

You should at least try a more recent build of OpenOCD.
E.g. a build of the latest version from this repo.
There have been a lot of changes since 0.11.0 and some of them may be relevant to your situation.

newfrogg, 2024-02-24:
Hmm, I have updated some deprecated command, but this still fails.
[image]
The new openocd script.
```
adapter driver ftdi
transport select jtag
ftdi tdo_sample_edge falling
ftdi device_desc "TUL"
ftdi vid_pid 0x0403 0x6010
ftdi channel 0
ftdi layout_init 0x0088 0x008b
reset_config none

# Configure JTAG chain and the target processor
set _CHIPNAME riscv

# Configure JTAG expected ID
set _EXPECTED_ID 0x23727093

jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id $_EXPECTED_ID -ignore-version

# just to avoid a warning about the auto-detected arm core
# see: https://github.com/pulp-platform/riscv-dbg/blob/master/doc/debug-system.md
jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

riscv set_ir idcode 0x09
riscv set_ir dtmcs 0x22
riscv set_ir dmi 0x23

adapter speed 10000

# riscv set_prefer_sba on
gdb_report_data_abort enable
gdb_report_register_access_error enable
gdb_breakpoint_override hard

reset_config none

init
halt
```

l1onog, 2024-04-15:
> P/S: I still fail when using riscv-openocd 0.12.0+dev-03692-g3c88a95d4. Got the same error.
> 
> Hmm, I have updated some deprecated command, but this still fails. ![Screenshot from 2024-02-24 21-25-45](https://private-user-images.githubusercontent.com/85973867/307528315-eed9cfbd-a963-48aa-98d5-27ea19209e5d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxNDk5OTYsIm5iZiI6MTcxMzE0OTY5NiwicGF0aCI6Ii84NTk3Mzg2Ny8zMDc1MjgzMTUtZWVkOWNmYmQtYTk2My00OGFhLTk4ZDUtMjdlYTE5MjA5ZTVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE1VDAyNTQ1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkwMWRlZDk0NjUyZjFjZWVjZjc0Y2EyOGQzMWVkOTdjMGU2ZjYyNjU5ZWUxODgzMjcyMzNhYzk0NTEzNGYzZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.01My37ZUeKZTnlNRY6LM9wepFFNl_HFTo-NsRkWgCls) The new openocd script.
> 
> ```
> adapter driver ftdi
> transport select jtag
> ftdi tdo_sample_edge falling
> ftdi device_desc "TUL"
> ftdi vid_pid 0x0403 0x6010
> ftdi channel 0
> ftdi layout_init 0x0088 0x008b
> reset_config none
> 
> # Configure JTAG chain and the target processor
> set _CHIPNAME riscv
> 
> # Configure JTAG expected ID
> set _EXPECTED_ID 0x23727093
> 
> jtag newtap $_CHIPNAME cpu -irlen 6 -expected-id $_EXPECTED_ID -ignore-version
> 
> # just to avoid a warning about the auto-detected arm core
> # see: https://github.com/pulp-platform/riscv-dbg/blob/master/doc/debug-system.md
> jtag newtap arm_unused tap -irlen 4 -expected-id 0x4ba00477
> 
> set _TARGETNAME $_CHIPNAME.cpu
> target create $_TARGETNAME riscv -chain-position $_TARGETNAME
> 
> riscv set_ir idcode 0x09
> riscv set_ir dtmcs 0x22
> riscv set_ir dmi 0x23
> 
> adapter speed 10000
> 
> # riscv set_prefer_sba on
> gdb_report_data_abort enable
> gdb_report_register_access_error enable
> gdb_breakpoint_override hard
> 
> reset_config none
> 
> init
> halt
> ```

Hi @newfrogg, have you fixed that issue. I get the same problem as you. The dmcontrol=0x00 and debug module is not active. Some thing like below. If you fixed that issue, It would like to know your solution. Thank you.

newfrogg, 2024-04-15:
Hi @l1onog, I don't know you are dealing with ibex-demo-system or not. For my case, adding the config udev device rule for Pynq Z2 is the solution. You could use my config as reference. Also, openocd cleaned & rebuilt from source is a good start.
[https://github.com/newfrogg/ibex-demo-system?tab=readme-ov-file#add-udev-rules-for-our-device](https://github.com/newfrogg/ibex-demo-system?tab=readme-ov-file#add-udev-rules-for-our-device).

TommyMurphyTM1234, 2024-04-15:
> Hi @l1onog, I don't know you are dealing with ibex-demo-system or not. For my case, adding the config udev device rule for Pynq Z2 is the solution. You could use my config as reference. Also, [...]

I don't see any evidence that the issue mentioned above is related to, or would be solved by, `udev` rules installation. If it was then the test would not have worked at all unless `root` privileges were used e.g. via `su/sudo` in order to connect to and use the JTAG probe.