riscv-collab/riscv-openocd #968: The error "unable to resume hart 0" appears

Tim-Dong-CN, 2023-11-22:
The error "unable to resume hart 0" appears

```gdb
(gdb) ni
unable to resume hart 0
  dmstatus =0x00400383
  was stepping, halting
```

I am using remote-bitbang for simulation testing. Upon inspecting the log information from OpenOCD and reviewing my simulation waveform files, I observed that Hart 0 has already received the resume request sent by OpenOCD, and the core has resumed execution. Subsequently, it paused via a single-step. However, the operations within this Hart occur very rapidly, while JTAG transmission is comparatively slow. This results in all these operations completing before OpenOCD reads the dmstatus register for the first time after sending the resume request. As a consequence, OpenOCD incorrectly believes that the core has not resumed execution and, therefore, reports an error.

By examining the OpenOCD source code, I speculate that the logic in the following code snippet might be causing the issue. I think relying solely on checking **DM_DMSTATUS_ALLHALTED** might be too simplistic. Would it be more rigorous to incorporate the detection with the **dcsr** register for a more comprehensive verification of the check result?

```c
static int riscv013_step_or_resume_current_hart(struct target *target,
		bool step)
{
	if (target->state != TARGET_HALTED) {
		LOG_TARGET_ERROR(target, "Hart is not halted!");
		return ERROR_FAIL;
	}
	LOG_TARGET_DEBUG(target, "resuming (for step?=%d)", step);

	if (riscv_flush_registers(target) != ERROR_OK)
		return ERROR_FAIL;

	dm013_info_t *dm = get_dm(target);
	/* Issue the resume command, and then wait for the current hart to resume. */
	uint32_t dmcontrol = DM_DMCONTROL_DMACTIVE | DM_DMCONTROL_RESUMEREQ;
	dmcontrol = set_dmcontrol_hartsel(dmcontrol, dm->current_hartid);
	dm_write(target, DM_DMCONTROL, dmcontrol);

	dmcontrol = set_field(dmcontrol, DM_DMCONTROL_RESUMEREQ, 0);

	uint32_t dmstatus;
	for (size_t i = 0; i < 256; ++i) {
		usleep(10);
		if (dmstatus_read(target, &dmstatus, true) != ERROR_OK)
			return ERROR_FAIL;
		if (get_field(dmstatus, DM_DMSTATUS_ALLUNAVAIL))
			return ERROR_FAIL;
		if (get_field(dmstatus, DM_DMSTATUS_ALLRESUMEACK) == 0)
			continue;
		if (step && get_field(dmstatus, DM_DMSTATUS_ALLHALTED) == 0)
			continue;

		dm_write(target, DM_DMCONTROL, dmcontrol);
		return ERROR_OK;
	}

	dm_write(target, DM_DMCONTROL, dmcontrol);

	LOG_TARGET_ERROR(target, "unable to resume");
	if (dmstatus_read(target, &dmstatus, true) != ERROR_OK)
		return ERROR_FAIL;
	LOG_TARGET_ERROR(target, "  dmstatus=0x%08x", dmstatus);

	if (step) {
		LOG_TARGET_ERROR(target, "  was stepping, halting");
		riscv_halt(target);
		return ERROR_OK;
	}

	return ERROR_FAIL;
}
```

aap-sc, 2023-11-22:
@Tim-Dong-CN are you talking about this part of the snippet?

```
		if (get_field(dmstatus, DM_DMSTATUS_ALLRESUMEACK) == 0)
			continue;
		if (step && get_field(dmstatus, DM_DMSTATUS_ALLHALTED) == 0)
			continue;
```

? 

Could you please elaborate what's wrong with it. If I read this right the sequence is the following:

1. wait till ALLRESUMEACK is set
2. wait till DMSTATUS_ALLHALTED is set if single-step was requested.

both conditions should be true simultaneously. Looks fine to me.

In the scenario that you've described:

> This results in all these operations completing before OpenOCD reads the dmstatus register for the first time after sending the resume request.

Upon accepting resume request the hart should have `resumeack` set and `allhalted` set if the step completes. This is aligned  quite well with the above analysis. Am I missing something?

Tim-Dong-CN, 2023-11-22:
Thank you for your reply.
I think there is a problem with my understanding of the resumeack protocol. 
Of course your code is correct.
Express thanks to you!

closed 2023-11-22 by Tim-Dong-CN

en-sc, 2023-11-22:
Which OpenOCD version are you using?

> The error "unable to resume hart 0" appears
> 
> ```gdb
> (gdb) ni
> unable to resume hart 0
>   dmstatus =0x00400383
>   was stepping, halting
> ```

From the error message reported, It seems to be prior to https://github.com/riscv/riscv-openocd/commit/550a66e720 (at least year old). Maybe the problem you have can be solved by using a more recent version?

jbw0410, 2024-01-09:
[image]
Hi, I met the similar issue when running bunch of step commands using a tcl script, while no error was reported when just send a single step command through telnet

I've already updated my OpenOCD to the latest version, commit: 6a61446, and the problem still exist..

aap-sc, 2024-01-09:
@jbw0410 could you please run OpenOCD with `-d3` option passed to the command line? This will enable verbose debug logging. Please attach the output to this discussion. Most likely the log file will be huge, so you should compress it.