riscv-collab/riscv-openocd #975: negative acknowledgement, but no packet pending

hyperion009, 2023-12-04:
Hi
I am trying to set up the scenario: riscv(RTL)<--->Openocd <---> gdb. 
Here is my gdb command (target remote local:3333 in test.gdb) :
riscv64-unknown-elf-gdb  -x test.dgb hello.elf
[image]
Openocd has also reported:
[image]
Any comments are appreciated, thanks

TommyMurphyTM1234, 2023-12-04:
More info needed - for example:

* What is your RTL target and how have you verified that it has been implemented (in FPGA or simulation?) correctly?
* Where did you obtain or how did you build OpenOCD?
* How are you invoking OpenOCD?
* What OpenOCD script(s) are you using?
* You probably also need to capture, compress and attach (don't copy/paste/embed it as it will be too long) a verbose `openocd -d3` log.

hyperion009, 2023-12-05:
What is your RTL target and how have you verified that it has been implemented (in FPGA or simulation?) correctly?

- It's an in-house RISC-V implementation. and I am doing a simulation on it. 

Where did you obtain or how did you build OpenOCD?

- Git clone https://github.com/riscv/riscv-openocd directly
- using the following comand to compile:
`` `
> ./bootstrap
> ./configure --prefix=/tools/openocd/ --enable-remote-bitbang
> make
> make install
```

How are you invoking OpenOCD?

- I invoked simualtion, openocd and gdb seperatedly in a different shell.

What OpenOCD script(s) are you using?
You probably also need to capture, compress and attach (don't copy/paste/embed it as it will be too long) a verbose openocd -d3 log.
[openocd.log](https://github.com/riscv/riscv-openocd/files/13554564/openocd.log)

hyperion009, 2023-12-05:
@TommyMurphyTM1234 Hi Tommy， this issue is my fault and program code was loaded to unexpected address in rtl. thanks again.

closed 2023-12-05 by hyperion009

TommyMurphyTM1234, 2023-12-06:
> @TommyMurphyTM1234 Hi Tommy， this issue is my fault and program code was loaded to unexpected address in rtl. thanks again.

Ok - maybe you can close the issue if it's no longer relevant?