riscv-collab/riscv-openocd #979: Track and clarify differences with upstream OpenOCD

en-sc, 2023-12-14:
After the most recent merge #976 the difference with upstream version used looks like so (some differences are omitted):
```
> git checkout 1f512eac32e614e893565741f93ea3739a522797
...
> git diff d4575b647a3603200a9bb4a784d170f792ab88d0 --stat
 .github/*
 ...
 HACKING                                                                     |   26 +
 Makefile.am                                                                 |    6 +-
 configure.ac                                                                |    3 +-
 contrib/loaders/flash/gd32v/gd32vf103.inc                                   |    8 +
 doc/openocd.texi                                                            |  278 +++++++++-
 git-hooks/*
 src/flash/nor/tcl.c                                                         |    3 +-
 src/helper/Makefile.am                                                      |    2 +
 src/helper/base64.c                                                         |  157 ++++++
 src/helper/base64.h                                                         |   17 +
 src/jtag/drivers/ftdi.c                                                     |  448 +++++++++++++++-
 src/rtos/FreeRTOS.c                                                         |  763 ++++++++++++++++++++-------
 src/rtos/hwthread.c                                                         |   49 +-
 src/rtos/rtos.c                                                             |  266 ++++++++--
 src/rtos/rtos.h                                                             |   52 +-
 src/rtos/rtos_standard_stackings.c                                          |  213 +++++++-
 src/rtos/rtos_standard_stackings.h                                          |    5 +
 src/server/gdb_server.c                                                     |  176 +++++--
 src/target/breakpoints.c                                                    |  213 ++++----
 src/target/breakpoints.h                                                    |    2 +-
 src/target/dsp563xx.c                                                       |    3 +-
 src/target/riscv/*
 ...
 src/target/target.c                                                         |   96 ++--
 src/target/target.h                                                         |   12 +-
 tcl/board/esp32c2-*
 ...
 tcl/board/sifive-*
 ...
 tcl/interface/ftdi/arty-onboard-ftdi.cfg                                    |    7 +
 tcl/interface/ftdi/digilent-hs2-cjtag.cfg                                   |   17 +
 tcl/interface/ftdi/olimex-arm-jtag-cjtag.cfg                                |   27 +
 tcl/interface/ftdi/olimex-arm-usb-ocd-h-cjtag.cfg                           |   22 +
 tcl/interface/ftdi/olimex-arm-usb-tiny-h-cjtag.cfg                          |   27 +
 "tcl/target/1986\320\262\320\2651\321\202.cfg" => tcl/target/1986Be1T.cfg   |    0
 "tcl/target/\320\2721879x\320\2611\321\217.cfg" => tcl/target/K1879x61R.cfg |    0
 tcl/target/esp*
 tcl/target/gd32vf103.cfg                                                    |  105 +++-
 tools/filter_openocd_log.py                                                 |  120 +++++
```

Some of these files seem suspicious, namely:
```
 contrib/loaders/flash/gd32v/gd32vf103.inc
 src/target/dsp563xx.c
 tcl/interface/ftdi/*
 tcl/target/gd32vf103.cfg 
 ```

en-sc, 2023-12-14:
Regarding `contrib/loaders/flash/gd32v/gd32vf103.inc` -- it was moved in the upstream.
```
> git grep gd32vf103.inc
contrib/loaders/flash/gd32vf103/Makefile:all: gd32vf103.inc
src/flash/nor/stm32f1x.c:#include "../../../contrib/loaders/flash/gd32vf103/gd32vf103.inc"
> git diff contrib/loaders/flash/gd32vf103/gd32vf103.inc contrib/loaders/flash/gd32v/gd32vf103.inc
```

en-sc, 2023-12-14:
The change to `src/target/dsp563xx.c` is made by 53ec10b61d to create `riscv repeat_read`

en-sc, 2023-12-14:
There is a significant difference in `src/jtag/drivers/ftdi.c` and `tcl/interface/ftdi/*`. Some are quite recent (2022). @timsifive, can you please clarify whether these changes are RISC-V specific in some way? Is it possible to start upstreaming them without upstreaming the main RISC-V-specific part?

timsifive, 2023-12-14:
The ftdi changes let a user debug a target using cJTAG OSCAN1 protocol. It was added in https://github.com/riscv/riscv-openocd/pull/320. I'm not sure if it's ever seen significant use.

There are significant FreeRTOS changes as well. They support 64-bit targets, and also reorganize a bunch of code so it was easier to support RISC-V with 2 different call stack conventions. This is the one that seems the hardest to upstream since you probably need to check that it still works on some non-RISC-V target before sending it upstream.

The breakpoints changes are recent, and are causing quite a few conflicts when merging changes down. I'd prioritize upstreaming these just to reduce the work required in resolving these conflicts.

The rtos changes primarily let gdb correctly access registers from individual threads that are not the current one. This is probably important in conjunction with hwthread and multiple harts.

TommyMurphyTM1234, 2025-05-23:
Might this be an opportune time to consider a full "two way/up and down" sync of the RISC-V fork with the upstream OpenOCD?

- https://sourceforge.net/p/openocd/mailman/message/59186930/

> **Preparing for OpenOCD v1.0.0-rc1**
> 
> From: Antonio B. <bor...@gm...> - 2025-05-23 10:16:54
>
> Hello,
>
> we are going to tag OpenOCD v1.0.0-rc1 in around one month.
> It will follow a period of code freeze, during which only patches for
> bug fixes will be merged.
> The release v1.0.0 is expected before the end of the year.
>
> ...

I understand that such work is always dependent/contingent on there being sufficient voluntary support to assist with execution and this is often in short supply. If I personally can be of any assistance I'm happy to help.

I'm not sure if there are specific issues that might represent difficulties with such a sync?

Copying @ilg-ul and @AntonioBorneo in case they are interested.

ilg-ul, 2025-05-23:
I'll be happy to know that the risc-v patches are upstreamed and I'll be glad to make a new xPack release once v1.0.0 is out, but I'm afraid I can't be of great help to review the patches, my experience with risc-v debuggers is minimal.

en-sc, 2025-10-24:
Also I have a suggestion:
Perhaps we could split the sync in two, first merging everything related to RISC-V and then dealing with other changes?

<details>
<summary>I've tried to come up with the list of patches that I suggest to merge first</summary>
* https://review.openocd.org/c/openocd/+/8893
* https://review.openocd.org/c/openocd/+/8896
* https://review.openocd.org/c/openocd/+/9069
* https://review.openocd.org/c/openocd/+/9071
* https://review.openocd.org/c/openocd/+/9072
* https://review.openocd.org/c/openocd/+/9123
* https://review.openocd.org/c/openocd/+/9171
* https://review.openocd.org/c/openocd/+/9127
* https://review.openocd.org/c/openocd/+/9130
* https://review.openocd.org/c/openocd/+/9132
* https://review.openocd.org/c/openocd/+/9133
* https://review.openocd.org/c/openocd/+/9135
* https://review.openocd.org/c/openocd/+/9138
* https://review.openocd.org/c/openocd/+/9141
* https://review.openocd.org/c/openocd/+/9142
* https://review.openocd.org/c/openocd/+/9143
* https://review.openocd.org/c/openocd/+/9144
* https://review.openocd.org/c/openocd/+/9163
* https://review.openocd.org/c/openocd/+/9164
* https://review.openocd.org/c/openocd/+/9165

</details>

This will make RISC-V support in mainline OpenOCD on par with the fork and will allow us to freeze the development of the fork.

The thing is: other patches, related to SMP/RTOS hwthread/unavailable targets seem quite complex.

As for testing with `riscv-tests` on Spike I would suggest to target `spike32/64` at this stage, disregarding failures of multiprocessor targets for now.

tom-van, 2025-10-26:
> From a quick check, the scripts in this fork that need to go upstream are only:
> 
> tcl/target/gd32vf103.cfg

I believe the upstream gd32vf103.cfg is the most recent, the riscv-collab version is screwed after bad merge from upstream.

tom-van, 2025-10-28:
After lot more testing I recommend adding
9177: server: rtos: don't fake step for hwthread rtos. | https://review.openocd.org/c/openocd/+/9177
9183: Draft: fix missing thread ID in stop reply when smp-configured hart (but not hart 0) single-stepped | https://review.openocd.org/c/openocd/+/9183
to @en_sc list.

riscv-tests:
targets spike64-2, spike64-2-hwthread - passed all but UnavailableHaltedTest (expected)
targets rp2350rv, rp2350rv-hwthread - all passed (UnavailableHaltedTest  isn't applicable)

tom-van, 2025-10-31:
I also noticed `Unknown DCSR cause` error and grep showed it in numerous logs. So probably OpenOCD reads 0 instead of real DCSR value.

No problem in re-running with `--remotelogfile-enable` but I would focus to OpenOCD log part you replaced by ...

Debug: 62196 3022 riscv-013.c:1636 register_read_direct(): [mpfs.u54_4] Reading dcsr
Debug: 62197 3022 riscv-013.c:927 is_command_unsupported(): [mpfs.u54_4] Abstract command 0x2207b0 is cached as not supported
Debug: 62198 3022 riscv-013.c:930 is_command_unsupported(): [mpfs.u54_4] access register=0x2207b0 {regno=0x7b0 write=arg0 transfer=enabled postexec=disabled aarpostincrement=disabled aarsiz
e=32bit}
Debug: 62199 3022 riscv-013_reg.c:375 riscv013_reg_save(): [mpfs.u54_4] Saving fp
...
Debug: 62218 3022 riscv_reg.c:988 riscv_reg_get(): [mpfs.u54_4] Read fp: 0x80017e0
Debug: 62219 3022 program.c:33 riscv_program_write(): [mpfs.u54_4] progbuf[00] = DASM(0x7b002473)
Debug: 62220 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [0, 2)
Debug: 62221 3022 batch.c:255 log_batch(): 41b w 7b002473 @20 -> + 00000000 @00; 3i
Debug: 62222 3022 batch.c:199 log_dmi_decoded(): write:
Debug: 62223 3022 batch.c:255 log_batch(): 41b - 00000000 @00 -> + 7b041073 @20; 3i
Debug: 62224 3022 program.c:33 riscv_program_write(): [mpfs.u54_4] progbuf[01] = DASM(0x00100073)
Debug: 62225 3022 riscv-013.c:5404 riscv013_write_progbuf(): [mpfs.u54_4] Cache hit for 0x100073 @1
Debug: 62226 3022 riscv-013.c:748 riscv013_execute_abstract_command(): [mpfs.u54_4] access register=0x241000 {regno=0x1000 write=arg0 transfer=disabled postexec=enabled aarpostincrement=disabled aarsize=32bit}
Debug: 62227 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [0, 3)
Debug: 62228 3022 batch.c:255 log_batch(): 41b w 00241000 @17 -> + 00000000 @00; 3i
Debug: 62229 3022 batch.c:199 log_dmi_decoded(): write: command=0x241000 {control=0x241000}
Debug: 62230 3022 batch.c:255 log_batch(): 41b r 00000000 @16 -> + 00321008 @17; 3i
Debug: 62231 3022 batch.c:255 log_batch(): 41b - 00000000 @00 -> + 10000002 @16; 0i
Debug: 62232 3022 batch.c:199 log_dmi_decoded(): read: abstractcs=0x10000002 {datacount=2 progbufsize=0x10}
Debug: 62233 3022 riscv-013.c:474 decrement_reset_delays_counter(): [mpfs.u54_4] **resetting learned delays** (reset_delays_wait counter expired)
Debug: 62234 3022 riscv-013.c:683 abstract_cmd_batch_check_and_clear_cmderr(): [mpfs.u54_4] abstractcs=0x10000002 {datacount=2 relaxedpriv=full_checks busy=ready progbufsize=0x10 cmderr=none}
Debug: 62235 3022 riscv-013.c:748 riscv013_execute_abstract_command(): [mpfs.u54_4] access register=0x321008 {regno=0x1008 write=arg0 transfer=enabled postexec=disabled aarpostincrement=disabled aarsize=64bit}
Debug: 62236 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [0, 3)
Debug: 62237 3022 batch.c:255 log_batch(): 41b w 00321008 @17 -> + 00000000 @00; 0i
Debug: 62238 3022 batch.c:199 log_dmi_decoded(): write: command=0x321008 {control=0x321008}
Debug: 62239 3022 batch.c:255 log_batch(): 41b r 00000000 @16 -> b 00000000 @00; 0i
Debug: 62240 3022 batch.c:255 log_batch(): 41b - 00000000 @00 -> b 00000000 @00; 0i
Debug: 62241 3022 riscv.c:451 dtmcs_scan(): TAP mpfs.cpu: DTMCS: 0x10000 -> ?
Debug: 62242 3022 batch.h:86 riscv_scan_set_delay(): DM access delay is set to 1.
Debug: 62243 3022 batch.c:139 add_idle_before_batch(): [mpfs.u54_4] **Adding 1 idle cycles before the batch.**
Debug: 62244 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [1, 3)
Debug: 62244 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [1, 3)
Debug: 62245 3022 batch.c:255 log_batch(): 41b r 00000000 @16 -> + 00000000 @00; 1i
Debug: 62246 3022 batch.c:255 log_batch(): 41b - 00000000 @00 -> b 00000000 @00; 1i
Debug: 62247 3022 riscv.c:451 dtmcs_scan(): TAP mpfs.cpu: DTMCS: 0x10000 -> ?
Debug: 62248 3022 batch.h:86 riscv_scan_set_delay(): DM access delay is set to 2.
Debug: 62249 3022 batch.c:139 add_idle_before_batch(): [mpfs.u54_4] **Adding 1 idle cycles before the batch.**
Debug: 62250 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [2, 3)
Debug: 62251 3022 batch.c:255 log_batch(): 41b - 00000000 @00 -> + 10000002 @16; 2i
Debug: 62252 3022 batch.c:199 log_dmi_decoded(): read: abstractcs=0x10000002 {datacount=2 progbufsize=0x10}
Debug: 62253 3022 riscv-013.c:683 abstract_cmd_batch_check_and_clear_cmderr(): [mpfs.u54_4] abstractcs=0x10000002 {datacount=2 relaxedpriv=full_checks busy=ready progbufsize=0x10 cmderr=none}
Debug: 62254 3022 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [0, 3)
Debug: 62255 3023 batch.c:255 log_batch(): 41b r 00000000 @04 -> + 00000000 @00; 2i
Debug: 62256 3023 batch.c:255 log_batch(): 41b r 00000000 @05 -> b 00000000 @00; 2i
Debug: 62257 3023 batch.c:255 log_batch(): 41b - 00000000 @00 -> b 00000000 @00; 2i
Debug: 62258 3023 riscv.c:451 dtmcs_scan(): TAP mpfs.cpu: DTMCS: 0x10000 -> ?
Debug: 62259 3023 batch.h:86 riscv_scan_set_delay(): DM access delay is set to 3.
Debug: 62260 3023 batch.c:139 add_idle_before_batch(): [mpfs.u54_4] **Adding 1 idle cycles before the batch.**
Debug: 62261 3023 batch.c:291 riscv_batch_run_from(): [mpfs.u54_4] Running batch of scans [1, 3)
Debug: 62262 3023 batch.c:255 log_batch(): 41b r 00000000 @05 -> + 00000000 @00; 3i
Debug: 62263 3023 batch.c:199 log_dmi_decoded(): read:
Debug: 62264 3023 batch.c:255 log_batch(): 41b - 00000000 @00 -> + 00000000 @05; 3i
Debug: 62265 3023 batch.c:199 log_dmi_decoded(): read:
Debug: 62266 3023 riscv-013.c:1655 register_read_direct(): [mpfs.u54_4] dcsr = 0x0

tom-van, 2025-10-31:
Looks like re-running riscv_batch_run_from() with start_idx > 0 leads to wrong read.

tom-van, 2025-10-31:
The problem is in repeating the operation following the read op, where read data should be scanned in without issuing read op itself

Just a very quick fix:
```
diff --git a/src/target/riscv/batch.c b/src/target/riscv/batch.c
index 5acbddaf3..784f8c264 100644
--- a/src/target/riscv/batch.c
+++ b/src/target/riscv/batch.c
@@ -282,8 +282,8 @@ int riscv_batch_run_from(struct riscv_batch *batch, size_t start_idx,
        assert(batch->used_scans);
        assert(start_idx < batch->used_scans);
        assert(batch->last_scan == RISCV_SCAN_TYPE_NOP);
-       assert(!batch->was_run || riscv_batch_was_scan_busy(batch, start_idx));
-       assert(start_idx == 0 || !riscv_batch_was_scan_busy(batch, start_idx - 1));
+//REVISIT!     assert(!batch->was_run || riscv_batch_was_scan_busy(batch, start_idx));
+//     assert(start_idx == 0 || !riscv_batch_was_scan_busy(batch, start_idx - 1));
 
        if (batch->was_run)
                add_idle_before_batch(batch, start_idx, delays);
diff --git a/src/target/riscv/riscv-013.c b/src/target/riscv/riscv-013.c
index 6fa5e025b..ca5cf7b85 100644
--- a/src/target/riscv/riscv-013.c
+++ b/src/target/riscv/riscv-013.c
@@ -2557,6 +2557,12 @@ static int batch_run_timeout(struct target *target, struct riscv_batch *batch)
                const size_t new_finished_scans = riscv_batch_finished_scans(batch);
                assert(new_finished_scans >= finished_scans);
                decrement_reset_delays_counter(target, new_finished_scans - finished_scans);
+               bool last_scan_was_read = false;
+               if (new_finished_scans > 0) {
+                       struct scan_field *field = batch->fields + (new_finished_scans - 1);
+                       last_scan_was_read = buf_get_u32(field->out_value, DTM_DMI_OP_OFFSET,
+                                                                                       DTM_DMI_OP_LENGTH) == DTM_DMI_OP_READ;
+               }
                finished_scans = new_finished_scans;
                if (!riscv_batch_was_batch_busy(batch)) {
                        assert(finished_scans == batch->used_scans);
@@ -2565,6 +2571,8 @@ static int batch_run_timeout(struct target *target, struct riscv_batch *batch)
                result = increase_dmi_busy_delay(target);
                if (result != ERROR_OK)
                        return result;
+               if (last_scan_was_read)
+                       finished_scans--;
        } while (time(NULL) - start < riscv_get_command_timeout_sec());
 
        assert(result == ERROR_OK);
```

tom-van, 2025-10-31:
As I wrote, _quick fix_, so quick that not correct: if the read op the repeated batch starts from also returned some useful data in the previous batch run, data gets overwritten. The proper solution should run the first read op but avoid saving scanned in data from it. I let it for you, riscv guys :-)

en-sc, 2025-11-05:
@tom-van, I believe the failure you are observing is a HW issue. I will post a detailed analysis in #1307 shortly.