riscv-collab/riscv-openocd #982: program.c, riscv_program_exec(): remove dead code for save/restore of registers

JanMatCodasip, 2023-12-18:
The code in riscv_program_exec() dealing with save and restore of registers is not used anymore.

Remove the following:
- 2x for loop in riscv_program_exec() that deal with saving and restoring register values
- saved_registers local array
- writes_xreg array in struct riscv_program

en-sc, 2023-12-19:
I will prepare a patch with the full suggestion.

en-sc, 2023-12-19:
Calling `riscv_save_register()` on a dirty register does not change the value in the cache, since the register's value is not read from the target -- it is read from the cache (`valid` bit is set).

en-sc, 2024-01-23:
I've implemented the patch: https://github.com/riscv/riscv-openocd/pull/1003.
However, throughout the codebase `riscv_program_insert()` is used explicitly, which makes impossible automatic tracking of clobbered registers.
Therefore, I believe removing the loop as you suggested will be better.
@JanMatCodasip, can you please implement your suggestion?

JanMatCodasip, 2024-01-25:
@en-sc Of course. I will submit a pull request shortly (probably on Monday).

JanMatCodasip, 2024-02-05:
@en-sc I am sorry for delayed response. 

I have just posted the patch with the dead code removal: https://github.com/riscv/riscv-openocd/pull/1012

en-sc, 2024-03-04:
AFAIU, the issue is fixed by #1012.

closed 2024-03-04 by en-sc