riscv-collab/riscv-openocd #983: How to ensure consistency between icache and dcache when setting breakpoints

lz-bro, 2023-12-18:
I encountered some errors while setting breakpoints：
Dsiabled Dcache/Icache, set bp and it could be hit.
Enabled  Dcache/Icache, set bp, but excution can't hit this bp.

So should OpenOCD take the following steps to maintain consistency？
1. load the instruction data that needs to be modified into dCache.
2. Change to 'ebreak' and write back to Dcache.
3. Write 'ebreak' back to memory.
4. Invalid Icache, ensuring that 'ebreak' are read from memory.
I'm not sure if my understanding is correct, I hope someone can give me some advice.

TommyMurphyTM1234, 2023-12-18:
Perhaps this Lauterbach article might be of relevance here?

* https://support.lauterbach.com/kb/articles/risc-v-hw-designer-instruction-data-cache-handling-in-trace32

TommyMurphyTM1234, 2023-12-18:
This closed issue might also be relevant:

* https://github.com/riscv/riscv-openocd/issues/462

Note this comment from @timsifive:

* https://github.com/riscv/riscv-openocd/issues/462#issuecomment-604518511

>>  Instruction Cache invalidation .
>
> That's a real weakness in the RISC-V specifications.

timsifive, 2023-12-18:
OpenOCD tries to execute `fence.i` and `fence.rw.rw` after memory writes. I think that's the best we can do, since the ISA does not provide any kind of explicit cache control. Arguably we should use `cbo.flush` as well, which is defined in https://github.com/riscv/riscv-CMOs/blob/master/specifications/cmobase-v1.0.1.pdf

lz-bro, 2023-12-19:
openocd seems to used the program buffer execution of the 'FENCE' and 'FENCE.I' instructions in the following scenarios:
https://github.com/riscv/riscv-openocd/blob/feff8e005482ce3a070131e62a14b3bc0b9fbdf9/src/target/riscv/riscv-013.c#L2892-L2928
`read_memory_progbuf`
https://github.com/riscv/riscv-openocd/blob/feff8e005482ce3a070131e62a14b3bc0b9fbdf9/src/target/riscv/riscv-013.c#L4146
`write_memory_progbuf`
https://github.com/riscv/riscv-openocd/blob/feff8e005482ce3a070131e62a14b3bc0b9fbdf9/src/target/riscv/riscv-013.c#L4713
`step_or_resume`
https://github.com/riscv/riscv-openocd/blob/feff8e005482ce3a070131e62a14b3bc0b9fbdf9/src/target/riscv/riscv-013.c#L5174-L5176

TommyMurphyTM1234, 2023-12-19:
Probably only relevant to a small subset of users (those using a "legacy" debug spec 0.11 system) but a `fence.i` is also employed here in `riscv-011.c`:

`execute_resume`
https://github.com/riscv/riscv-openocd/blob/feff8e005482ce3a070131e62a14b3bc0b9fbdf9/src/target/riscv/riscv-011.c#L1154

closed 2023-12-27 by lz-bro