riscv-collab/riscv-openocd #984: Is AMP(Asymmetric multiprocessing) supported in Openocd

panciyan, 2023-12-20:
Currently seems only SMP is supported in Openocd, but AMP not supported ?
Is there any plan for AMP feature?
Thanks a lot!

en-sc, 2023-12-20:
To clarify:

* As far as I know, SMP support in OpenOCD consists of `target smp ...` command, which allows to assign targets to an SMP group. This affects the behavior of some commands, e.g. `halt` will halt all the targets in an SMP group.

* If you don't create an SMP group, you will still be able to work with the targets you define, though most actions will affect only the selected hart.
e.g.: You have 4 targets named `t0 - t3`. You wish to halt target `t0` and `t2` but leave `t1` and `t3` running:
```
> targets t0
> halt
> targets t2
> halt
```
Will do the trick.

* Note that an SMP group can be created for a subset of defined targets.

So, to sum up, I don't quite understand what form of AMP support you expect?

panciyan, 2023-12-20:
Thanks for your comment. @en-sc 
My debug scenario is:
Two riscv core, one is rv32, another is rv64. connected by daisy chain for debugging.
My cfg file like:
...
jtag newtap  1
jtag newtap  2
target create 1
target create 2
...
I can start two gdb to connect to the openocd, the gdb port is 3333 and 3334, and load two elf files to the targets by the two gdb,  then debug the elf as normal, in this way only one target/thread can be observed by gdb.

But I want to start only one gdb to connect to the openocd, but seems can't switch thread by command "thread xxx", and can't do step/continue for debugging the two elf files run respectively on the two core as smp can.

TommyMurphyTM1234, 2023-12-20:
> Thanks for your comment. @en-sc My debug scenario is: Two riscv core, one is rv32, another is rv64. connected by daisy chain for debugging. My cfg file like: ... jtag newtap 1 jtag newtap 2 target [...]

Whatever about OpenOCD support for simultaneously debugging heterogeneous targets aren't you effectively also asking about gdb support for same? I don't think that standard gdb supports that? Unlike some forks such as AMD's ROCgdb.

* https://rocm.docs.amd.com/projects/ROCgdb/en/latest/

timsifive, 2023-12-20:
> Two riscv core, one is rv32, another is rv64. connected by daisy chain for debugging.

This works fine in OpenOCD. We even have regression tests for that scenario.

However, as @TommyMurphyTM1234 points out, gdb can't handle it. gdb requires each target to be the same. (Almost the same. Having one target with FPU and one without mostly works, but differing XLEN does not.)

TommyMurphyTM1234, 2023-12-20:
FWIW another (commercial) solution for heterogenous target debug is Ashling's RiscFree:

* https://www.ashling.com/multi-core/

I suspect that there are others too.
Just mentioning this in case it's useful for background info.
I have no affiliation with Ashling.

panciyan, 2023-12-21:
Yes, for Ashling's RiscFree，two gdb started in eclipse IDE. I suspect we can do the similar thing for riscv64 + riscv32 heterogeneous targets, also two gdb connect to the openocd.

closed 2024-01-04 by panciyan