riscv-collab/riscv-openocd #985: IR capture error

lz-bro, 2023-12-22:
I'm using OpenOCD to connect the board successfully
```
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link V9 compiled May  7 2021 16:26:12
Info : Hardware version: 9.40
Info : VTarget = 3.264 V
Info : clock speed 2000 kHz
Info : JTAG tap: post.unknown tap/device found: 0x08501c0d
Info : JTAG tap: pre.unknown enabled
Info : JTAG tap: riscv.cpu enabled
Info : [riscv.cpu.0] datacount=2 progbufsize=2
Info : [riscv.cpu.0] Vector support with vlenb=32
Info : [riscv.cpu.0] Examined RISC-V core; found 4 harts
Info : [riscv.cpu.0]  XLEN=64, misa=0x8000000000b4112f
[riscv.cpu.0] Target successfully examined.
adapter speed: 12000 kHz
Info : starting gdb server for riscv.cpu.0 on 1024
Info : Listening on port 1024 for gdb connections
```
But, 'reset halt'
```
Info : JTAG tap: post.unknown tap/device found: 0x08501c0d 
Error: IR capture error at bit 9, saw 0x0401 not 0x...3
Warn : Bypassing JTAG setup events due to errors
```
What is the reason for this?
Thanks.

TommyMurphyTM1234, 2023-12-22:
What's the nature of the RISC-V target?
Seems a bit odd that the IDCODE cannot be decoded at all.
Can you post a full `openocd -d3` log, please?

lz-bro, 2023-12-22:
```
Debug: 25 1 command.c:152 script_debug(): command - jtag newtap pre unknown -irlen 1 -expected-id 0x00000000 -disable
Debug: 26 1 tcl.c:418 handle_jtag_newtap_args(): Creating New Tap, Chip: pre, Tap: unknown, Dotted: pre.unknown, 5 params
Debug: 27 1 core.c:1474 jtag_tap_init(): Created Tap: pre.unknown @ abs position 0, irlen 1, capture: 0x1 mask: 0x3
Debug: 28 1 command.c:152 script_debug(): command - jtag configure pre.unknown -event tap-enable
Debug: 29 1 command.c:152 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x10000E21 -disable
Debug: 30 1 tcl.c:418 handle_jtag_newtap_args(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 5 params
Debug: 31 1 core.c:1474 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 32 1 command.c:152 script_debug(): command - jtag configure riscv.cpu -event tap-enable
Debug: 33 1 command.c:152 script_debug(): command - jtag newtap post unknown -irlen 9 -expected-id 0x08501C0D
Debug: 34 1 tcl.c:418 handle_jtag_newtap_args(): Creating New Tap, Chip: post, Tap: unknown, Dotted: post.unknown, 4 params
Debug: 35 1 core.c:1474 jtag_tap_init(): Created Tap: post.unknown @ abs position 2, irlen 9, capture: 0x1 mask: 0x3
Debug: 36 1 command.c:152 script_debug(): command - target create riscv.cpu.0 riscv -chain-position riscv.cpu
Debug: 37 1 target.c:2171 target_free_all_working_areas_restore(): freeing all working areas
Debug: 38 1 riscv.c:437 riscv_create_target(): [riscv.cpu.0] riscv_create_target()
...
Info : 74 2 jlink.c:707 jlink_init(): J-Link V9 compiled May  7 2021 16:26:12
Info : 87 3 jlink.c:748 jlink_init(): Hardware version: 9.40
Info : 96 3 jlink.c:789 jlink_init(): VTarget = 3.264 V
Debug: 102 4 jlink.c:514 jaylink_log_handler(): Last read operation left 16 bytes in the buffer
Debug: 112 4 jlink.c:942 jlink_reset(): TRST: 0, SRST: 0
Debug: 132 8 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 133 8 adapter.c:217 adapter_khz_to_speed(): have adapter set up
Debug: 140 8 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 141 8 adapter.c:217 adapter_khz_to_speed(): have adapter set up
Info : 142 8 adapter.c:177 adapter_init(): clock speed 2000 kHz
Debug: 143 8 openocd.c:133 handle_init_command(): Debug Adapter init complete
Debug: 144 8 command.c:152 script_debug(): command - transport init
Debug: 145 8 transport.c:219 handle_transport_init(): handle_transport_init
Debug: 146 8 jlink.c:942 jlink_reset(): TRST: 0, SRST: 0
Debug: 151 8 core.c:830 jtag_add_reset(): SRST line released
Debug: 152 8 core.c:855 jtag_add_reset(): TRST line released
Debug: 153 8 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 154 8 command.c:152 script_debug(): command - jtag arp_init
Debug: 155 8 core.c:1509 jtag_init_inner(): Init JTAG chain
Debug: 156 8 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 166 8 core.c:1234 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 167 8 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Info : 177 9 core.c:1133 jtag_examine_chain_display(): JTAG tap: post.unknown tap/device found: 0x08501c0d (mfg: 0x606 (Shenzhen Chixingzhe Technology Co Ltd), part: 0x8501, ver: 0x0)
Debug: 178 9 core.c:1364 jtag_validate_ircapture(): IR capture validation scan
Debug: 188 10 core.c:1421 jtag_validate_ircapture(): post.unknown: IR capture 0x1
Debug: 189 10 tcl.c:551 jtag_tap_handle_event(): JTAG tap: post.unknown event: 1 (setup)
        action:
    global jtag_configured
    global _CHIPNAME
    global _TARGETNAME
    global _SPEED

    if { $jtag_configured == 0 } {
       #adapter speed 2000
       irscan post.unknown 0x98
       drscan post.unknown 16 0xa

       jtag tapenable pre.unknown
       jtag tapenable $_CHIPNAME.cpu

       set jtag_configured  1

       #adapter speed $_SPEED
    }
...
Debug: 7668 11679 gdb_server.c:416 gdb_log_incoming_packet(): [riscv.cpu.0] received packet: qRcmd,72657365742068616c74
Debug: 7669 11679 gdb_server.c:2813 gdb_query_packet(): qRcmd: reset halt
Debug: 7670 11679 command.c:152 script_debug(): command - reset halt
Debug: 7671 11679 target.c:1834 target_call_reset_callbacks(): target reset 2 (halt)
Debug: 7672 11679 command.c:152 script_debug(): command - target names
Debug: 7673 11679 command.c:152 script_debug(): command - riscv.cpu.0 invoke-event reset-start
Debug: 7674 11679 command.c:152 script_debug(): command - transport select
Debug: 7675 11679 command.c:152 script_debug(): command - jtag arp_init-reset
Debug: 7676 11679 core.c:1606 jtag_init_reset(): Initializing with hard TRST+SRST reset
Debug: 7677 11679 core.c:843 jtag_add_reset(): JTAG reset with TLR instead of TRST
Debug: 7678 11679 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 7688 11679 core.c:1509 jtag_init_inner(): Init JTAG chain
Debug: 7689 11679 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 7699 11679 core.c:1234 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 7700 11679 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Info : 7710 11680 core.c:1133 jtag_examine_chain_display(): JTAG tap: post.unknown tap/device found: 0x08501c0d (mfg: 0x606 (Shenzhen Chixingzhe Technology Co Ltd), part: 0x8501, ver: 0x0)
[... cut ...]
Debug: 7722 11680 core.c:1421 jtag_validate_ircapture(): post.unknown: IR capture 0x1
Error: 7723 11680 core.c:1431 jtag_validate_ircapture(): IR capture error at bit 9, saw 0x0401 not 0x...3
Debug: 7724 11680 gdb_server.c:432 gdb_log_outgoing_packet(): [riscv.cpu.0] sending packet: $O49522063617074757265206572726f722061742062697420392c2073617720307830343031206e6f742030782e2e2e330a#a0
Debug: 7725 11680 core.c:328 jtag_call_event_callbacks(): jtag event: TAP reset
Warn : 7735 11681 core.c:1577 jtag_init_inner(): Bypassing JTAG setup events due to errors

```

TommyMurphyTM1234, 2023-12-22:
That's not the full log.

lz-bro, 2023-12-22:
This is the full log.
[Uploading debug.txt…]()

lz-bro, 2023-12-22:
@TommyMurphyTM1234  I have solved it, thank you!

closed 2023-12-22 by lz-bro

flying-radish, 2024-03-15:
may i ask you how you solve your problem?

lz-bro, 2024-03-16:
> may i ask you how you solve your problem?

First of all, I'm not sure if you and I are the same issue. It is a clock speed issue in my HW.
Clock speed set 2000 kHz firstly，IR scan and Dr scan under 2000kHz. Then set 12000kHz after tapenable， So IR scan and Dr scan under 12000kHz if reset halt.
A reset-start event hook that set 2000 kHz to clock speed was use to solve this issue.