riscv-collab/riscv-openocd #993: semihosting issues

ZZH-Finzlize, 2024-01-11:
code: 
```C
printf("123");

FILE* fp = open("test.txt", "r");
```
linked with libsemihost by use flag '--specs=semihost.specs' and the "test.txt" is in the openocd working directory.

issues:
1. printf doesn't output string to openocd console.
2. open function return NULL

en-sc, 2024-01-11:
Can you please attach a debug log?

This can be done in two ways:
* Passing `-d3` argument to OpenOCD via command line.
* Running `debug_level 3` command (`monitor debug_level 3` if you are connected via GDB).

The log will probably be quite big. Please, consider compressing it before attaching to the issue.

You can also redirect the log into a file by either:
* Specifying `-l <filename>' argument in the command line.
* Running `log_output <filename>' OpenOCD command.

ZZH-Finzlize, 2024-01-11:
I add these commands to my cfg file:
```
riscv.cpu0 arm semihosting enable
riscv.cpu0 arm semihosting_fileio enable
debug_level 3
log_output semihosting_log.txt
```

and here is the semihosting_log.txt:
[semihosting_log.zip](https://github.com/riscv/riscv-openocd/files/13898651/semihosting_log.zip)

c code:
```C
printf("printf semihosting test.\r\n");

FILE* fp = fopen("help.txt", "r");
```

en-sc, 2024-01-11:
If I'm correct, the issue is that the OpenOCD can not detect the instructions expected to be the semihosting call.
```
Debug: 40804 15985 riscv_semihosting.c:96 riscv_semihosting(): [riscv.cpu0]    -> NONE (no magic)
```

The following sequence is expected:
```
        uint32_t magic[] = {
                0x01f01013,     /* slli    zero,zero,0x1f */
                0x00100073,     /* ebreak */
                0x40705013      /* srai    zero,zero,0x7 */
        };
```

To investigate further I will need the binary you are loading to the target. Preferably in ELF format. Can you attach it?

ZZH-Finzlize, 2024-01-11:
This is a correct behavior, because I set some breakpoints in the program, so I gusee there will be some 'ebrak' instructions for breakpoints instead of semihosting call, and if you search the key word "semihosting_common" in the log, you will find sonthing like this:
```
Debug: 52641 20339 semihosting_common.c:390 semihosting_common(): op=0xc (FLEN), param=0x4002a5c8
Debug: 52730 20355 semihosting_common.c:696 semihosting_common(): fstat(0)=0
Debug: 53256 20553 semihosting_common.c:390 semihosting_common(): op=0x9 (ISTTY), param=0x4002a5c8
Debug: 53797 20751 semihosting_common.c:390 semihosting_common(): op=0x5 (WRITE), param=0x4002a638
Debug: 66608 25001 semihosting_common.c:390 semihosting_common(): op=0x1 (OPEN), param=0x4002a6d8
```
I think that means the openocd has already identified the semihosting instruction sequence, and knows the type of the syscall, but somehow it just can't get the correct result.

en-sc, 2024-01-11:
> That is a correct behavior, because I set some breakpoints in the program.

Is the issue still present without the breakpoints?

ZZH-Finzlize, 2024-01-11:
I just found that if I remove the "riscv.cpu0 arm semihosting_fileio enable" command from my cfg file, the "printf" will work as expected, whatever the breakpoint is set or not, but "fopen" seems always return NULL, add the "semihosting_fileio" command to the cfg file doesn't make it work.

en-sc, 2024-01-11:
Have you tried using `perror() ` in the code to report why `NULL` is returned?  If I'm correct `errno` should have been set by semihosting.

TommyMurphyTM1234, 2024-01-11:
> That is a correct behavior, because I set some breakpoints in the program, so I guess there will be some 'ebreak' instructions for breakpoints instead of semihosting call

That doesn't make sense to me. According to your log, when OpenOCD tries to find the RISC-V semihosting "magic preamble sequence" mentioned by @en-sc [above](https://github.com/riscv/riscv-openocd/issues/993#issuecomment-1886589760) it finds this instead:
> ```
> Debug: 40803 15985 riscv_semihosting.c:93 riscv_semihosting(): [riscv.cpu0] compare 0xfaa43c23 from 0x40040c4c against 0x01f01013
> ```
That bears no resemblance to the expected "magic preamble sequence" or one overwritten by breakpoint instructions as far as I can see.

> I just found that if I remove the "riscv.cpu0 arm semihosting_fileio enable" command from my cfg file, the "printf" will work as expected, whatever the breakpoint is set or not, but "fopen" seems always return NULL, add the "semihosting_fileio" command to the cfg file doesn't make it work.

This also doesn't make sense to me - if `semihosting_fileio` is not enabled then file I/O cannot be routed via semihosting as far as I can see from the OpenOCD documentation and semihosting implementation.

What version of OpenOCD are you using?

erhankur, 2024-01-11:
File I/O will be routed to the gdb console if;
```
arm semihosting enable + arm semihosting_fileio enable
```

File I/O will be routed to the OpenOCD console if;
```
arm semihosting enable
```

ZZH-Finzlize, 2024-01-12:
> That doesn't make sense to me. According to your log, when OpenOCD tries to find the RISC-V semihosting "magic preamble sequence" mentioned by @en-sc [...]

The address in that log corresponds with these instructions, and the binary value of the instruction is the same as the log.

[image]

I did set a temporary break point at main function by gdb command `tb main`, so I think openocd will meet an ebreak instruction at here and find the previous instruction is not `slli zero,zero,0x1f` then openocd will handle it as a breakpoint, maybe the openocd will replace the ebreak instruction back to the original instruction, I think this does make sense.

By the way, these codes printed that log, at src/target/riscv/riscv_semihosting.c, and it should return SEMIHOSTING_NONE in that case.

```C
uint32_t magic[] = {
		0x01f01013,	/* slli    zero,zero,0x1f */
		0x00100073,	/* ebreak */
		0x40705013	/* srai    zero,zero,0x7 */
	};

	/* Read three uncompressed instructions: The previous, the current one (pointed to by PC) and the next one */
	for (int i = 0; i < 3; i++) {
		uint8_t buf[4];
		/* Instruction memories may not support arbitrary read size. Use any size that will work. */
		target_addr_t address = (pc - 4) + 4 * i;
		*retval = riscv_read_by_any_size(target, address, 4, buf);
		if (*retval != ERROR_OK)
			return SEMIHOSTING_ERROR;
		uint32_t value = target_buffer_get_u32(target, buf);
		LOG_TARGET_DEBUG(target, "compare 0x%08x from 0x%" PRIx64 " against 0x%08x",
			value, address, magic[i]);
		if (value != magic[i]) {
			LOG_TARGET_DEBUG(target, "   -> NONE (no magic)");
			return SEMIHOSTING_NONE;
		}
	}
````

> This also doesn't make sense to me - if `semihosting_fileio` is not enabled then file I/O cannot be routed via semihosting as far as I can see from the OpenOCD documentation and semihosting [...]

About this point, I Think the answer mentioned by @erhankur is correct.

> File I/O will be routed to the gdb console if;
>>arm semihosting enable + arm semihosting_fileio enable

>File I/O will be routed to the OpenOCD console if;
>>arm semihosting enable

But I didn't see anything being printed out on the gdb console when semihosting_fileio is enabled though.

[image]

> What version of OpenOCD are you using?

[image]

ZZH-Finzlize, 2024-01-12:
> Have you tried using `perror() ` in the code to report why `NULL` is returned? If I'm correct `errno` should have been set by semihosting.

It shows "Too many open files in system" when "semihosting enable".

[image]

and it crashed when "semihosting enable" + "semihosting_fileio enable", in this case, the printf will return but nothing is printed out, the perror will never return, even crtl+c can't stop the program.

[image]

erhankur, 2024-01-13:
Can you try with `arm semihosting enable` only and send us -d3 log again. Also please send your complete test application  codes related to file calls. And it would be helpful to see how did you prepare semihosting calls with the file parameters.

erhankur, 2024-01-16:
Please try
```c
#include <stdio.h>
#include <string.h>
#include <fcntl.h>
#include <unistd.h>
#include <errno.h>

int main(int argc, char** argv)
{
    printf("This is semihosting test\r\n");

    int fd = open("help.txt", 4/*O_RDWR | O_CREAT | O_TRUNC*/, 0);
    if (fd < 0) {
        printf("File open error! errno:%d\r\n", errno);
        return 0;
    }
    char str[] = "Hello from RiscV World\n";
    write(fd, str, strlen(str));
    close(fd);

    return 0;
}

```

ZZH-Finzlize, 2024-01-16:
> Please try
>
> ```c
> #include <stdio.h>
> #include <string.h>
> #include <fcntl.h>
> #include <unistd.h>
> #include <errno.h>
>
> int main(int argc, char** argv)
> {
>     printf("This is semihosting test\r\n"); [...]

It goes here:
[image]
[image]

elf file and log:
[test.zip](https://github.com/riscv/riscv-openocd/files/13944272/test.zip)

erhankur, 2024-01-16:
According to OpenOCD log OpenOCD should return 14 (fd).  What is your toolchain?

ZZH-Finzlize, 2024-01-16:
I'm using sifive toolchain: [releases](https://github.com/sifive/freedom-tools/releases)

TommyMurphyTM1234, 2024-01-16:
> I'm using sifive toolchain: [releases](https://github.com/sifive/freedom-tools/releases)

That's quite old and may not be using an old version of the Newlib semihosting support:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/802

Maybe it would be worth trying a more recent build of the toolchain - e.g.:

* https://github.com/riscv-collab/riscv-gnu-toolchain/releases

ZZH-Finzlize, 2024-01-16:
> Maybe it would be worth trying a more recent build of the toolchain - e.g.:
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain/releases

I tried this toolchain:
[image]

But still can't open file by semihosting:
[image]

erhankur, 2024-01-16:
Ok, I tried with [v11.3.0-1](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v11.3.0-1)

This is the test code
```c
    printf("This is semihosting test\r\n");

    FILE *fp = fopen("help.txt", "w+");
    if (!fp) {
        printf("File open error! errno:%d\r\n", errno);
        return 0;
    }
    fprintf(fp, "Hello again!\n");
    fclose(fp);
```

My flags;
```
-march=rv32imc --specs=semihost.specs
```

It works for me.

TommyMurphyTM1234, 2024-01-16:
I'm confused...

You say [here](https://github.com/riscv/riscv-openocd/issues/993#issuecomment-1892940068) that you get this error:

> ![image](https://private-user-images.githubusercontent.com/33342565/296905635-6af451fb-29df-404b-80c6-5543df48ae70.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0MTc2MDEsIm5iZiI6MTcwNTQxNzMwMSwicGF0aCI6Ii8zMzM0MjU2NS8yOTY5MDU2MzUtNmFmNDUxZmItMjlkZi00MDRiLTgwYzYtNTU0M2RmNDhhZTcwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE2VDE1MDE0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA3OGZjMGNkYzk1ZGIwZDA2YTBmODEwMTJhYjg1MWMxYjc3ZWRhZjc3YTIyNmJlNjJmYjk5ZWQwYTJmNzc5YmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.neKCbY7XfRFD0PYEdQDAefDrHA3mkzH2HKy_0HrJMpc)

but the log file in this zip:

> elf file and log: [test.zip](https://github.com/riscv/riscv-openocd/files/13944272/test.zip)

doesn't mention that error:
```
Debug: 26378 2617 riscv_semihosting.c:93 riscv_semihosting(): [riscv.cpu0] compare 0xfcf42623 from 0x4004024e against 0x01f01013
Debug: 26379 2617 riscv_semihosting.c:96 riscv_semihosting(): [riscv.cpu0]    -> NONE (no magic)

Line 68976: Error: 69003 60649 mpsse.c:908 mpsse_flush(): libusb_handle_events() failed with LIBUSB_ERROR_INTERRUPTED
Line 68978: Error: 69005 60652 ftdi.c:722 ftdi_execute_queue(): error while flushing MPSSE queue: -4
Line 68979: Error: 69006 60652 riscv-013.c:533 dmi_scan(): dmi_scan failed jtag scan
Line 68981: Error: 69008 60653 riscv-013.c:619 dmi_op_timeout(): [riscv.cpu1] Failed DMI read at 0x11; status=2
```

ZZH-Finzlize, 2024-01-17:
> Ok, I tried with [v11.3.0-1](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v11.3.0-1)

I still have the same issue when using that toolchain, but my flag is -march=rv64imafdc_zicsr -abi=lp64d -cmodel=medany, may be 64bit causes this issue? It seems that I have to find another board with 32bit riscv chip.

[image]

ZZH-Finzlize, 2024-01-17:
I tried to run the test code on 32bit core, but nothing changed.

[image]

Test codes are compiled with these flags -march=rv32imac_zicsr -mabi=ilp32 -mcmodel=medlow --specs=semihost.specs

ZZH-Finzlize, 2024-01-17:
Hold on, I just found that the content of "help.txt" are cleared, it is not an empty file before I open it by semihosting, I think that means "help.txt" has already been opened by OpenOCD, so this entire issue should be my problem, whatever the toolchain or anything else, I'll find it by myself. 

Thanks to everyone.

closed 2024-01-17 by ZZH-Finzlize

TommyMurphyTM1234, 2024-01-17:
> Hold on, I just found that the content of "help.txt" are cleared, it is not an empty file before I open it by semihosting, I think that means "help.txt" has already been opened by OpenOCD

That doesn't really make much sense or seem to tally with the `errno == ENFILE (23)` "Too many open files in system" error?

> may be 64bit causes this issue?

I doubt it.

erhankur, 2024-01-17:
Since `open()` is successful, errno shows the previous error or garbage value. OpenOCD doesn't clear it.

Without checking `fp` is null or not, printing `perror()` is not make sense.

ZZH-Finalize, 2024-01-17:
> Without checking `fp` is null or not, printing `perror()` is not make sense.

I manually checked fp in gdb, it is NULL, and whatever `perror` prints, `fopen` did opened the "help.txt"(beacuse the content are cleared), it just can't return a correct file pointer back.

new test code and debug view:
[image]

ZZH-Finalize, 2024-01-17:
I traced the program on instruction level, the address `0x400405ac` is the position that the semihosting call is done, so the `fd` should in register `a0`, at that position, `a0` is 13 which should be a correct `fd`, but after the `__add_fdentry` (address 0x400405bc), the value in `a0` is changed to -1

[image]
[image]

And I tried several times, I found it will always get `fd` as 13, so I write these crazy codes:
[image]

But the string `Hello from RiscV World` is printed on openocd console, "help.txt" did't change, this makes me very confused. In my opinion, the "help.txt" should be cleared or change to`Hello from RiscV World`.
[image]

erhankur, 2024-01-17:
That was my test code but you must use it with the original flags. Semihosting library should take care of the flag conversion for the OpenOCD
```c
int fd = open("help.txt", O_RDWR | O_CREAT | O_TRUNC, 0);
```

Or use the latest
```c
printf("This is semihosting test\r\n");

FILE *fp = fopen("help.txt", "w+");
if (!fp) {
    printf("File open error! errno:%d\r\n", errno);
    return 0;
}
fprintf(fp, "Hello again!\n");
fclose(fp);
```

__add_fdentry,  __get_fdentry are the internal implementation details and must be synced in each syscall. So, your hack will not work. 

Can you also define this before semihosting calls?

```c
#define RISCV_MAX_OPEN_FILES 160 /* default 16. Just increase it some big number */
```

ZZH-Finalize, 2024-01-17:
> That was my test code but you must use it with the original flags. Semihosting library should take care of the flag conversion for the OpenOCD
> 
> ```c
> int fd = open("help.txt", O_RDWR | O_CREAT | O_TRUNC, 0);
> ```

Oh, that's my fault, the "help.txt" is cleared after changing the flags.

> Can you also define this before semihosting calls?
> 
> ```c
> #define RISCV_MAX_OPEN_FILES 160 /* default 16. Just increase it some big number */
> ```

I don't know where should I define it, so I add a compile flag `-DRISCV_MAX_OPEN_FILES=160`, but it still says "Too many open files in system"
[image]

erhankur, 2024-01-17:
Yes, it is too late to define. 

What happens as a summary; OpenOCD can open the file, and return back the file descriptor. But semihost library can not add that descriptor to the internal table because of the maximum amount of open files. (16) So it returns -1 and sets errno to 23 

Question; why there is 16 open files already?

ZZH-Finalize, 2024-01-18:
> Question; why there is 16 open files already?

Test code:
```C
printf("This is semihosting test\r\n");
printf("RISCV_MAX_OPEN_FILES: %d\r\n", RISCV_MAX_OPEN_FILES);

char test_str[] = "write fd: 00";

for (size_t i = 0; i < 32; i++) {
    test_str[10] = i / 10 + '0';
    test_str[11] = i % 10 + '0';
    printf(", i: %d, ret: %d\r\n", i, write(i, test_str, sizeof(test_str)));
}
```

[image]

As we can see, fd in range 0-15 is openocd console, they occupied all fd.

TommyMurphyTM1234, 2024-01-18:
> Test code:
>
> ```c
> printf("This is semihosting test\r\n");
> printf("RISCV_MAX_OPEN_FILES: %d\r\n", RISCV_MAX_OPEN_FILES);
>
> char test_str[] = "write fd: 00";
>
> for (size_t i = 0; i < 32; i++) { [...]

Are you sure that that screenshot matches your code snippet?
That `printf()` statement also looks very odd to me.

> I should use another toolchain which can support semihosting correctly.

RISC-V GCC + Newlib + OpenOCD can do this (as @erhankur 
[showed earlier](https://github.com/riscv/riscv-openocd/issues/993#issuecomment-1893462126)) but there seems to be something strange with your setup that is causing problems for some reason.

ZZH-Finalize, 2024-01-18:
> Are you sure that that screenshot matches your code snippet?

Yes, they are matched. The `write` will run first and output "write fd: 00", then printf will output ", i: %d, ret: %d\r\n", so you can see when `i` is greater than 15, the message will start by ", i: %d, ret: %d\r\n" insted of "write fd:" because of `write` is failed and return -1.

[image]

> RISC-V GCC + Newlib + OpenOCD can do this (as @erhankur https://github.com/riscv/riscv-openocd/issues/993#issuecomment-1893462126) but there seems to be something strange with your setup that is causing problems for some reason.

Yes, I will check my setup also.