riscv-collab/riscv-openocd #999: What version of the RISC-V Debug Specification is used?

mateoconlechuga, 2024-01-17:
Hello, I am wondering if there is a way to know either through a tag, command line argument, or documentation which RISC-V debug specification is implemented? Currently it seems like 1.0, but I would like to target hardware which uses 0.13, and I am unsure where/how to get/compile a version of OpenOCD that works with different versions of the specification.

Thanks for the help.

mateoconlechuga, 2024-01-17:
The issue I am trying to deal with is the backwards-incompatible change to allow access to `tdata2` while the trigger is disabled, specifically these lines: https://github.com/riscv/riscv-openocd/blob/4fc0d86ff0c5ac45058777c9f0e5c53d3d17d7f9/src/target/riscv/riscv.c#L653-L659

The hardware I am working with does not support this modification.

TommyMurphyTM1234, 2024-01-17:
You don't need to build it for any/each specific debug specification. The one build supports most or all of all relevant debug specs:

* 0.11nov12: https://static.dev.sifive.com/pdfjs-1.9.426-dist/web/viewer.html?file=https://static.dev.sifive.com/riscv-debug-spec-0.11nov12.pdf
* 0.13.2: https://github.com/riscv/riscv-debug-spec/blob/release/riscv-debug-release.pdf
* Latest frozen draft of 1.0.0-rc1: https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-frozen.pdf

TommyMurphyTM1234, 2024-01-17:
> The issue I am trying to deal with is the backwards-incompatible change to allow access to `tdata2` while the trigger is disabled, specifically these lines: [...]

To add to what I posted above - that does not rule out the possibility of bugs where, for example, something that works in one version of the debug spec may not work in another. If you can explain in more detail what you understand the issue to be here it might help others to figure out if there's a possible explanation or fix.

TommyMurphyTM1234, 2024-01-17:
Also - it might be worth collecting a verbose (`openocd -d3`) log for the issue that you are experiencing and attaching that to this thread. That may shed more light on the problem.

TommyMurphyTM1234, 2024-01-17:
> The issue I am trying to deal with is the backwards-incompatible change to allow access to `tdata2` while the trigger is disabled, specifically these lines: [...]

I had a quick skim of the 0.13.2 debug spec and can't see anything that says that a trigger must be enabled for `tdata2` to be accessed/written. Where are you seeing this?

mateoconlechuga, 2024-01-17:
Thanks for the info! Yes I am seeing that now. Here is the debug log with the write of 0 to `tdata1` to disable the trigger, followed by the write to `tdata2`. Since the trigger is disabled, the hardware does not accept the write, and leads to the failing readback verification of `tdata2` later in the function.

aap-sc, 2024-01-17:
@mateoconlechuga could you double check this issue? https://github.com/riscv/riscv-openocd/issues/870 Does this have relevance to your case?

mateoconlechuga, 2024-01-17:
@aap-sc Yes that issue is relevant - the hardware I am using is 0.13 specification compliant and doesn't accept writes to `tdata2` when the trigger is disabled. The fix in that issue implemented the 1.0 specification which assumes that `tdata2` is writable even when the trigger is disabled, which is what I am running into.

TommyMurphyTM1234, 2024-01-17:
> @aap-sc Yes that issue is relevant - the hardware I am using is 0.13 specification compliant and doesn't accept writes to `tdata2` when the trigger is disabled. 

Can you clarify where in the 0.13 spec it states that `tdata2` cannot be written unless the trigger is disabled? I can't find it.

aap-sc, 2024-01-17:
Regarding the supported version.

@mateoconlechuga , @TommyMurphyTM1234 the current OpenOCD codebase does not make a distinction between 1.0 and 0.13. The intention is to support both. There may be indeed be inconsistencies between 1.0 and 0.13 - it's just that no one encountered so case before (at least there was no pending issue to indicate that).

1. RISCV spec allows for runtime detection of which spec is used. See `DM_DMSTATUS_VERSION_1_0 `, `DM_DMSTATUS_VERSION_0_13` , `DM_DMSTATUS_VERSION_0_11` defines. So in theory one could query this information duting examine and adjust the codebase. You may wonder why these defines are not used. This is because there another mechanism to identify if we on 0_11 or 0_13+ (see `DTM_DTMCS_VERSION_0_11`).
2. It would be nice to double check if there is really a non-backwards compatible change or is it a bug in HW implementation. Note: the reason for the non-conform ant implementation may be a non-backward compatible change to the spec itself, but it does not matter much.  I'm unable to express my opinion on the matter **today**, maybe someone else will (like @TommyMurphyTM1234), or I'll take a look tomorrow.
3. If this is a HW bug , depending of the complexity of a workaround  OpenOCD maintainers tend to accept such patches. It's just this needs to be properly documented.
4. But I should warn that there is a major problem. It's testing. In OS OpenOCD is tested on spike. And spike implementation of debug subsystem focuses mostly on 1.0 spec. It may be difficult to publish relevant changes there (though not impossible). Without a proper test case the code will eventually break even if workaround is implemented.

mateoconlechuga, 2024-01-17:
Under section `5.2 Trigger Registers`:

> If a debugger writes an unsupported configuration, the register will read back a value that is supported (which may simply be a disabled trigger).

When OpenOCD writes 0 to the `tdata1`, it effectively disables the trigger in hardware, even though it still exists.
This masks the return of the `tdata2` register and causes it to return a "valid" value of 0, even though it was written while the register was in a disabled state.

The 1.0 specification clarifies the intended behavior a lot better, and that `tdata2` should be writable in the new section:

> The combination of these rules means that a debugger cannot simply set a trigger by writing
tdata1, then tdata2, etc. The current value of tdata2 might not be legal with the new value of
tdata1. To help with this situation, it is guaranteed that writing 0 to tdata1 disables the trigger,
and leaves it in a state where tdata2 and tdata3 can be written with any value that makes sense
for any trigger type supported by this trigger.

TommyMurphyTM1234, 2024-01-17:
> @TommyMurphyTM1234
> 
> Under section `5.2 Trigger Registers`:
> 
> > If a debugger writes an unsupported configuration, the register will read back a value that is supported (which may simply be a disabled trigger).
> 
> When OpenOCD writes 0 to the `tdata1`, it effectively disables the trigger in hardware, even though it still exists. This masks the return of the `tdata2` register and causes it to return a "valid" value of 0, even though it was written while the register was in a disabled state.

Yes - but I still don't see how that mandates that accesses/writes to `tdata2` cannot be made when the trigger is enabled even if it might make more sense to only do this when the trigger is disabled?

mateoconlechuga, 2024-01-17:
I don't think I read your comment here correctly the first time:

> Can you clarify where in the 0.13 spec it states that tdata2 cannot be written unless the trigger is disabled? I can't find it.

The behavior I am experiencing is the other way around - `tdata2` cannot be written *when* the trigger is disabled.

---

Section 5.7 of the 1.0 specification clarifies this a lot more. This clarification about writing a 0 to disable the register and allow writes to tdata2 is nonexistent in the 0.13 specification, which the hardware was built against.

> Almost all trigger functionality is optional. All tdata registers follow write-any-read-legal semantics. If a debugger writes an unsupported configuration, the register will read back a value that is supported (which may simply be a disabled trigger). This means that a debugger must always read back values it writes to tdata registers, unless it already knows already what is supported. Writes to one tdata register must not modify the contents of other tdata registers, nor the configuration of any trigger besides the one that is currently selected.

> **The combination of these rules means that a debugger cannot simply set a trigger by writing tdata1, then tdata2, etc. The current value of tdata2 might not be legal with the new value of tdata1. [...]

aap-sc, 2024-01-17:
@mateoconlechuga a note regarding the spec interpretation. Such questions are better to be raised here: https://github.com/riscv/riscv-debug-spec . I believe RISCV debug workgroup still provides support for both versions of the spec

mateoconlechuga, 2024-01-17:
@aap-sc Thanks! I don't believe that this is a specification interpretation issue though.

This is the backwards-incompatible specification change that is called out in the 1.0.0 specification that OpenOCD currently does not support:

`1.2.1.2 Incompatible Changes from 0.13 to 1.0`
> 6. When a selected trigger is disabled, tdata2 and tdata3 can be written with any value supported by any of the types this trigger supports.

aap-sc, 2024-01-18:
@mateoconlechuga 

> I don't believe that this is a specification interpretation issue though.

Yes, the note you've mentioned looks convincing enough to me. So if you are going to submit a patch - please, consider creating a patch for spike, so we could test this case.