riscv-collab/riscv-openocd #1000: Is AP_TYPE_JTAG_AP supported?

coolwinner, 2024-01-22:
I checked the code, and I found it seems only MEM_AP is supported in the file mem_ap.c , but I can't find any files related with JTAG_AP.

Thanks!

TommyMurphyTM1234, 2024-01-22:
Any use? Seems to have been an attempt to add JTAG_AP support.

* https://review.openocd.org/c/openocd/+/4875

But it seems to be still open after several years.

What exactly is the relevance of your query to RISC-V support?

Maybe this is something that you need to raise on the upstream mailing lists or issue tracker?

* https://openocd.org/pages/mailing-lists.html
* https://openocd.org/pages/bug-tracker.html

coolwinner, 2024-01-22:
Yes, the RISC-V implementation that use ADI (Arm Debug Interface) v5 is as below:

[image]

But openocd seems not support Dap in riscv target, could you give some suggestions?

Thanks!

TommyMurphyTM1234, 2024-01-22:
> Yes, the RISC-V implementation that use ADI (Arm Debug Interface) v5 is as below: [...]

What is this RISC-V implementation?

coolwinner, 2024-01-23:
> What is this RISC-V implementation?

Sorry, I am afraid you may misunderstand my mean. I mean our hardware design is as the image I shown in above comment, but we want to debug with openocd, it seems current openocd not support the design. So could you give me some suggestions about how to support the hardware design with openocd?

Thanks!

TommyMurphyTM1234, 2024-01-28:
According to the OpenOCD documentation:

* https://openocd.org/doc/html/Architecture-and-Core-Commands.html#:~:text=16.11%20RISC%2DV,implement%20version%200.11.

>  OpenOCD supports JTAG debug of RV32 and RV64 cores 

There is no mention of any ADI support.

It might be worth asking this question in the upstream OpenOCD mailing lists?

> How to support ADI (Arm Debug Interface) v5 in RISC-V

* https://openocd.org/pages/mailing-lists.html

zqb-all, 2024-06-05:
hi @coolwinner  ,has there been any progress on this issue so far ? It seems that JTAG-AP is completely without any support, and even though openocd has some support for MEM-AP, it will require further modifications to support RISC-V DM behind MEM-AP

coolwinner, 2024-06-11:
> hi @coolwinner ,has there been any progress on this issue so far ? It seems that JTAG-AP is completely without any support, and even though openocd has some support for MEM-AP, it will require further modifications to support RISC-V DM behind MEM-AP

Hi @zqb-all ,
We changed our design and no longer used JTAG-AP, and we are doing some modifications to support RISC-V DM behind MEM-AP.

zqb-all, 2024-06-11:
> We changed our design and no longer used JTAG-AP, and we are doing some modifications to support RISC-V DM behind MEM-AP.

Got it. Thanks for your reply.

ashig8, 2024-11-27:
> Hi @zqb-all , We changed our design and no longer used JTAG-AP, and we are doing some modifications to support RISC-V DM behind MEM-AP.

Hello @coolwinner 
I am working on a similar design where RISCV-DM is behind MEM-AP. As you mentioned that you have done some modifications to support this, were you able to debug riscv ?  can you pls share some details on how this can be achieved ? is there any commit or branch for reference ?