riscv-collab/riscv-openocd #1004: Minor issue: Different error message on removal of non-existent breakpoint

MarekVCodasip, 2024-01-25:
Due to differences between vanilla and riscv-openocd, there is a different error message printed when removing non-existent breakpoint on SMP vs non-SMP targets. (Difference manifests on non-SMP)

Non SMP target:
`Error during removal of breakpoint at address (addr)`

SMP target:
`no breakpoint at address (addr) found`

See [src/target/breakpoints.c:335](https://github.com/riscv/riscv-openocd/blob/riscv/src/target/breakpoints.c#L335) for more details.

aap-sc, 2024-01-25:
@MarekVCodasip , @en-sc in my opinion the commit that introduced this discrepancy is rather controversial . I believe it was @timsifive who introduced the change, though.

Does anyone know what was the reason for introducing this and why the patch is not in the upstream openocd?

aap-sc, 2024-01-25:
To be clear, i'm talking about this commit: 39a4f37f84d52aa38ffa1a7db703ad57deec4fdc

aap-sc, 2024-01-25:
@TommyMurphyTM1234 actually it does :). To my surprise I've already asked a question regarding the upstream changes. https://github.com/riscv/riscv-openocd/pull/879#discussion_r1310614788 O_o. And even got a reply (and acknowledged it). LOL. So basically there is a chunk of functionality related to unavailable target support which is not (or was not) upstreamed at the time. 

Looking at the current code now - I have minor concerns that it may be difficult to upstream this specific functionality.