riscv-collab/riscv-openocd #1005: Flush dirty registers on shutdown is broken

fborisovskii, 2024-01-25:
Hello,

After merging to commit
https://github.com/riscv/riscv-openocd/commit/8dbb1250f587484774752e8610f04a98157ca10a
I've got Error in oenocd log after sending monitor shutdown from gbd session.

```
Debug: 17077 1640 gdb_server.c:438 gdb_log_outgoing_packet(): [riscv.cpu0] {1} sending packet: $OK#9a
Debug: 17078 1641 gdb_server.c:420 gdb_log_incoming_packet(): [riscv.cpu0] {1} received packet: qRcmd,73687574646f776e
Debug: 17079 1641 gdb_server.c:2820 gdb_query_packet(): qRcmd: shutdown
Debug: 17080 1641 command.c:154 script_debug(): command - shutdown
User : 17081 1641 server.c:758 handle_shutdown_command(): shutdown command invoked
Debug: 17082 1641 gdb_server.c:438 gdb_log_outgoing_packet(): [riscv.cpu0] {1} sending packet: $O73687574646f776e20636f6d6d616e6420696e766f6b65640a#87
Debug: 17083 1641 gdb_server.c:438 gdb_log_outgoing_packet(): [riscv.cpu0] {1} sending packet: $EA8#be
Debug: 17084 1641 gdb_server.c:1154 gdb_connection_closed(): {1} GDB Close, Target: riscv.cpu0, state: halted, gdb_actual_connections=0
Debug: 17085 1641 target.c:1790 target_call_event_callbacks(): target event 8 (gdb-end) for core riscv.cpu0
Debug: 17086 1641 target.c:1790 target_call_event_callbacks(): target event 23 (gdb-detach) for core riscv.cpu0
Debug: 17087 1641 breakpoints.c:319 breakpoint_remove_all_internal(): [riscv.cpu0] Delete all breakpoints
Debug: 17088 1641 riscv.c:535 riscv_deinit_target(): [riscv.cpu0] riscv_deinit_target()
Debug: 17089 1641 riscv.c:1788 riscv_flush_registers(): [riscv.cpu0] Flushing register cache
Debug: 17090 1641 riscv.c:1800 riscv_flush_registers(): [riscv.cpu0] fp is dirty; write back 0x80004054
Debug: 17091 1641 riscv-013.c:4874 riscv013_set_register(): [riscv.cpu0] writing 0x80004054 to register fp
Debug: 17092 1641 riscv-013.c:1691 register_write_direct(): [riscv.cpu0] Writing 0x80004054 to fp
Debug: 17093 1641 riscv-013.c:892 execute_abstract_command(): [riscv.cpu0] access register=0x231008 {regno=0x1008 write=register transfer=enabled postexec=disabled aarpostincrement=disabled aarsize=32bit}
Error: 17094 1641 riscv-013.c:1679 register_write_progbuf(): [riscv.cpu0] Unexpected write to fp via program buffer.
Error: 17095 1641 riscv.c:543 riscv_deinit_target(): [riscv.cpu0] Failed to flush registers. Ignoring this error.
Debug: 17096 1641 riscv-013.c:1861 deinit_target(): [riscv.cpu0] Deinitializing target.
Debug: 17097 1641 target.c:2144 target_free_all_working_areas_restore(): freeing all worki
```

Futher invetigations comes to check  in dmi_op_timeout function:
```
	if (openocd_is_shutdown_pending()) {
		return ERROR_SERVER_INTERRUPTED;
	}
```

When riscv_deinit_target is called after shutdown command received shutdown pending is already turned on.
So errors from dmi_op_timeout is unavoidable if there are any dirty register.

Regards,
Fedor

JanMatCodasip, 2024-01-26:
Hi @fborisovskii, thank you for the report and taking the time to investigate further.

@en-sc It looks like we should not react on `openocd_is_shutdown_pending()` in `dmi_op_timeout()` (and maybe also not in riscv_*_register()). That's because there may still be valid DMI operations to make while deinitializing the target, as @fborisovskii proved.

en-sc, 2024-01-26:
@fborisovskii, thank you for the catch!
@JanMatCodasip, I don't think this will be sufficient. IMHO, the patch should be reverted (as well as mainline patch).
The reason is: there is `pre_shutdown_commands` list. Any command can be added to this list by the user, and it should be possible for these commands to be executed.
However, in the current implementation it is not possible since `openocd_is_shutdown_pending()` is already `true` (`shutdown_openocd` is set to `SHUTDOWN_REQUESTED`) prior to executing the commands from the list.
I will prepare the revert.

JanMatCodasip, 2024-01-30:
@fborisovskii Please let us know if #1006 resolved your issue, and whether we can close this thread. Thank you.

fborisovskii, 2024-01-30:
Hello,

I've checked the revert. Issue is resolved. 

Thank you

closed 2024-01-30 by fborisovskii