riscv-collab/riscv-openocd #1024: Debug Module did not become active. dmcontrol=0x0

lz-bro, 2024-03-01:
Hello, the following error occurred when I try to connect to openocd:

```
Debug: 243 14 riscv.c:1727 riscv_examine(): [riscv.cpu0] Starting examination
Debug: 253 15 riscv.c:406 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x7101
Debug: 254 15 riscv.c:1741 riscv_examine(): [riscv.cpu0] dtmcontrol=0x7101
Debug: 255 15 riscv.c:1743 riscv_examine(): [riscv.cpu0] version=0x1
Debug: 256 15 riscv-013.c:2776 init_target(): [riscv.cpu0] Init.
Debug: 257 15 riscv-013.c:1922 examine(): [riscv.cpu0] dbgbase=0x0
Debug: 267 15 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x0 -> 0x7101
Debug: 268 15 riscv-013.c:1930 examine(): [riscv.cpu0] dtmcontrol=0x7101
Debug: 269 15 riscv-013.c:1931 examine(): [riscv.cpu0] dtmcs=0x7101 {version=1_0 idle=7 errinfo=not_implemented abits=0x10}
Debug: 270 15 riscv-013.c:791 check_dbgbase_exists(): [riscv.cpu0] Searching for DM with DMI base address (dbgbase) = 0x0
Debug: 271 15 riscv-013.c:250 get_dm(): [riscv.cpu0] Coreid [0] Allocating new DM
Debug: 272 15 riscv-013.c:520 dmi_scan(): [riscv.cpu0] reset_delays_wait done
Debug: 282 15 riscv-013.c:420 dump_field(): 50b w 00000000 @10 -> + 00000000 @16; 0i
Debug: 283 15 riscv-013.c:428 dump_field(): write: dmcontrol=0 {}
Debug: 293 16 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000000 @10; 0i
Debug: 294 16 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=1, ac_busy_delay=0
Debug: 304 16 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 314 16 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 1i
Debug: 324 17 riscv-013.c:420 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 1i
Debug: 325 17 riscv-013.c:428 dump_field(): write: dmcontrol=1 {dmactive=active}
Debug: 335 17 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000001 @10; 1i
Debug: 336 17 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Debug: 346 17 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 356 18 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 2i
Debug: 357 18 riscv-013.c:5181 riscv013_invalidate_cached_progbuf(): [riscv.cpu0] Invalidating progbuf cache
Debug: 367 18 riscv-013.c:420 dump_field(): 50b w 07ffffc1 @10 -> + 00000000 @10; 2i
Debug: 368 18 riscv-013.c:428 dump_field(): write: dmcontrol=0x7ffffc1 {dmactive=active hartsello=0x3ff hasel=multiple hartselhi=0x3ff}
Debug: 378 18 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 07ffffc1 @10; 2i
Debug: 379 18 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=3, ac_busy_delay=0
Debug: 389 19 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 399 19 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 3i
Debug: 409 19 riscv-013.c:420 dump_field(): 50b r 00000000 @10 -> + 00000000 @10; 3i
Debug: 419 20 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 04010001 @10; 3i
Debug: 420 20 riscv-013.c:434 dump_field(): read: dmcontrol=0x4010001 {dmactive=active hartsello=1 hasel=multiple}
Debug: 430 20 riscv-013.c:420 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 3i
Debug: 431 20 riscv-013.c:428 dump_field(): write: dmcontrol=1 {dmactive=active}
Debug: 441 20 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000001 @10; 3i
Debug: 451 21 riscv-013.c:420 dump_field(): 50b r 00000000 @11 -> + 00000000 @10; 3i
Debug: 461 21 riscv-013.c:420 dump_field(): 50b - 00000000 @11 -> + 004003a3 @11; 3i
Debug: 462 21 riscv-013.c:434 dump_field(): read: dmstatus=0x4003a3 {version=1_0 impebreak=1 hasresethaltreq=1 authenticated=true anyhalted=1 allhalted=1}
Debug: 463 21 riscv-013.c:1986 examine(): [riscv.cpu0] dmstatus:  0x004003a3
...
...
...
Debug: 2144 71 riscv.c:1727 riscv_examine(): [riscv.cpu1] Starting examination
Debug: 2154 72 riscv.c:406 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x7101
Debug: 2155 72 riscv.c:1741 riscv_examine(): [riscv.cpu1] dtmcontrol=0x7101
Debug: 2156 72 riscv.c:1743 riscv_examine(): [riscv.cpu1] version=0x1
Debug: 2157 72 riscv-013.c:2776 init_target(): [riscv.cpu1] Init.
Debug: 2158 72 riscv-013.c:1922 examine(): [riscv.cpu1] dbgbase=0x0
Debug: 2168 72 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x0 -> 0x7101
Debug: 2169 72 riscv-013.c:1930 examine(): [riscv.cpu1] dtmcontrol=0x7101
Debug: 2170 72 riscv-013.c:1931 examine(): [riscv.cpu1] dtmcs=0x7101 {version=1_0 idle=7 errinfo=not_implemented abits=0x10}
Debug: 2171 72 riscv-013.c:791 check_dbgbase_exists(): [riscv.cpu1] Searching for DM with DMI base address (dbgbase) = 0x0
Debug: 2172 72 riscv-013.c:5181 riscv013_invalidate_cached_progbuf(): [riscv.cpu1] Invalidating progbuf cache
Debug: 2173 72 riscv-013.c:520 dmi_scan(): [riscv.cpu1] reset_delays_wait done
Debug: 2183 73 riscv-013.c:420 dump_field(): 50b w 07ffffc1 @10 -> + 00000000 @16; 0i
Debug: 2184 73 riscv-013.c:428 dump_field(): write: dmcontrol=0x7ffffc1 {dmactive=active hartsello=0x3ff hasel=multiple hartselhi=0x3ff}
Debug: 2194 73 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 07ffffc1 @10; 0i
Debug: 2195 73 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=1, ac_busy_delay=0
Debug: 2205 73 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 2215 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 1i
Debug: 2225 74 riscv-013.c:420 dump_field(): 50b r 00000000 @10 -> + 00000000 @10; 1i
Debug: 2235 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000000 @10; 1i
Debug: 2236 74 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Debug: 2246 75 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 2256 75 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 2i
Debug: 2257 75 riscv-013.c:434 dump_field(): read: dmcontrol=0 {}
Debug: 2267 75 riscv-013.c:420 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 2i
Debug: 2268 75 riscv-013.c:428 dump_field(): write: dmcontrol=1 {dmactive=active}
Debug: 2278 75 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000001 @10; 2i
Debug: 2279 75 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=3, ac_busy_delay=0
Debug: 2289 76 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 2299 76 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 3i
Error: 2300 76 riscv-013.c:1976 examine(): [riscv.cpu1] Debug Module did not become active. dmcontrol=0x0
Error: 2301 76 target.c:681 target_examine_one(): [riscv.cpu1] Examination failed
```
I made the following modifications to successfully connect：
1. Reset every DM
dm_write(target, DM_DMCONTROL, 0);
dm_write(target, DM_DMCONTROL, 1);
https://github.com/riscv-collab/riscv-openocd/blob/ca7d88252633a5d452ee92e6f8f3eb20cadfb633/src/target/riscv/riscv-013.c#L1954-L1958
2. dmi_busy_delay set equal to dtmcs_idle

TommyMurphyTM1234, 2024-03-01:
Probably worth reading these previously posted issues and heeding the suggestions and feedback posted there:

* https://github.com/riscv-collab/riscv-openocd/issues/966
* https://github.com/riscv-collab/riscv-openocd/issues/552
* https://github.com/riscv-collab/riscv-openocd/issues/427

If yours is an FPGA implementation of a RISC-V then you should first double check your target (e.g. in simulation, checking all relevant reset/clock signals, checking all synthesis/place and route logs for warnings/errors, using logic analysis if necessary etc.) to ensure that it's fully functional.

lz-bro, 2024-03-01:
Is this a bit unreasonable? The success of the core0 exam feels like a coincidence. When performing a scan operation, increasing dmi_busy_delay after busy will not repeat the current operation.
```
Debug: 282 15 riscv-013.c:420 dump_field(): 50b w 00000000 @10 -> + 00000000 @16; 0i
Debug: 283 15 riscv-013.c:428 dump_field(): write: dmcontrol=0 {}
Debug: 293 16 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000000 @10; 0i
Debug: 294 16 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=1, ac_busy_delay=0
Debug: 304 16 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 314 16 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 1i
Debug: 324 17 riscv-013.c:420 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 1i
Debug: 325 17 riscv-013.c:428 dump_field(): write: dmcontrol=1 {dmactive=active}
Debug: 335 17 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000001 @10; 1i
Debug: 336 17 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Debug: 346 17 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 356 18 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 2i
Debug: 357 18 riscv-013.c:5181 riscv013_invalidate_cached_progbuf(): [riscv.cpu0] Invalidating progbuf cache
Debug: 367 18 riscv-013.c:420 dump_field(): 50b w 07ffffc1 @10 -> + 00000000 @10; 2i
Debug: 368 18 riscv-013.c:428 dump_field(): write: dmcontrol=0x7ffffc1 {dmactive=active hartsello=0x3ff hasel=multiple hartselhi=0x3ff}
Debug: 378 18 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 07ffffc1 @10; 2i
Debug: 379 18 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu0] dtmcs_idle=7, dmi_busy_delay=3, ac_busy_delay=0
Debug: 389 19 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 399 19 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 3i
Debug: 409 19 riscv-013.c:420 dump_field(): 50b r 00000000 @10 -> + 00000000 @10; 3i
Debug: 419 20 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 04010001 @10; 3i
Debug: 420 20 riscv-013.c:434 dump_field(): read: dmcontrol=0x4010001 {dmactive=active hartsello=1 hasel=multiple}
Debug: 430 20 riscv-013.c:420 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 3i
```

en-sc, 2024-03-01:
Are you running on Spike or on some other simulator/HW?

lz-bro, 2024-03-01:
> Are you running on Spike or on some other simulator/HW?

In our HW, and my cfg is:
```
adapter driver jlink
transport select jtag
adapter speed 1000
# adapter serial xxx

bindto 0.0.0.0
gdb_port 1024

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1

target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -coreid 0
target create $_TARGETNAME_1 riscv -chain-position $_CHIPNAME.cpu -coreid 1

$_TARGETNAME_0 configure -event examine-end {
    #write register, enable multicore
    mww 0xf00110004 0x3
}

foreach t [target names] {
    targets $t
    #riscv set_mem_access sysbus progbuf
    riscv set_mem_access progbuf
}
```

en-sc, 2024-03-01:
AFAIU, this is a HW issue.
Let's look at the examination of the `riscv.cpu1` hart, the one that resulted in the error.
* Request to reset the DM is issued (`dmcontrol.dmactive = 0`):
```
Debug: 2183 73 riscv-013.c:420 dump_field(): 50b w 07ffffc1 @10 -> + 00000000 @16; 0i
Debug: 2184 73 riscv-013.c:428 dump_field(): write: dmcontrol=0x7ffffc1 {dmactive=active hartsello=0x3ff hasel=multiple hartselhi=0x3ff}
```
* DMI NOP is issued to check the result of the write
```
Debug: 2194 73 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 07ffffc1 @10; 0i
```
* DMI NOP returned `dmi.busy`, so delays are increased and `dtmcs.dmireset` is issued. Note, that `dtmcs.dmireset` **does not cancel the outstanding transaction** [6.1.4 DTM Control and Status (dtmcs, at 0x10) in RISC-V Debug Specification Version 1.0.0-rc1].
```
Debug: 2195 73 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=1, ac_busy_delay=0
Debug: 2205 73 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
```
* One more NOP is issued to check the result of the write. The result indicates success.
```
Debug: 2215 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 1i
```
* The debugger reads `dmcontrol` to check that the DM has been reset. (DMI busy is encountered along the way and processed in a similar fashion). Returned value indicates `dmcontrol.dmactive == 0`, meaning the DM did not become active.
```
Debug: 2225 74 riscv-013.c:420 dump_field(): 50b r 00000000 @10 -> + 00000000 @10; 1i
Debug: 2235 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000000 @10; 1i
Debug: 2236 74 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Debug: 2246 75 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 2256 75 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 2i
Debug: 2257 75 riscv-013.c:434 dump_field(): read: dmcontrol=0 {}
```
* Which is exactly what is reported in the end:
```
....
Error: 2300 76 riscv-013.c:1976 examine(): [riscv.cpu1] Debug Module did not become active. dmcontrol=0x0
Error: 2301 76 target.c:681 target_examine_one(): [riscv.cpu1] Examination failed
```

### TLDR:
Seems to me, writing `dtmcs.dmireset` cancels the outstanding DMI transaction on your HW. This is explicitly prohibited by the spec ( see [6.1.4 DTM Control and Status (dtmcs, at 0x10) in RISC-V Debug Specification Version 1.0.0-rc1]).
You can hide the issue by increasing the delay and avoiding DMI busy. However, this will not fix the underlying issue and I would advise against doing so -- DMI busy will be encountered eventually, and it's much easier to analyze the issue when it's a part of examination routine rather then when it's encountered during some other complex operation.

en-sc, 2024-03-01:
I have misread the logs and my initial analysis was not correct. I've updated the comment.

lz-bro, 2024-03-04:
@en-sc Thank you for your detailed answer, writing dtmcs.dmireset cancels the outstanding DMI transaction on our HW, this is not in compliance with the spec. Is this what you mean?

en-sc, 2024-03-04:
> ... writing dtmcs.dmireset cancels the outstanding DMI transaction on our HW, this is not in compliance with the spec. Is this what you mean?

Yes, this is my assumption.

However, I would not focus on the write. The data, which is read back when the NOP is issued is unspecified (it is wrong to assume that on the NOP the written value would be read back).

The canceling of the DMI transaction is better presented by the following:
1. There was a write to `dmcontrol` (`dmcontrol.dmactive` is set):
```
Debug: 2183 73 riscv-013.c:420 dump_field(): 50b w 07ffffc1 @10 -> + 00000000 @16; 0i
```
2. The write was successful:
```
Debug: 2215 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 1i
```
3. `dmcontrol` is read.
```
Debug: 2225 74 riscv-013.c:420 dump_field(): 50b r 00000000 @10 -> + 00000000 @10; 1i
```
4. `dmi.busy` encountered, so `dtmcs.dmireset` is issued.
```
Debug: 2235 74 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> b 00000000 @10; 1i
Debug: 2236 74 riscv-013.c:489 increase_dmi_busy_delay(): [riscv.cpu1] dtmcs_idle=7, dmi_busy_delay=2, ac_busy_delay=0
Debug: 2246 75 riscv-013.c:478 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
```
5. The value read is zero (so `dmcontrol.dmactive == 0`).
```
Debug: 2256 75 riscv-013.c:420 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 2i
Debug: 2257 75 riscv-013.c:434 dump_field(): read: dmcontrol=0 {}
```

So, I think, issuing `dmireset` between (4) and (5) caused the read of `dmcontrol` to be aborted.
The thing is, I'm almost sure DM was actually active at the time, since this is the examination of the second hart on the same DM.

Nevertheless, I would like to emphasize,  this is only an assumption -- I do not know enough about your HW to be sure.

en-sc, 2024-03-07:
Somewhat related to #332

lz-bro, 2024-03-11:
@en-sc Thank you very much for your answer,  we have determined this is a hardware issue by using logic analysis.

closed 2024-03-11 by en-sc

zhajio1988, 2024-04-02:
@lz-bro 
Hello,I had the same problem as you, may I ask how you finally solved it? [1035](https://github.com/riscv-collab/riscv-openocd/issues/1035)

l1onog, 2024-04-26:
> @en-sc Thank you very much for your answer, we have determined this is a hardware issue by using logic analysis.

Hi, I'm experiencing the same problem as you, how exactly did you solve it? I will be very grateful for your answer!

TommyMurphyTM1234, 2024-04-26:
> > @en-sc Thank you very much for your answer, we have determined this is a hardware issue by using logic analysis.
> 
> Hi, I'm experiencing the same problem as you, how exactly did you solve it? I will be very grateful for your answer!

Did you read this? 

* https://github.com/riscv-collab/riscv-openocd/issues/1035

If it's the same issue then, as @en-sc says here, the hardware is faulty:

* https://github.com/riscv-collab/riscv-openocd/issues/1035#issuecomment-2027181134

You may be able to work around the problem to some extent as @en-sc suggests there but the only real fix is probably to fix the hardware bug. Maybe feasible if it's an FPGA or pre-tape-out ASIC but if it's already in silicon then probably not?

l1onog, 2024-04-26:
> > @zhajio1988:
> > Hello,I had the same problem as you, may I ask how you finally solved it? 1035
> 
> > @l1onog:
> > Hi, I'm experiencing the same problem as you, how exactly did you solve it? I will be very grateful for your answer!
> 
> Did you read this?
> 
> * [there is something error when openocd debug riscv multicore  #1035](https://github.com/riscv-collab/riscv-openocd/issues/1035)
> 
> If it's the same issue then, as @en-sc says here, the hardware is faulty:
> 
> * [there is something error when openocd debug riscv multicore  #1035 (comment)](https://github.com/riscv-collab/riscv-openocd/issues/1035#issuecomment-2027181134)
> 
> You may be able to work around the problem to some extent as @en-sc suggests there but the only real fix is probably to fix the hardware bug. Maybe feasible if it's an FPGA or pre-tape-out ASIC but if it's already in silicon then probably not?

Hi, I'm using a Zcu102 FPGA board, but I don't think it's a board issue because I'm experiencing the same problem after replacing the board. Also, I did attempts like updating the openocd version and so on didn't solve the problem... Finally, I think it's a chance problem because the same operation worked for me once .

en-sc, 2024-04-26:
> Hi, I'm using a Zcu102 FPGA board, but I don't think it's a board issue ...

The issue is not with the board, but with the implementation of the Debug Transport Module (firmware) you are programming the FPGA board with. 

> Finally, I think it's a chance problem because the same operation worked for me once .

The issue in #1035 and #1024 is timing related -- it will most probably be unstable if you are running on HW, not on a simulator.

If you think your problem is not the same, please file a new issue and attach a `-d3` log.

l1onog, 2024-04-26:
> > Hi, I'm using a Zcu102 FPGA board, but I don't think it's a board issue ...
> 
> The issue described in #1024 and #1035 is not with the board, but with the implementation of the Debug Transport Module (firmware) you are programming the FPGA board with.
> 
> > Finally, I think it's a chance problem because the same operation worked for me once .
> 
> The issue in #1035 and #1024 is timing related -- it will most probably be unstable if you are running on HW, not on a simulator.
> 
> If you think your problem is not the same, please file a new issue and attach a `-d3` log.

Hi, I have submitted a new question. https://github.com/riscv-collab/riscv-openocd/issues/1060