riscv-collab/riscv-openocd #1030: halted at 0x800001d8 due to debug interrupt

sx2025, 2024-03-14:
I use this cfg file to config my openocd .
//
adapter_khz     1000

interface cmsis-dap

transport select jtag

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

init
halt
//
my interface is cmsis-dap, my board has a core of E203 riscv .  
when i run the cfg file with openocd , it always comes out a message which is 'halted at 0x800001d8 due to debug interrupt'. I cannot do anything . I want to listen to your advice. Is it an error of cfg file or some of errors in  my riscv verilog file.

[image]

TommyMurphyTM1234, 2024-03-14:
Why do you think that this is an error/anomaly?
Your script has a `halt` command at the end so is going to put the target into debug halt after connecting.

What exactly do you mean by "I can't do anything"?

What happens if you run OpenOCD manually (outside of Eclipse) and then use the telnet interface to interact with it using the low level [OpenOCD commands](https://openocd.org/doc/html/General-Commands.html)?
For example:
```
# in one terminal
openocd -f <your-script>

# in another terminal
telnet localhost:4444
poll
reg
step
step

etc.
```
You should probably capture and attach a verbose `openocd -d3` log to help with analysis.

sx2025, 2024-03-14:
thank you for answering me. Because i am a rookie. I try to download my program into riscv . But it seems not working . And it also remind me 'halted at 0x800001d8 due to debug interrupt'. Whether i use halt or not ,it will still come out this infomation 'halted at 0x800001d8 due to debug interrupt'. I don't know why.
Here are the debug infomations.
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
`Open On-Chip Debugger 0.11.0+dev-01875-g13d7bc6cf (2021-09-25-22:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 3 options.c:64 configuration_output_handler(): debug_level: 3
User : 4 4 options.c:64 configuration_output_handler():
Debug: 5 5 options.c:245 add_default_dirs(): bindir=bin
Debug: 6 6 options.c:246 add_default_dirs(): pkgdatadir=
Debug: 7 7 options.c:247 add_default_dirs(): exepath=E:/hbird/NucleiStudio/toolchain/openocd/bin
Debug: 8 9 options.c:248 add_default_dirs(): bin2data=../
Debug: 9 11 configuration.c:44 add_script_search_dir(): adding C:/Users/pastgone/AppData/Roaming/OpenOCD
Debug: 10 13 configuration.c:44 add_script_search_dir(): adding E:/hbird/NucleiStudio/toolchain/openocd/bin/..//site
Debug: 11 16 configuration.c:44 add_script_search_dir(): adding E:/hbird/NucleiStudio/toolchain/openocd/bin/..//scripts
Debug: 12 19 command.c:201 script_debug(): command - ocd_find .\e203.cfg
Debug: 13 20 configuration.c:99 find_file(): found .\e203.cfg
Debug: 14 22 command.c:201 script_debug(): command - echo DEPRECATED! use 'adapter speed' not 'adapter_khz'
User : 15 24 command.c:729 jim_echo(): DEPRECATED! use 'adapter speed' not 'adapter_khz'
Debug: 16 26 command.c:201 script_debug(): command - adapter speed 1000
Debug: 17 27 command.c:201 script_debug(): command - adapter speed 1000
Debug: 18 29 core.c:1818 jtag_config_khz(): handle jtag khz
Debug: 19 30 core.c:1781 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 20 33 core.c:1781 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 21 35 command.c:201 script_debug(): command - reset_config srst_only
Debug: 22 36 command.c:201 script_debug(): command - echo DEPRECATED! use 'adapter srst pulse_width' not 'adapter_nsrst_assert_width'
User : 23 39 command.c:729 jim_echo(): DEPRECATED! use 'adapter srst pulse_width' not 'adapter_nsrst_assert_width'
Debug: 24 41 command.c:201 script_debug(): command - adapter srst pulse_width 100
Debug: 25 43 command.c:201 script_debug(): command - adapter srst pulse_width 100
Debug: 26 44 command.c:201 script_debug(): command - adapter srst pulse_width 100
Debug: 27 47 command.c:201 script_debug(): command - echo DEPRECATED! use 'adapter driver' not 'interface'
User : 28 49 command.c:729 jim_echo(): DEPRECATED! use 'adapter driver' not 'interface'
Debug: 29 51 command.c:201 script_debug(): command - adapter driver cmsis-dap
Debug: 30 52 command.c:201 script_debug(): command - adapter driver cmsis-dap
Debug: 31 54 command.c:201 script_debug(): command - transport select jtag
Debug: 32 55 command.c:201 script_debug(): command - transport select jtag
Debug: 33 57 command.c:201 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x1e200a6d
Debug: 34 59 command.c:201 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x1e200a6d
Debug: 35 61 tcl.c:566 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 36 64 tcl.c:590 jim_newtap_cmd(): Processing option: -irlen
Debug: 37 66 tcl.c:590 jim_newtap_cmd(): Processing option: -expected-id
Debug: 38 67 core.c:1484 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 39 69 command.c:201 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu
Debug: 40 71 command.c:201 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu
Debug: 41 74 target.c:2185 target_free_all_working_areas_restore(): freeing all working areas
Debug: 42 76 riscv.c:427 riscv_create_target(): riscv_create_target()
Debug: 43 78 command.c:201 script_debug(): command - riscv set_reset_timeout_sec 1
Debug: 44 80 command.c:201 script_debug(): command - riscv set_reset_timeout_sec 1
Debug: 45 81 command.c:201 script_debug(): command - init
Debug: 46 82 command.c:201 script_debug(): command - target init
Debug: 47 84 command.c:201 script_debug(): command - target init
Debug: 48 85 command.c:201 script_debug(): command - target names
Debug: 49 87 command.c:201 script_debug(): command - target names
Debug: 50 88 command.c:201 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 51 90 command.c:201 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 52 95 command.c:201 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 53 97 command.c:201 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 54 100 command.c:201 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 55 102 command.c:201 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 56 104 command.c:201 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 57 106 command.c:201 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
[... cut ...]
(3284) hpmcounter19h (/32)
(3285) hpmcounter20h (/32)
(3286) hpmcounter21h (/32)
(3287) hpmcounter22h (/32)
(3288) hpmcounter23h (/32)
(3289) hpmcounter24h (/32)
(3290) hpmcounter25h (/32)
(3291) hpmcounter26h (/32)
(3292) hpmcounter27h (/32)
(3293) hpmcounter28h (/32)
(3294) hpmcounter29h (/32)
(3295) hpmcounter30h (/32)
(3296) hpmcounter31h (/32)
(3667) hgeip (/32)
(3922) mvendorid (/32)
(3923) marchid (/32)
(3924) mimpid (/32)
(3925) mhartid (/32)
(4161) priv (/8)

> step
halted at 0x8000036a due to step
>

sx2025, 2024-03-14:
> Why do you think that this is an error/anomaly? Your script has a `halt` command at the end so is going to put the target into debug halt after connecting.
>
> What exactly do you mean by "I can't do [...]

> thank you for answering me. Because i am a rookie. I try to download my program into riscv . But it seems not working . And it also remind me 'halted at 0x800001d8 due to debug interrupt'. Whether i [...]

TommyMurphyTM1234, 2024-03-14:
> I try to download my program into riscv . But it seems not working . 

I can't see any evidence of any attempt to load the program in your OpenOCD log.

> And it also remind me 'halted at 0x800001d8 due to debug interrupt'. Whether i use halt or not ,it will still come out this infomation 'halted at 0x800001d8 due to debug interrupt'. 

Possibly because your debug launch configuration in Eclipse is telling the target to halt (at `main()`?) after connecting?
It's difficult to tell exactly what's going on when you are using so many things (Eclipse/plugins, GDB, OpenOCD) all at once.

It might be better to try to get things working using just OpenOCD and GDB from the command line first.

Aren't there existing resources that explain how to debug a Nuclei E203 RISC-V target using OpenOCD?
For example?

* https://www.bilibili.com/read/cv28484104/
* https://www.rvmcu.com/community-topic-id-1628.html
* https://www.riscv-mcu.com/community-topic-id-913.html

> I don't know why. Here are the debug infomations. 

It's better to capture the log to a file and then attach/upload it rather than embedding the huge amount of text in a post.

TommyMurphyTM1234, 2024-03-14:
To run OpenOCD and GDB standalone on the command line (outside of the scope of Eclipse) in order to simplify matters you should do something like this (this is off the top of my head so there may errors or omissions):
```
# In one terminal
openocd -f e203.cfg

# In another terminal
gdb myprogram
target remote localhost:3333

load
break main
continue
```

TommyMurphyTM1234, 2024-03-14:
> Open On-Chip Debugger
> 
> > load_image "E:\hbird\sdk\hel\Debug\hel.bin" 0x80000000
> > Core got an exception (0xffffffff) while writing to 0x80002108
> > keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (2381 ms). Workaround: increase "set remotetimeout" in GDB
> > (It may have failed between 0x80000000 and 0x80002107 as well, but we didn't check then.)
> 
> it seems my riscv has some problems，maybe about clk. i use a board with 27M clk, while the e203 use 50M clk. By using pll i have an clock with 16.2Mhz. but the original verilog files use 16Mhz clk. And i have also devide a clk which is about 32.768khz . maybe they cause the errors,what's your opinion?

If you are using an FPGA implementation of a RISC-V CPU/SoC then you need to ensure that it's fully functional before trying to debug against it. E.g. check all synthesis/P&R reports for any issues, check all clocks/resets, maybe double check functionality in simulation etc.

sx2025, 2024-03-14:
thank you ! i will examine my verilog files.

closed 2024-03-14 by sx2025