riscv-collab/riscv-openocd #1032: OpenOCD prints confusing status messages during resume command

aap-sc, 2024-03-21:
Example (reproducible on spike in a 4-hart configuration)
Recently, (after https://github.com/riscv-collab/riscv-openocd/commit/b503fdef02fa6fb76442d84e3a088fcf01cc6394) OpenOCD started to notify user about hart state updates. This causes confusion in some cases since some internal updates to the hart state should not be visible to the user as these are implementation details. For example situation like this:

```
> reset halt
JTAG tap: riscv.tap tap/device found: 0xdeadbeef ...
> resume
[riscv.cpu0] Found 4 triggers
riscv.cpu0 halted due to single-step.
[riscv.cpu1] Found 4 triggers
riscv.cpu1 halted due to single-step.
[riscv.cpu2] Found 4 triggers
riscv.cpu2 halted due to single-step.
[riscv.cpu3] Found 4 triggers
riscv.cpu3 halted due to single-step.
```

aap-sc, 2024-03-21:
MR that fixes the situation for 0.13: https://github.com/riscv-collab/riscv-openocd/pull/1031

NOTE: 0.11 is still affected (most likely).

aap-sc, 2024-05-01:
Now that https://github.com/riscv-collab/riscv-openocd/pull/1031 is merged in, could someone double-check if 0.11 is still affected? @TommyMurphyTM1234 , @JanMatCodasip any chance you could a simple test on your SiFive board?

The test is simple:

```
reset halt
resume
```

If during `resume` the message `riscv.cpu0 halted due to single-step` appears, then 0.11 is still affected.

TommyMurphyTM1234, 2024-05-01:
Hi @aap-sc - does this tell you what you need to know? 
With OpenOCD built from the latest sources at commit https://github.com/riscv-collab/riscv-openocd/commit/687f00c06055026331fa34250dcb05c9f652c361:
```
openocd -f board/sifive-hifive1.cfg
Open On-Chip Debugger 0.12.0+dev-03752-ge51f8695e (2024-04-26-21:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
riscv.cpu halted due to debug-request.
halted at 0x204002a4 due to debug interrupt
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
```
openocd -f board/sifive-hifive1.cfg -c "reset halt" -c "resume"
Open On-Chip Debugger 0.12.0+dev-03752-ge51f8695e (2024-04-26-21:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
riscv.cpu halted due to debug-request.
halted at 0x204002a4 due to debug interrupt
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
riscv.cpu halted due to breakpoint.
halted at 0x400 due to software breakpoint
riscv.cpu halted due to single-step.
halted at 0x404 due to step
riscv.cpu halted due to single-step.
halted at 0x404 due to step
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
I built OpenOCD from the latest sources but I'm not sure why the banner reports this `Open On-Chip Debugger 0.12.0+dev-03752-ge51f8695e (2024-04-26-21:57)`.

If I build with the previous commit (https://github.com/riscv-collab/riscv-openocd/commit/9563cd67e650426ceaac1d330fda84f7ad1b26a6) I still get the same version banner and the same log...
```
openocd -f board/sifive-hifive1.cfg -c "reset halt" -c "resume"
Open On-Chip Debugger 0.12.0+dev-03752-ge51f8695e (2024-04-26-21:57)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
riscv.cpu halted due to debug-request.
halted at 0x204002a4 due to debug interrupt
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
riscv.cpu halted due to breakpoint.
halted at 0x400 due to software breakpoint
riscv.cpu halted due to single-step.
halted at 0x404 due to step
riscv.cpu halted due to single-step.
halted at 0x404 due to step
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

aap-sc, 2024-05-01:
> Hi @aap-sc - does this tell you what you need to know?

Yep. This part:

```
riscv.cpu halted due to single-step.
halted at 0x404 due to step
riscv.cpu halted due to single-step.
halted at 0x404 due to step
```

So it looks like 0.11 is still affected. Will keep the issue open then