riscv-collab/riscv-openocd #1034: OpenOCD return error when using GDB to hit hardware breakpoints on hardware that doesn't support them.

Alan-19950616, 2024-03-26:
I think it's better to report a warning when this behaviour is detected than to report an error outright.

reporting an error will cause GDB to disconnect

https://github.com/riscv-collab/riscv-openocd/blob/cb2b3945143efda7196d14ae60c9877c289026a7/src/server/gdb_server.c#L1832

Alan-19950616, 2024-03-26:
Sorry, I got the wrong line number.

https://github.com/riscv-collab/riscv-openocd/blob/cb2b3945143efda7196d14ae60c9877c289026a7/src/server/gdb_server.c#L1859

en-sc, 2024-03-26:
AFAIU, `ERROR_NOT_IMPLEMENTED` should be returned here by `watchpoint_add()`.

Please, attach OpenOCD `-d3` log so it is possible to figure out why `watchpoint_add()` does not return `ERROR_NOT_IMPLEMENTED` in the case in question.

Alan-19950616, 2024-03-27:
OpenOCD log
```
Info : [riscv.cpu] Found 0 triggers
Error: Couldn't find an available hardware trigger.
Error: can't add breakpoint: resource not available
```
OpenOCD `-d3` log
```
84595 Error: 84593 31200 riscv.c:673 add_trigger(): Couldn't find an available hardware trigger.
84596 Error: 84594 31200 breakpoints.c:93 breakpoint_add_internal(): can't add breakpoint: resource not available
84597 Debug: 84595 31200 gdb_server.c:1495 gdb_error(): Reporting -308 to GDB as generic error
```

aap-sc, 2024-03-27:
ERROR_NOT_IMPLEMENTED is returned if OpenOCD does not implement something (not the target). Here the behavior is as expected. You try to set a HW breakpoint -> you have no available resources -> you get an error. GDB should not disconnect in such a case. Which GDB version are you using?

Alan-19950616, 2024-03-27:
GDB version `GNU gdb (GDB) 13.2.90.20230712-git`

I want to make sure that the error code is `#define ERROR_TARGET_RESOURCE_NOT_AVAILABLE	(-308)`  and not `#define ERROR_NOT_IMPLEMENTED			(-7)`.

aap-sc, 2024-03-27:
```
I want to make sure that the error code is 
```

Again, OpenOCD should return ERROR_TARGET_RESOURCE_NOT_AVAILABLE here since no HW resources available. ERROR_NOT_IMPLEMENTED is for the cases when something is not impleneted in OpenOCD itself.

Here is the debug session from gdb 13.2:

```
(gdb) watch *(int*)0x0
Hardware watchpoint 1: *(int*)0x0
(gdb) c
Continuing.
Warning:
Could not insert hardware watchpoint 1.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.

Command aborted.
```

GDB is not disconnected. it simply aborts the "continue" command since it can't set HW watchpoint.

Alan-19950616, 2024-03-27:
That's the problem I'm having, and I'm not expressing it clearly.

At this point it is not possible to use commands such as continue or single-step unless the hardware breakpoint is removed, is it possible to just warn without returning an error?

aap-sc, 2024-03-27:
{noformat}
At this point it is not possible to use commands such as continue or single-step unless the hardware breakpoint is removed, is it possible to just warn without returning an error?
{noformat}

But hat's the intended behavior for GDB. You've explicitly asked to set a HW watchpoint. GDB is unable to proceed unless you cancel this request.

TommyMurphyTM1234, 2024-03-27:
If you're trying to report and analyse an OpenOCD issue then it might make more sense to use OpenOCD standalone and its Telnet interface to interact with the program/target obviating the need for GDB and thus simplifying the diagnosis?

aap-sc, 2024-06-05:
I presume that the question is answered. The behavior you have is the expected one.

closed 2024-06-05 by aap-sc