riscv-collab/riscv-openocd #1035: thers is something error when openocd debug riscv multicore

zhajio1988, 2024-03-28:
Chip has 4 riscv cores, when openocd debug riscv multicore, riscv.cpu0 examine pass, but other cores encounter below errors:

Error: 3406 6331102 riscv-013.c:1921 examine(): [riscv.cpu1] Debug Module did not become active. dmcontrol=0x0

After  debug waveform, i find DMI is busy,  thiese are some clues in riscv-debug specification.
If a debugger sees this status, it needs to give the target more TCK edges between UpdateDR and Capture-DR. The simplest way to do that is to add extra transitions in Run-Test/Idle.

I want to know how to add add extra transitions in Run-Test/Idle when i use openocd?

en-sc, 2024-03-28:
> I want to know how to add add extra transitions in Run-Test/Idle when i use openocd?

Currently, there is no such command. The reason is: the necessary delay is learned during the session. Meaning, when `dmi.busy` is encountered, the delay used by OpenOCD is increased.

This makes it so OpenOCD can connect to target without knowing the number of RTI cycles needed.
There seems to be another issue, please, attach verbose OpenOCD log.

You can get it by adding `-d3` flag in the command line.
The file can be quite large. It can be redirected using `-l <filename>` flag.

TommyMurphyTM1234, 2024-03-28:
Maybe a shot in the dark, but...

Any chance your target is behaving the same as the hardware in this issue and incorrectly cancelling an outstanding DMI transaction when there is a delay and `dtmcs.dmireset` is written?

* https://github.com/riscv-collab/riscv-openocd/issues/1024#issuecomment-1973173367

> Seems to me, writing dtmcs.dmireset cancels the outstanding DMI transaction on your HW. This is explicitly prohibited by the spec ( see [6.1.4 DTM Control and Status (dtmcs, at 0x10) in RISC-V Debug Specification Version 1.0.0-rc1]).

As @en-sc suggests, a verbose `openocd -d3 ...` log will probably help to clarify exactly what's going on/wrong.

TommyMurphyTM1234, 2024-03-29:
> Thanks for your help. I added 'debug_level 4' to the openocd config file. Is this the same as openocd -d3?

`-d3` is sufficient. `-d4` gives too much additional info.

* https://openocd.org/doc/html/General-Commands.html

> level 3 adds debugging messages; and level 4 adds verbose low-level debug messages.

en-sc, 2024-03-29:
```
Debug: 3527 1254872 riscv-013.c:397 dump_field(): 50b w 00000001 @10 -> + 00000000 @10; 2i
Debug: 3528 1254872 riscv-013.c:405 dump_field(): write: dmcontrol=1 {dmactive=active}
Debug: 3540 1265929 riscv-013.c:397 dump_field(): 50b - 00000000 @10 -> b 00000001 @10; 2i
Debug: 3541 1265929 riscv-013.c:468 increase_dmi_busy_delay(): [riscv.cpu2] dtmcs_idle=7, dmi_busy_delay=3, ac_busy_delay=0
Debug: 3565 1275016 riscv-013.c:455 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x7d01
Debug: 3577 1287308 riscv-013.c:397 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 3i
Error: 3578 1287308 riscv-013.c:1921 examine(): [riscv.cpu2] Debug Module did not become active. dmcontrol=0x0
```
Seems like @TommyMurphyTM1234 was right. It looks like the same HW issue as in #1024.
@zhajio1988, please, take a look there for the explanation.
In short:
Writing `dtmcs.dmireset` cancels the request to activate debug module, which does not correspond to the spec ([6.1.4 DTM Control and Status (dtmcs, at 0x10) in RISC-V Debug Specification Version 1.0.0-rc1] (AFAIK, it was the same in RISCV Debug Spec 0.13)).

en-sc, 2024-03-29:
On a side note, @zhajio1988, AFAIU you are running `riscv-tests/debug` testsuite?

You can try to mitigate the issue by removing the call to `riscv reset_delays`
https://github.com/riscv-software-src/riscv-tests/blob/fd01e46314636666b5bed1f08465f734751a4cda/debug/testlib.py#L837

**This is not a fix, since the HW is faulty.**
But you most probably will be able to successfully examine the targets and figure out if there are other issues with the design / OpenOCD.

TommyMurphyTM1234, 2024-03-29:
> Seems like @TommyMurphyTM1234 was right. It looks like the same HW issue as in #1024. @zhajio1988, please, take a look there for the explanation. In short: Writing `dtmcs.dmireset` cancels the [...]

Maybe the same hardware is used in both cases?

> Attatch is log file, search 'Error:' keyword. [20240326-150458-C920Sim-MulticoreRegTest.log](https://github.com/riscv-collab/riscv-openocd/files/14801685/20240326-150458-C920Sim-MulticoreRegTest.log)

> ```
> Info : 260 133115 core.c:1133 jtag_examine_chain_display(): JTAG tap: riscv.cpu tap/device found: 0x10000b6f (mfg: 0x5b7 (T-HEAD Semiconductor Co Ltd), part: 0x0000, ver: 0x1)
> ```

The precise nature of the target hardware in #1024 doesn't seem to be clarified as far as I can see but maybe it's the same T-Head RISC-V implementation or another one that has the same debug bug?

* https://github.com/riscv-collab/riscv-openocd/issues/1024#issuecomment-1972984549

zhajio1988, 2024-04-01:
@TommyMurphyTM1234 @en-sc 
Thanks.
We are using a third party IP, AE suggested me to increase the tck between update_dr and idle, specifically to increase the number of tcks in the idle, so far this method solved the "Debug Module did not become active" problem I encountered. But looking at the description in [1024,](https://github.com/riscv-collab/riscv-openocd/issues/1024) it seems that this is a HW issue!

zhajio1988, 2024-04-01:
After debug i find below code snippet in RTL: The dmi doesn't implement errinfo function in dmtcs.

`
    assign capture_dmi_in_running = dmi_req_running & ctrl_idr_capture_dr & dtm_ir[DTM_IRREG_WIDTH-1:0] ==  DMI;

    always @ (posedge tclk or negedge trst_b) begin
        if (~trst_b)
            op_stat[1:0] <= 2'b00;
        else if (dmihardreset | escape_rst)
            op_stat[1:0] <= 2'b00;
        else if (dmireset)
            op_stat[1:0] <= 2'b00;
        else if (dmi_req_running & ctrl_idr_update_dr & dtm_ir[DTM_IRREG_WIDTH-1:0] ==  DMI & (^chain_idr_data[1:0]) |
                 capture_dmi_in_running)
            op_stat[1:0] <= 2'b11;
    end

    assign dmi_total[CHAIN_DW-1:0] = capture_dmi_in_running ? {address[DTM_ABITS-1:0], data[31:0], 2'b11} : {address[DTM_ABITS-1:0], data[31:0], op_stat[1:0]};
`
I think dmireset reset op_stat from busy to (idle or nop)? It's out of protocol. What do you guys think?

TommyMurphyTM1234, 2024-04-01:
> I find below log： Debug: 336454 55700304 jtag_vpi.c:287 jtag_vpi_state_move(): tap_set_state(RUN/IDLE) Debug: 336455 55700304 jtag_vpi.c:107 jtag_vpi_send_cmd(): sending JTAG VPI cmd: cmd=CMD_SCAN_CHAIN, length=1, nb_bits=3, buf_out=0x07 Debug: 336456 55701352 jtag_vpi.c:322 jtag_vpi_queue_tdi_xfer(): recvd JTAG VPI data: nb_bits=3, buf_in=0x00 Debug: 336457 55701352 core.c:962 default_interface_jtag_execute_queue(): JTAG DR SCAN to RUN/IDLE Debug: 336458 55701352 core.c:967 default_interface_jtag_execute_queue(): 50b out: 00004000000000 Debug: 336459 55701352 core.c:972 default_interface_jtag_execute_queue(): 50b in: 00004000000000 Debug: 336460 55701352 core.c:984 default_interface_jtag_execute_queue(): JTAG RUNTEST 3 cycles to RUN/IDLE Debug: 336461 55701352 riscv-013.c:397 dump_field(): 50b - 00000000 @10 -> + 00000000 @10; 3i Error: 336462 55701352 riscv-013.c:1921 examine(): [riscv.cpu1] Debug Module did not become active. dmcontrol=0x0 Why does it exit dmi_op_timeout after only adding up to 3 idle tcks?

This seems to be a different run to the one described in the earlier log?
As such you probably need to capture and attach the `openocd -d3 ...` log for this run.

TommyMurphyTM1234, 2024-04-01:
> But looking at the description in [1024,](https://github.com/riscv-collab/riscv-openocd/issues/1024) it seems that this is a HW issue! What do you guys think?

The problem in that issue is clearly explained as a hardware error in the implementation of the debug mechanism.

zhajio1988, 2024-04-01:
Hi, @TommyMurphyTM1234 
The log is same except the debug time,  you can find ' JTAG RUNTEST 3 cycles' too,  [20240326-150458-C920Sim-MulticoreRegTest.log](https://github.com/riscv-collab/riscv-openocd/files/14801685/20240326-150458-C920Sim-MulticoreRegTest.log)

zhajio1988, 2024-04-01:
@TommyMurphyTM1234 
I find op field in DMI has blow description：
0 (nop): Ignore data and address.
Don’t send anything over the DMI during
Update-DR. This operation should never result in
a busy or error response. The address and data
reported in the following Capture-DR are undefined.
This operation leaves the values in address and
data unspecified.
but [1024](https://github.com/riscv-collab/riscv-openocd/issues/1024)
say use DMI NOP is issued to check the result of the write， DMI NOP returned dmi.busy, so delays are increased and dtmcs.dmireset is issued. 
As you can see，riscv-debug 6.1.5 say NOP This operation should never result in a busy or error response.

zhajio1988, 2024-04-02:
Hi, Attached is the log for openocd -d3.
[20240402-145801-C920Sim-MulticoreRegTest.log](https://github.com/riscv-collab/riscv-openocd/files/14833789/20240402-145801-C920Sim-MulticoreRegTest.log)

zhajio1988, 2024-04-02:
Also i attach the dmi_dtm_dir RTL. Since github doesn't support uploading .v attachments, I've changed the extension to .txt.
[tdt_dmi_dtm_idr.txt](https://github.com/riscv-collab/riscv-openocd/files/14833849/tdt_dmi_dtm_idr.txt)

zhajio1988, 2024-05-06:
After run enough long time，this case can pass. I think we can close this tickets.

closed 2024-05-06 by TommyMurphyTM1234