riscv-collab/riscv-openocd #1038: Openocd: write to memory with dmi_write before init

hananyasegal1, 2024-03-31:
Hi All,

I want to ask if there is any option to write to memory for example, to de-assert the reset line of the core before init/examine completion?

Thanks,
Hananya

TommyMurphyTM1234, 2024-03-31:
Any use?

> I want to ask if there is any option to write to memory 

* https://openocd.org/doc/html/General-Commands.html#index-mwd

> to de-assert the reset line of the core before init/examine completion?

* https://openocd.org/doc/html/Reset-Configuration.html

hananyasegal1, 2024-04-02:
Hi,

I want to clarify my question: I want to write memory before target creation.

For example, do the following commands (in main script not in proc):
1. $_TARGETNAME write_memory 0x40040000 32 0x1
2.  init

Thanks,
Hananya

aap-sc, 2024-04-02:
@hananyasegal1 **short answer** - **no**. That's not how OpenOCD works.

There are few issues:
1. Before `init` you don't even have defined JTAG scan chain. So you can't initiate any transactions at all.
2. All these `write_memory` functions require a target to be present (and examined).

**Long answer** - **it depends** on what your are trying  to archive. If your design allows to initiate memory operations via a dedicated TAP controller,  you can do the following:

1. define JTAG scan chain without defining actual targets.
2. call init
3. use low-level `drscan` / `irscan` commands to do whatever you need with your taps.

hananyasegal1, 2024-04-02:
Hi Anatoly,

1. Is it possible to bypass the target creation and the init/examine stages?
2. In the case of using JTAG directly, can you explain how to do it with `drscan` / `irscan` commands?

Thanks,
Hananya

TommyMurphyTM1234, 2024-04-02:
> Hi Anatoly,
> 
> 1. Is it possible to bypass the target creation and the init/examine stages?
> 2. In the case of using JTAG directly, can you explain how to do it with `drscan` / `irscan` commands?
> 
> Thanks, Hananya

As far as I can see, most or all of this is covered in the OpenOCD documentation and nothing that you're asking about is really RISC-V specific.

* https://openocd.org/doc-release/html/index.html

For example:

* https://openocd.org/doc-release/html/Config-File-Guidelines.html
* https://openocd.org/doc-release/html/JTAG-Commands.html

aap-sc, 2024-06-05:
I assume that the original question was answered. Closing the ticket. Feel free to re-open if something is needed to be clarifed.

closed 2024-06-05 by aap-sc