riscv-collab/riscv-openocd #1042: examine(): Debug Module did not become active. dmcontrol=0x0

l1onog, 2024-04-15:
I implemented RISCV SoC on a Zcu102 fpga. I am trying to use openOCD and jtag-hs2 to download code to the board as instructed in the README. I have checked the physical connection and make sure it be correctly connected to the Zcu102. Does anyone know what could potentially cause this error and ways to fix it? @en-sc  I would really appreciate it.

here is the error I got:
```
Open On-Chip Debugger 0.10.0+dev-00830-ga88cc98a0-dirty (2024-04-09-21:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.unknown0 tap/device found: 0x10102001 (mfg: 0x000 (<invalid>), part: 0x0102, ver: 0x1)
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
Error: Debug Module did not become active. dmcontrol=0x0
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Here is what it shows when I run with -d3 flag
```
Open On-Chip Debugger 0.10.0+dev-00830-ga88cc98a0-dirty (2024-04-09-21:46)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 3 options.c:60 configuration_output_handler(): debug_level: 3
User : 14 3 options.c:60 configuration_output_handler(): 
Debug: 15 3 options.c:184 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 3 options.c:185 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 3 options.c:186 add_default_dirs(): exepath=/usr/local/bin
Debug: 18 3 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 19 3 configuration.c:42 add_script_search_dir(): adding /home/fpga/.openocd
Debug: 20 3 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 21 3 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 22 3 configuration.c:97 find_file(): found /opt/riscv/pulp/fpga/pulp-zcu102/openocd-zcu102-digilent-jtag-hs2.cfg
Debug: 23 3 command.c:143 script_debug(): command - adapter_khz adapter_khz 1000
Debug: 25 3 core.c:1636 jtag_config_khz(): handle jtag khz
Debug: 26 3 core.c:1599 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 27 3 core.c:1599 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 28 3 command.c:143 script_debug(): command - interface interface ftdi
Debug: 30 3 command.c:355 register_command_handler(): registering 'ftdi_device_desc'...
Debug: 31 3 command.c:355 register_command_handler(): registering 'ftdi_serial'...
Debug: 32 3 command.c:355 register_command_handler(): registering 'ftdi_channel'...
Debug: 33 3 command.c:355 register_command_handler(): registering 'ftdi_layout_init'...
Debug: 34 3 command.c:355 register_command_handler(): registering 'ftdi_layout_signal'...
Debug: 35 3 command.c:355 register_command_handler(): registering 'ftdi_set_signal'...
Debug: 36 3 command.c:355 register_command_handler(): registering 'ftdi_get_signal'...
Debug: 37 3 command.c:355 register_command_handler(): registering 'ftdi_vid_pid'...
Debug: 38 3 command.c:355 register_command_handler(): registering 'ftdi_tdo_sample_edge'...
Debug: 39 3 command.c:355 register_command_handler(): registering 'ftdi_oscan1_mode'...
Debug: 40 3 command.c:143 script_debug(): command - ftdi_vid_pid ftdi_vid_pid 0x0403 0x6014
Debug: 42 4 command.c:143 script_debug(): command - ftdi_channel ftdi_channel 0
Debug: 44 4 command.c:143 script_debug(): command - ftdi_layout_init ftdi_layout_init 0x00e8 0x60eb
Debug: 46 4 command.c:143 script_debug(): command - transport transport select
Info : 47 4 transport.c:285 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 48 4 command.c:355 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 49 4 command.c:355 register_command_handler(): registering 'jtag_rclk'...
Debug: 50 4 command.c:355 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 51 4 command.c:355 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 52 4 command.c:355 register_command_handler(): registering 'scan_chain'...
Debug: 53 4 command.c:355 register_command_handler(): registering 'jtag_reset'...
Debug: 54 4 command.c:355 register_command_handler(): registering 'runtest'...
Debug: 55 4 command.c:355 register_command_handler(): registering 'irscan'...
Debug: 56 4 command.c:355 register_command_handler(): registering 'verify_ircapture'...
Debug: 57 4 command.c:355 register_command_handler(): registering 'verify_jtag'...
Debug: 58 4 command.c:355 register_command_handler(): registering 'tms_sequence'...
Debug: 59 4 command.c:355 register_command_handler(): registering 'wait_srst_deassert'...
Debug: 60 4 command.c:355 register_command_handler(): registering 'jtag'...
Debug: 61 4 command.c:355 register_command_handler(): registering 'jtag'...
Debug: 62 4 command.c:355 register_command_handler(): registering 'jtag'...
Debug: 63 4 command.c:355 register_command_handler(): registering 'jtag'...
Debug: 64 4 command.c:355 register_command_handler(): registering 'jtag'...
[... 223 lines cut ...]
Debug: 298 21 riscv-013.c:452 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Debug: 299 21 riscv-013.c:397 scan(): 41b 7i r 00000000 @10 -> + 00000000 @10
Debug: 300 21 riscv-013.c:397 scan(): 41b 7i - 00000000 @10 -> b 00000000 @10
Debug: 301 21 riscv-013.c:461 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=8, ac_busy_delay=0
Debug: 302 21 riscv-013.c:452 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Debug: 303 21 riscv-013.c:397 scan(): 41b 8i - 00000000 @10 -> + 00000000 @10
Error: 304 21 riscv-013.c:1510 examine(): Debug Module did not become active. dmcontrol=0x0
Debug: 305 21 openocd.c:160 handle_init_command(): target examination failed
Debug: 306 21 command.c:143 script_debug(): command - flash flash init
Debug: 308 21 tcl.c:1222 handle_flash_init_command(): Initializing flash devices...
Debug: 309 21 command.c:143 script_debug(): command - nand nand init
Debug: 311 21 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 312 21 command.c:143 script_debug(): command - pld pld init
Debug: 314 21 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Debug: 315 21 gdb_server.c:3490 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 316 21 server.c:310 add_service(): Listening on port 3333 for gdb connections
Debug: 317 21 command.c:143 script_debug(): command - halt halt
Debug: 319 21 target.c:3070 handle_halt_command(): -
Error: 320 21 target.c:567 target_halt(): Target not examined yet
Debug: 321 21 command.c:629 run_command(): Command 'halt' failed with error code -4
User : 322 21 command.c:695 command_run_line(): 
Debug: 323 21 riscv.c:476 riscv_deinit_target(): riscv_deinit_target()
Debug: 324 21 riscv-013.c:1448 deinit_target(): riscv_deinit_target()
Debug: 325 23 target.c:1966 target_free_all_working_areas_restore(): freeing all working areas
```

TommyMurphyTM1234, 2024-04-15:
>```
>Open On-Chip Debugger 0.10.0+dev-00830-ga88cc98a0-dirty (2024-04-09-21:46)
>```

You're using a very out of date version of OpenOCD.
There have been a lot of changes/enhancements since that version that could impact behaviour in this context.
You should try again with a more up to date version - ideally a build of the latest `0.12.0+dev` using the sources in this repository.

l1onog, 2024-04-15:
When I use the latest version, I get the following results
```
Open On-Chip Debugger 0.12.0+dev-03744-g3991492cc (2024-04-15-21:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
riscv
  riscv authdata_read [index]
  riscv authdata_write [index] value
  riscv dm_read reg_address
  riscv dm_write reg_address value
  riscv dmi_read address
  riscv dmi_write address value
  riscv dump_sample_buf [base64]
  riscv etrigger set [vs] [vu] [m] [s] [u] <exception_codes>|clear
  riscv exec_progbuf instr1 [instr2 [... instr16]]
  riscv expose_csrs n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv expose_custom n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv hide_csrs {n0|n-m0}[,n1|n-m1]......
  riscv icount set [vs] [vu] [m] [s] [u] [pending] <count>|clear
  riscv info
  riscv itrigger set [vs] [vu] [nmi] [m] [s] [u] <mie_bits>|clear
  riscv memory_sample bucket address|clear [size=4]
  riscv repeat_read count address [size=4]
  riscv reset_delays [wait]
  riscv resume_order normal|reversed
  riscv set_bscan_tunnel_ir value
  riscv set_command_timeout_sec [sec]
  riscv set_ebreakm [on|off]
  riscv set_ebreaks [on|off]
  riscv set_ebreaku [on|off]
  riscv set_enable_trigger_feature [('eq'|'napot'|'ge_lt'|'all')
            ('wp'|'none')]
  riscv set_enable_virt2phys on|off
  riscv set_enable_virtual on|off
  riscv set_ir [idcode|dtmcs|dmi] value
  riscv set_maskisr ['off'|'steponly']
  riscv set_mem_access method1 [method2] [method3]
  riscv set_reset_timeout_sec [sec]
  riscv use_bscan_tunnel value [type]
riscv.cpu
  riscv.cpu arm
    riscv.cpu arm semihosting ['enable'|'disable']
    riscv.cpu arm semihosting_basedir [dir]
    riscv.cpu arm semihosting_cmdline arguments
    riscv.cpu arm semihosting_fileio ['enable'|'disable']
    riscv.cpu arm semihosting_read_user_param
    riscv.cpu arm semihosting_redirect (disable | tcp <port>
              ['debug'|'stdio'|'all'])
    riscv.cpu arm semihosting_resexit ['enable'|'disable']
  riscv.cpu arp_examine ['allow-defer']
  riscv.cpu arp_halt
  riscv.cpu arp_halt_gdb
  riscv.cpu arp_poll
  riscv.cpu arp_reset 'assert'|'deassert' halt
  riscv.cpu arp_waitstate statename timeoutmsecs
  riscv.cpu cget target_attribute
  riscv.cpu configure [target_attribute ...]
  riscv.cpu curstate
  riscv.cpu debug_reason
  riscv.cpu eventlist
  riscv.cpu examine_deferred
  riscv.cpu get_reg list
  riscv.cpu invoke-event event_name
[... 67 lines cut ...]

set _CHIPNAME riscv

jtag newtap $_CHIPNAME unknown0 -irlen 5 -expected-id 0x10102001
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511c3

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x3e0

gdb_report_data_abort enable
gdb_report_register_access_error enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

# prefer to use sba for system bus access
riscv set_prefer_sba on

# dump jtag chain
scan_chain

init
halt
echo "Ready for Remote Connections"
```

l1onog, 2024-04-15:
If the version of opoencd corresponds to the version of the debug script, can we ignore the specific version issue, and how can we solve the 'examine(): Debug Module did not become active. dmcontrol=0x0' problem?

TommyMurphyTM1234, 2024-04-15:
> If the version of opoencd corresponds to the version of the debug script, can we ignore the specific version issue

No. Because, as I said, there have been many changes, enhancements and fixes between 0.10.0 and the latest 0.12.0 development version and these make it effectively pointless to report and analyse issues with the old version. You need to get things working with the latest 0.12.0 and see if the problem persists with that. It's not that difficult to update your script and I've given you the documentation link to assist with this.

l1onog, 2024-04-15:
Thanks, 
I will try it as you pointed out.

TommyMurphyTM1234, 2024-04-22:
@l1onog can you clarify what happened please? 
Did it work ok with a build of the latest OpenOCD once your OpenOCD script was updated to work with that?