riscv-collab/riscv-openocd #1049: Random/sporadic failures running test suite interactively against Spike

TommyMurphyTM1234, 2024-04-21:
I'm not sure if this should be logged here or in [the test suite repo](https://github.com/riscv-software-src/riscv-tests/issues) - but I'll log it here for now. :-)

I have built the latest toolchain and OpenOCD and am running the basic smoke tests against Spike:

* https://github.com/riscv-software-src/riscv-tests/tree/master/debug#usage

but have been getting random/sporadic failures - e.g. the following from three separate attempts to run the smoke tests:

1. `MemorySampleMixed`: [20240421-125738-spike64-MemorySampleMixed.log](https://github.com/riscv-collab/riscv-openocd/files/15051888/20240421-125738-spike64-MemorySampleMixed.log)
2. `Registers`: [20240421-130859-spike64-Registers.log](https://github.com/riscv-collab/riscv-openocd/files/15051889/20240421-130859-spike64-Registers.log)
3. `ProgramSwWatchpoint`: [20240421-131950-spike64-ProgramSwWatchpoint.log](https://github.com/riscv-collab/riscv-openocd/files/15051892/20240421-131950-spike64-ProgramSwWatchpoint.log)

Does anybody know why this is happening?
In case it matters I'm doing this on a VirtualBox VM (host OS = Windows 10. guest OS = ZorinOS 17.1 Core) with what seems to be adequate resources (CPUs, memory, disk space etc.).

Maybe it's due to some local issue since the GitHib Action automated tests seem to be running OK?

* https://github.com/riscv-collab/riscv-openocd/actions/workflows/spike-openocd-tests.yml

TommyMurphyTM1234, 2024-04-21:
I just noticed that most if not all of the failures that I have encoutered seem to have this in common:
```
-------------------------[ /tmp/openocdxsztiom6.log ]-------------------------
+ REMOTE_BITBANG_HOST=localhost REMOTE_BITBANG_PORT=41607 WORK_AREA=0x1212340000 USE_FREERTOS=0 openocd --command 'gdb_port 0' --command 'tcl_port 0' --command 'telnet_port disabled' -f targets/RISC-V/spike-1.cfg
Open On-Chip Debugger 0.12.0+dev-03744-g3991492cc-dirty (2024-04-19-19:06)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:41607
Info : remote_bitbang driver initialized
Info : Note: The adapter "remote_bitbang" doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0xdeadbeef (mfg: 0x777 (Fabric of Truth Inc), part: 0xeadb, ver: 0xd)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0xdeadbeef (mfg: 0x777 (Fabric of Truth Inc), part: 0xeadb, ver: 0xd)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x4c0c02). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: [riscv.cpu] Examination failed
```
Does anybody know why this might happen sporadically/randomly?

aap-sc, 2024-04-21:
```
Does anybody know why this might happen sporadically/randomly?
```

Ah.. Sporadic failures. I've never encountered sporadic failures on these exact tests you've mentioned, but I have some experience with debugging sporadic failures in the test suite. For example, on the current TOT I periodically get sporadic failures of `UnavailableMultiTest` test (had not time to investigate it, yet)

A while ago there was an issue like this: https://github.com/riscv-software-src/riscv-tests/issues/520 which was fixed here: https://github.com/riscv-software-src/riscv-tests/pull/522 . Most likely there are still issues of these nature in the testsuite. 

I also remember that some tests depend on host machine perfomance. For example I know that **MemorySampleSingle** is affected, see call to:

```
self.check_incrementing_samples(raw_samples, addr, tolerance=0x500000)
```

Internally at Syntacore we just patch-out the source code of this test for it to work on our CI machine. Again, had no time fix this one, it was easier to patch-out and forget about the problem.

I have suspicion that you may encounter issue of similar nature. 

> I'm not sure if this should be logged here or in [the test suite repo](https://github.com/riscv-software-src/riscv-tests/issues) - but I'll log it here for now. :-)

I would recommend to file the issue there. That being said - both me and @en-sc has write permissions to riscv-tests repository with an expectation that we do reviews/fixes. So at least you've notified us about the issue.

Now, if you need help with debugging these it would be great if you could produce test execution logs that capture this sporadic failure. Sometimes careful analysis of these logs is enough to debug the issue.

If you choose to attach log files, please run tests as follows (below is just an example):

```
gdbserver.py \
  $PLATFORM_NAME \
 --isolate \
 --logs $OG_DIR \
 --gcc riscv64-unknown-elf-gcc \
 --gdb riscv64-unknown-elf-gdb \
  --sim_cmd spike --target-timeout 180 \
  --remotelogfile-enable \
  --server_cmd "$OPENCOD_BIN_PATH -d3'
```

Important arguments are: 
- ` --remotelogfile-enable` enables logging for gdb-remote commands sent to OpenOCD by GDB. This is relatively new option that was added here: https://github.com/riscv-software-src/riscv-tests/commit/b89570c845e21b0dbf94b93a9685542002eda957
-  `--server_cmd ` with `-d3` option. d3 enables debug-level printing during openocd execution.

Passing these extra arguments may simplify log analysis a little bit

TommyMurphyTM1234, 2024-04-21:
I edited the [`Makefile`](https://github.com/riscv-software-src/riscv-tests/blob/6b1d7372d951ed75811e0a09c0fe9e065c141c2d/debug/Makefile#L18-L24) to change this:
```
run.%:
	$(GDBSERVER_PY) \
		$(src_dir)/targets/RISC-V/$(word 2, $(subst ., ,$@)).py \
		$(word 3, $(subst ., ,$@)) \
		--isolate \
		--print-failures \
		$(if $(EXCLUDE_TESTS),--exclude-tests $(EXCLUDE_TESTS))
```
to this:
```
run.%:
	$(GDBSERVER_PY) \
		$(src_dir)/targets/RISC-V/$(word 2, $(subst ., ,$@)).py \
		$(word 3, $(subst ., ,$@)) \
		--isolate \
		--print-failures \
		$(if $(EXCLUDE_TESTS),--exclude-tests $(EXCLUDE_TESTS)) \
		--remotelogfile-enable --server_cmd "openocd -d3"
```
And then ran the smoke tests:
```
openocd -v
Open On-Chip Debugger 0.12.0+dev-03744-g3991492cc (2024-04-21-21:55)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

make

...

-------------------------------[ End of logs ]--------------------------------
Result: exception
Logfile: logs/20240421-222154-spike64_2-DownloadTest.log
Reproduce: ./gdbserver.py ./targets/RISC-V/spike64-2.py DownloadTest --hart 0
Time elapsed: 34.86s
::::::::::::::::::::::::::::[ ran 1 tests in 38s ]::::::::::::::::::::::::::::
1 tests returned exception
   DownloadTest > logs/20240421-222154-spike64_2-DownloadTest.log
make: *** [Makefile:19: run.spike64-2.DownloadTest] Error 1
```
Zipped `20240421-222154-spike64_2-DownloadTest.log`: [20240421-222154-spike64_2-DownloadTest.log.zip](https://github.com/riscv-collab/riscv-openocd/files/15054559/20240421-222154-spike64_2-DownloadTest.log.zip)

aap-sc, 2024-05-01:
@TommyMurphyTM1234 I've finally managed to look at the  20240421-222154-spike64_2-DownloadTest.log.zip log in a greater detail.

## Why the test fails.

I think I know the reason for this specific failure. 

1. Let's take a look at gdb remote logs:

```
224255 c compare-sections
224256 w $qCRC:1212340000,63c#68                                                                                           
224257 r $C53cfcfd2#d3
224258 w $qCRC:1212340640,40000#9a                                                                                          
224259 r <Timeout: 5 seconds>
224260 w -
224261 r <Timeout: 5 seconds> 
224262 w -
224263 r <Timeout: 5 seconds>
```

So, when gdb requested to calculate crc, this operation resulted int timeout.

2. Looking at the OpenOCD sections of the logs we can see that:

```
Debug: 108730 9011 gdb_server.c:424 gdb_log_incoming_packet(): [riscv.cpu0] {1} received packet: qCRC:1212340640,40000
...
Debug: 209697 29895 image.c:1300 image_calculate_checksum(): Calculating checksum done; checksum=0x9492e72e          
Debug: 209698 29895 gdb_server.c:441 gdb_log_outgoing_packet(): [riscv.cpu0] {1} sending packet: $C9492e72e#4e
```

Indeed, it took 20 seconds (29895 - 9011) to calculate the sum.

This is why we have test failure. 

## The root cause

As for the root cause - this is because spike is a rather slow simulator.  In addition to that halt mode in spike is implemented as a busy loop that constantly executes ROM code (at adress 0x1000 if I remember correctly). This means that whenever you have spike halted, the CPU usage by spike is still 100%.

## How to fix

gdbserver.py has an additional `--target-timeout` argument. Increasing the value passed as a parameter value should help. Here is what we us in our CI at Syntacore:

```
--target-timeout 180
```

TommyMurphyTM1234, 2024-05-01:
Thanks a lot @aap-sc - as ever, you've provided excellent analysis/insight. I've been meaning to revisit this issue but just didn't get a chance yet. When I do I'll definitely take cognizance of your analysis/advice above. 

Perhaps the test scripts need tweaking (e.g. to increase the relevant timeout(s)) as I would have expected the tests to run successfully out of the box barring "real" issues. But maybe it's difficult to ensure that this happens on all possible hosts. Mine is a VirtualBox VM running Zorin OS 17.1 Core on a 4th generation i5 host so obviously not the most powerful hardware available these days. 

Anyway, I'll try to revisit this issue again soon. Thanks again. 👍

aap-sc, 2024-05-01:
> Perhaps the test scripts need tweaking (e.g. to increase the relevant timeout(s)) as I would have expected the tests to run successfully out of the box barring "real" issues.

Actually, you are right! If the `--target-timeout` is not specified via command line test script takes timeout from the board definitions (the *.py files from here, https://github.com/riscv-software-src/riscv-tests/tree/master/debug/targets/RISC-V).

Now, looking at the value of `timeout_sec` (the default remote timeout for the target), we can see that some targets like spike64-2.py and spike32-2-hwthread.py have this value WAT too low. I think it is reasonable to increase this value for all the targets. I'll create a MR.

```
RISC-V/spike-multi.py:    timeout_sec = 30
RISC-V/spike-multi.py:    server_timeout_sec = 120
RISC-V/spike32-2-hwthread.py:    timeout_sec = 5
RISC-V/spike32-2.py:    timeout_sec = 30
RISC-V/spike32.py:    timeout_sec = 30
RISC-V/spike64-2-hwthread.py:    timeout_sec = 20
RISC-V/spike64-2-rtos.py:    timeout_sec = 60
RISC-V/spike64-2.py:    timeout_sec = 5
RISC-V/spike64.py:    timeout_sec = 30
SiFive/Freedom/U500Sim.py:    timeout_sec = 6000
```

TommyMurphyTM1234, 2024-05-01:
> Should be addressed by [riscv-software-src/riscv-tests#553](https://github.com/riscv-software-src/riscv-tests/pull/553)

WIthout this PR I ran the "smoke test" twice and each time it failed:

1. [20240501-140256-spike64_2-DownloadTest.log](https://github.com/riscv-collab/riscv-openocd/files/15177616/20240501-140256-spike64_2-DownloadTest.log)
2. [20240501-140858-spike64-EbreakTest.log](https://github.com/riscv-collab/riscv-openocd/files/15177618/20240501-140858-spike64-EbreakTest.log)

Then I applied the changes in [riscv-software-src/riscv-tests#553](https://github.com/riscv-software-src/riscv-tests/pull/553) and ran it again but it still failed. Seems to be a different failure to the timeout failures though?

3. [20240501-142710-spike64-MemorySampleSingle.log](https://github.com/riscv-collab/riscv-openocd/files/15177628/20240501-142710-spike64-MemorySampleSingle.log)

aap-sc, 2024-05-01:
>  ran it again but it still failed

The set of failed tests is different, right?

Regarding this one:

> 20240501-142710-spike64-MemorySampleSingle.log

Yes, this seems to be another issue. In fact if you read my comment here: https://github.com/riscv-collab/riscv-openocd/issues/1049#issuecomment-2068081650 :

```
I also remember that some tests depend on host machine perfomance. For example I know
that MemorySampleSingle is affected, see call to:

    self.check_incrementing_samples(raw_samples, addr, tolerance=0x500000)

Internally at Syntacore we just patch-out the source code of this test for it to work on our CI machine.
```

So yes, this test is known to be affected by some other issue that needs to be weeded out.

TommyMurphyTM1234, 2024-05-01:
> > ran it again but it still failed
> 
> The set of failed tests is different, right?

Just to clarify - when the test run fails in these cases it fails on one test which terminates the tests altogether.
It's not a case of an unexpected failure or exception on a particular test reported after everything has completed but allowing subsequent tests to proceed.
It's some sort of fatal failure that causes the test run to terminate prematurely.

> Regarding this one:
> 
> > 20240501-142710-spike64-MemorySampleSingle.log
> 
> Yes, this seems to be another issue. In fact if you read my comment here: [#1049 (comment)](https://github.com/riscv-collab/riscv-openocd/issues/1049#issuecomment-2068081650) :
> 
> ```
> I also remember that some tests depend on host machine perfomance. For example I know
> that MemorySampleSingle is affected, see call to:
> 
>     self.check_incrementing_samples(raw_samples, addr, tolerance=0x500000)
> 
> Internally at Syntacore we just patch-out the source code of this test for it to work on our CI machine.
> ```
> 
> So yes, this test is known to be affected by some other issue that needs to be weeded out. Another known test is "Unavailable*"-something :) one that still sporadically fails in our environment.

Ah - sorry, I forgot that you had posted about this.

For what it's worth I did a 4th run after a reset of the board and this time all tests completed.

So...

1. https://github.com/riscv-software-src/riscv-tests/pull/553 looks like it addresses the timeout failures
2. The `MemorySampleSingle` test failure is a known issue

TommyMurphyTM1234, 2024-05-01:
I did a 5th run and another failure:

* [20240501-185209-spike64-CustomRegisterTest.log](https://github.com/riscv-collab/riscv-openocd/files/15179648/20240501-185209-spike64-CustomRegisterTest.log)

Maybe this is the same class of failure as the aforementioned `MemorySampleSingle` test failure?

TommyMurphyTM1234, 2024-05-01:
Another run and another failure...

* 
[20240501-210945-spike64_2-RepeatReadTest.log](https://github.com/riscv-collab/riscv-openocd/files/15181257/20240501-210945-spike64_2-RepeatReadTest.log)

I need to go back through these failures to see if they are all of the same class - and maybe I should shift this issue/discussion to the `riscv-tests` repo assuming that these failures are not a result of any OpenOCD issue per se (if at all)?

aap-sc, 2024-05-01:
@TommyMurphyTM1234 well,

1.  As for MemorySampleSingle it would not hurt to report this issue to riscv-tests. There is a long-standing issue that needs some research
2. As for CustomRegisterTest.log / RepeatReadTest.log I'm planning to look into these failures (unless someone else won't do it sonner), though it may take some time.

TommyMurphyTM1234, 2024-05-01:
> @TommyMurphyTM1234 well,
> 
> 1. As for MemorySampleSingle it would not hurt to report this issue to riscv-tests. There is a long-standing issue that needs some research

Thanks - I will log that upstream in the `riscv-tests` repo so.

> 2. As for CustomRegisterTest.log / RepeatReadTest.log I'm planning to look into these failures (unless someone else won't do it sonner), though it may take some time.

Let me first (a) collect verbose OpenOCD/GDB remote protocol logs for such test failures and (b) see if I can have a look sooner. :-)

aap-sc, 2024-05-13:
@TommyMurphyTM1234 

1. RepeatReadTest looks like an issue with the test framework and may be related to https://github.com/riscv-software-src/riscv-tests/pull/491 .

The issue there is with:

```
     values.append(int(v, 16))                                                                                                                              
 ValueError: invalid literal for int() with base 16: 'keep_alive()'
```

This makes me think that there is not enough sanitization of the command output. 
 
2. As for CustomRegisterTest - this looks some other issue.

For now I suggest to:

- I'm planning to try and fix MemorySampleSingle.
- Wait till we have a clarification from @en-sc regarding https://github.com/riscv-software-src/riscv-tests/pull/491
- Wait till https://review.openocd.org/c/openocd/+/8227 is backported to riscv-openocd 

then we should re-run these tests with verbose logs.

TommyMurphyTM1234, 2024-05-13:
Thanks @aap-sc.

> * Wait till https://review.openocd.org/c/openocd/+/8227 is backported to riscv-openocd
> 
> then we should re-run these tests with verbose logs.

Is there any point in me manually applying that patch and running the tests to see what happens?

aap-sc, 2024-05-14:
@TommyMurphyTM1234 

> Is there any point in me manually applying that patch and running the tests to see what happens?

This looks like way too much hassle for my taste. We'll get this changes anyway in due time. Changes from upstream are merged fairly regularly these days, so it should not be too long.

TommyMurphyTM1234, 2024-05-14:
> @TommyMurphyTM1234
>
>
> This looks like way too much hassle for my taste. We'll get this changes anyway in due time. Changes from upstream are merged fairly regularly these days, so it should not be [...]

Ok, thanks. That's fine so. 

BTW, running the tests repeatedly again I sometimes get other tests failing (and terminating the test run) and not just the three mentioned in the first post in this issue. I'm not sure if I should keep posting logs for such failures?

aap-sc, 2024-05-14:
I suggest to wait till the situation with MemorySampleSingle is resolved.

I've manged to reproduce the problem in our lab environment with MemorySampleSingle and MemorySampleMixed .  In addition, I plan to merge the fixed for some sporadic failures observed in other tests (UnavailableMultiTest) for example. 

If we keep constantly to update this ticket it may be hard to follow the status of the current situation.

TommyMurphyTM1234, 2024-05-14:
> I suggest to wait till the situation with MemorySampleSingle is resolved.

Ok, thanks - I'll do that so. 

> I've manged to reproduce the problem in our lab environment with MemorySampleSingle and MemorySampleMixed . In addition, I plan to merge the fixed for some sporadic failures observed in other tests (UnavailableMultiTest for example).

That's great. 👍 

> If we keep constantly to update this ticket it may be hard to follow the status of the current situation.

I understand and agree. 

Thanks again @aap-sc.

aap-sc, 2024-05-15:
@TommyMurphyTM1234 

> I've manged to reproduce the problem in our lab environment with MemorySampleSingle and MemorySampleMixed . In addition, I plan to merge the fixed for some sporadic failures observed in other tests (UnavailableMultiTest for example).

The respected fixes are:

- https://github.com/riscv-software-src/riscv-tests/pull/556
- https://github.com/riscv-software-src/riscv-tests/pull/555

Once these fixes are  applied I do not observe sporadic failures in our lab environment anymore (just FYI)

Now, I suggest to wait till these are merged in and till https://review.openocd.org/c/openocd/+/8227 is available in riscv-openocd (as discussed above).

aap-sc, 2024-06-05:
@TommyMurphyTM1234 all the changes are merged to riscv-openocd (including https://review.openocd.org/c/openocd/+/8227).  We can try to re-run the testsuite on your machine to weed out remaining bugs. Please, let me know if you still have the will and desire to debug this.

TommyMurphyTM1234, 2024-06-05:
> @TommyMurphyTM1234 all the changes are merged to riscv-openocd (including https://review.openocd.org/c/openocd/+/8227). The number of sporadic failures should decrease now. We can try to re-run the testsuite on your machine to weed out remaining bugs. Please, let me know if you still have the will and desire to debug this.

Thanks a lot @aap-sc. Yes, I'm happy and able to do further testing on this. I might get to it later today but, if not, then tomorrow.

TommyMurphyTM1234, 2024-06-05:
> Thanks a lot @aap-sc. Yes, I'm happy and able to do further testing on this. I might get to it later today but, if not, then tomorrow.

@aap-sc - any idea about this? Maybe those "PRNG" related lines were supposed to be removed before the commit/PR was merged?

* https://github.com/riscv-software-src/riscv-tests/pull/531#issuecomment-2151026889

aap-sc, 2024-06-05:
@TommyMurphyTM1234 no these messages are intentional. The intention is to report the selected seed to the user, so we could reproduce the run. This is important because even the compiled binaries sometimes depend on this seed. So it's quite confusing to debug issues and get different binaries every time.

TommyMurphyTM1234, 2024-06-06:
> > Thanks a lot @aap-sc. Yes, I'm happy and able to do further testing on this. I might get to it later today but, if not, then tomorrow.
> 
> @aap-sc - any idea about this? Maybe those "PRNG seed ..." related lines were supposed to be removed before the commit/PR was merged?
> 
> * [ improvements to debug tests infrastructure to help with triaging process riscv-software-src/riscv-tests#531 (comment)](https://github.com/riscv-software-src/riscv-tests/pull/531#issuecomment-2151026889)

* https://github.com/riscv-software-src/riscv-tests/pull/561

TommyMurphyTM1234, 2024-06-06:
I manually did a run of all tests against the four targets that the `Makefile` default rule executes and got these results:
```
./gdbserver.py ./targets/RISC-V/spike64.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike64.log

Using $misa from hart definition: 0x8000000000141125
[CeaseStepiTest] Starting > logs/20240606-195706-spike64-CeaseStepiTest.log
[CeaseStepiTest] not_applicable in 0.00s
[CheckMisa] Starting > logs/20240606-195706-spike64-CheckMisa.log
[CheckMisa] pass in 1.49s
[CustomRegisterTest] Starting > logs/20240606-195708-spike64-CustomRegisterTest.log
[CustomRegisterTest] pass in 5.23s
[DebugBreakpoint] Starting > logs/20240606-195713-spike64-DebugBreakpoint.log
[DebugBreakpoint] pass in 5.46s
[DebugChangeString] Starting > logs/20240606-195718-spike64-DebugChangeString.log
[DebugChangeString] pass in 5.53s
[DebugCompareSections] Starting > logs/20240606-195724-spike64-DebugCompareSections.log
[DebugCompareSections] pass in 3.83s
[DebugExit] Starting > logs/20240606-195728-spike64-DebugExit.log
[DebugExit] pass in 4.08s
[DebugFunctionCall] Starting > logs/20240606-195732-spike64-DebugFunctionCall.log
[DebugFunctionCall] pass in 6.12s
[DebugSymbols] Starting > logs/20240606-195738-spike64-DebugSymbols.log
[DebugSymbols] pass in 4.53s
[DebugTurbostep] Starting > logs/20240606-195742-spike64-DebugTurbostep.log
[DebugTurbostep] pass in 7.13s
[DisconnectTest] Starting > logs/20240606-195749-spike64-DisconnectTest.log
[DisconnectTest] pass in 4.10s
[DownloadTest] Starting > logs/20240606-195754-spike64-DownloadTest.log
[DownloadTest] pass in 17.56s
[EbreakTest] Starting > logs/20240606-195811-spike64-EbreakTest.log
[EbreakTest] pass in 4.79s
[EtriggerTest] Starting > logs/20240606-195816-spike64-EtriggerTest.log
[EtriggerTest] pass in 4.76s
[FreeRtosTest] Starting > logs/20240606-195821-spike64-FreeRtosTest.log
[FreeRtosTest] pass in 11.37s
[Hwbp1] Starting > logs/20240606-195832-spike64-Hwbp1.log
[Hwbp1] pass in 5.44s
[Hwbp2] Starting > logs/20240606-195838-spike64-Hwbp2.log
[Hwbp2] pass in 6.13s
[HwbpManual] Starting > logs/20240606-195844-spike64-HwbpManual.log
[HwbpManual] pass in 7.37s
[IcountTest] Starting > logs/20240606-195851-spike64-IcountTest.log
[IcountTest] pass in 4.48s
[InfoTest] Starting > logs/20240606-195855-spike64-InfoTest.log
[InfoTest] pass in 1.58s
[InstantChangePc] Starting > logs/20240606-195857-spike64-InstantChangePc.log
[InstantChangePc] pass in 2.81s
[InstantHaltTest] Starting > logs/20240606-195900-spike64-InstantHaltTest.log
[InstantHaltTest] pass in 2.08s
[InterruptTest] Starting > logs/20240606-195902-spike64-InterruptTest.log
[InterruptTest] pass in 6.44s
[ItriggerTest] Starting > logs/20240606-195908-spike64-ItriggerTest.log
[ItriggerTest] pass in 4.18s
[JumpHbreak] Starting > logs/20240606-195913-spike64-JumpHbreak.log
[JumpHbreak] pass in 3.95s
[MemTest16] Starting > logs/20240606-195917-spike64-MemTest16.log
[MemTest16] pass in 1.94s
[MemTest32] Starting > logs/20240606-195918-spike64-MemTest32.log
[MemTest32] pass in 1.93s
[MemTest64] Starting > logs/20240606-195920-spike64-MemTest64.log
[MemTest64] pass in 1.94s
[MemTest8] Starting > logs/20240606-195922-spike64-MemTest8.log
[MemTest8] pass in 1.96s
[MemTestBlock0] Starting > logs/20240606-195924-spike64-MemTestBlock0.log
[MemTestBlock0] pass in 3.84s
[MemTestBlock1] Starting > logs/20240606-195928-spike64-MemTestBlock1.log
[MemTestBlock1] pass in 3.74s
[MemTestBlock2] Starting > logs/20240606-195932-spike64-MemTestBlock2.log
[MemTestBlock2] pass in 3.73s
[MemTestBlockReadInvalid] Starting > logs/20240606-195936-spike64-MemTestBlockReadInvalid.log
[... 637 lines cut ...]
    self.close()
  File "/usr/lib/python3.10/tempfile.py", line 585, in close
    unlink(self.name)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/gdb@45513-fb64jz0f.log'
Exception ignored in: <function _TemporaryFileCloser.__del__ at 0x7fbf77a50670>
Traceback (most recent call last):
  File "/usr/lib/python3.10/tempfile.py", line 589, in __del__
    self.close()
  File "/usr/lib/python3.10/tempfile.py", line 585, in close
    unlink(self.name)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/remote.gdb@45513-airs7m3i.log'
[UnavailableRunTest] pass in 5.14s
[UserInterrupt] Starting > logs/20240606-204947-spike32_2-UserInterrupt.log
[UserInterrupt] pass in 8.00s
[VectorTest] Starting > logs/20240606-204955-spike32_2-VectorTest.log
[VectorTest] not_applicable in 0.00s
[WriteCsrs] Starting > logs/20240606-204955-spike32_2-WriteCsrs.log
[WriteCsrs] pass in 5.85s
[WriteGprs] Starting > logs/20240606-205001-spike32_2-WriteGprs.log
[WriteGprs] pass in 11.66s
:::::::::::::::::::::::::::[ ran 73 tests in 583s ]:::::::::::::::::::::::::::
10 tests returned not_applicable
63 tests returned pass
```
Note that previous tests were performed on a VirtualBox VM running Zorin OS Core 17.1 as a guest OS on a Windows 10 host whereas these were performed on WSL2/Ubuntu 22.04 on the same Windows 10 host. I'm not sure if the probable effective lower spec resources of the former may have given different results (and more sporadic failures?) than the latter?

aap-sc, 2024-06-06:
@TommyMurphyTM1234 so these ones do not show any signs of sporadic failues (for now). Do you have the capability to run this experiment on VirtualBox  ?

TommyMurphyTM1234, 2024-06-06:
> @TommyMurphyTM1234 so these ones do not show any signs of sporadic failues (for now). Do you have the capability to run this experiment on VirtualBox ?

Yes - I'll revert to my VirtualBox setup and run the tests again.

I'm not sure why there are a few Python exceptions in the logs, if they are sporadic or reproducible (I suspect the former), and if they are anything to be concerned about?

aap-sc, 2024-06-06:
> I'm not sure why there are a few Python exceptions in the logs, if they are sporadic or reproducible (I suspect the former), and if they are anything to be concerned about?

They are sporadic. They are somewhat reproducible. The reason for this warnings is described in greater detail here: https://github.com/riscv-software-src/riscv-tests/pull/555 .

*We* should not be concerned about these warning. If someone should be concern - it's Python developers (I mean literally the developers of CPython interpreter) who don't follow their own guidelines and allow exceptions to propagate from finalizers of  TemporaryFile objects. One day I may file a bug against that - but right now I'm still hold the grudge against them (since I had to debug this).

TommyMurphyTM1234, 2024-06-06:
> > I'm not sure why there are a few Python exceptions in the logs, if they are sporadic or reproducible (I suspect the former), and if they are anything to be concerned about?
> 
> They are sporadic. They are somewhat reproducible. The reason for this warnings is described in greater detail here: [riscv-software-src/riscv-tests#555](https://github.com/riscv-software-src/riscv-tests/pull/555) .
> 
> **We** should not be concerned about these warnings. If someone should be concerned - its Python developers (I mean literally the developers of CPython interpreter) who don't follow their own guidelines and allow exceptions to propagate from finalizers of TemporaryFile objects. One day I may file a bug against that - but right now I'm still hold the grudge against them (since I had to debug this).

@aap-sc - thanks a lot for that and for the link to the related/explanatory PR which you had previously tagged me on but which I must've missed. 👍

TommyMurphyTM1234, 2024-06-08:
Hi @aap-sc, I've reverted to my original VirtualBox/Zorin OS 17.1 setup and have run the tests (just once) there with the latest of everyting (`riscv-openocd`, `riscv-gnu-toolchain` bare-metal GCC and Spike, `riscv-tests`) and got these results:
```
./gdbserver.py ./targets/RISC-V/spike64.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike64.log

Using $misa from hart definition: 0x8000000000141125
[CeaseStepiTest] Starting > logs/20240607-142956-spike64-CeaseStepiTest.log
[CeaseStepiTest] not_applicable in 0.00s
[CheckMisa] Starting > logs/20240607-142956-spike64-CheckMisa.log
[CheckMisa] pass in 2.88s
[CustomRegisterTest] Starting > logs/20240607-142959-spike64-CustomRegisterTest.log
[CustomRegisterTest] pass in 8.89s
[DebugBreakpoint] Starting > logs/20240607-143008-spike64-DebugBreakpoint.log
[DebugBreakpoint] pass in 47.33s
[DebugChangeString] Starting > logs/20240607-143055-spike64-DebugChangeString.log
[DebugChangeString] pass in 8.51s
[DebugCompareSections] Starting > logs/20240607-143104-spike64-DebugCompareSections.log
[DebugCompareSections] pass in 6.23s
[DebugExit] Starting > logs/20240607-143110-spike64-DebugExit.log
[DebugExit] pass in 6.27s
[DebugFunctionCall] Starting > logs/20240607-143116-spike64-DebugFunctionCall.log
[DebugFunctionCall] pass in 10.60s
[DebugSymbols] Starting > logs/20240607-143127-spike64-DebugSymbols.log
[DebugSymbols] pass in 6.69s
[DebugTurbostep] Starting > logs/20240607-143134-spike64-DebugTurbostep.log
[DebugTurbostep] pass in 10.71s
[DisconnectTest] Starting > logs/20240607-143144-spike64-DisconnectTest.log
[DisconnectTest] pass in 7.37s
[DownloadTest] Starting > logs/20240607-143152-spike64-DownloadTest.log
[DownloadTest] pass in 19.81s
[EbreakTest] Starting > logs/20240607-143212-spike64-EbreakTest.log
[EbreakTest] pass in 6.32s
[EtriggerTest] Starting > logs/20240607-143218-spike64-EtriggerTest.log
[EtriggerTest] pass in 7.28s
[FreeRtosTest] Starting > logs/20240607-143225-spike64-FreeRtosTest.log
[FreeRtosTest] pass in 15.59s
[Hwbp1] Starting > logs/20240607-143241-spike64-Hwbp1.log
[Hwbp1] pass in 57.63s
[Hwbp2] Starting > logs/20240607-143338-spike64-Hwbp2.log
[Hwbp2] pass in 9.20s
[HwbpManual] Starting > logs/20240607-143348-spike64-HwbpManual.log
[HwbpManual] pass in 27.58s
[IcountTest] Starting > logs/20240607-143415-spike64-IcountTest.log
[IcountTest] pass in 6.17s
[InfoTest] Starting > logs/20240607-143421-spike64-InfoTest.log
[InfoTest] pass in 2.02s
[InstantChangePc] Starting > logs/20240607-143423-spike64-InstantChangePc.log
[InstantChangePc] pass in 3.65s
[InstantHaltTest] Starting > logs/20240607-143427-spike64-InstantHaltTest.log
[InstantHaltTest] pass in 2.63s
[InterruptTest] Starting > logs/20240607-143430-spike64-InterruptTest.log
[InterruptTest] pass in 8.23s
[ItriggerTest] Starting > logs/20240607-143438-spike64-ItriggerTest.log
[ItriggerTest] pass in 29.75s
[JumpHbreak] Starting > logs/20240607-143508-spike64-JumpHbreak.log
[JumpHbreak] pass in 9.29s
[MemTest16] Starting > logs/20240607-143517-spike64-MemTest16.log
[MemTest16] pass in 2.32s
[MemTest32] Starting > logs/20240607-143519-spike64-MemTest32.log
[MemTest32] pass in 2.28s
[MemTest64] Starting > logs/20240607-143521-spike64-MemTest64.log
[MemTest64] pass in 2.40s
[MemTest8] Starting > logs/20240607-143524-spike64-MemTest8.log
[MemTest8] pass in 2.32s
[MemTestBlock0] Starting > logs/20240607-143526-spike64-MemTestBlock0.log
[MemTestBlock0] pass in 4.52s
[MemTestBlock1] Starting > logs/20240607-143531-spike64-MemTestBlock1.log
[MemTestBlock1] pass in 64.45s
[MemTestBlock2] Starting > logs/20240607-143635-spike64-MemTestBlock2.log
[MemTestBlock2] pass in 29.44s
[MemTestBlockReadInvalid] Starting > logs/20240607-143705-spike64-MemTestBlockReadInvalid.log
[... 693 lines cut ...]
Traceback (most recent call last):
  File "/usr/lib/python3.10/tempfile.py", line 589, in __del__
    self.close()
  File "/usr/lib/python3.10/tempfile.py", line 585, in close
    unlink(self.name)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/gdb@42785-rvv3g3ag.log'
Exception ignored in: <function _TemporaryFileCloser.__del__ at 0x73dbfd94d630>
Traceback (most recent call last):
  File "/usr/lib/python3.10/tempfile.py", line 589, in __del__
    self.close()
  File "/usr/lib/python3.10/tempfile.py", line 585, in close
    unlink(self.name)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/remote.gdb@42785-xte4lyc1.log'
[UserInterrupt] pass in 37.23s
[VectorTest] Starting > logs/20240607-192907-spike32_2-VectorTest.log
[VectorTest] not_applicable in 0.00s
[WriteCsrs] Starting > logs/20240607-192907-spike32_2-WriteCsrs.log
[WriteCsrs] pass in 7.58s
[WriteGprs] Starting > logs/20240607-192915-spike32_2-WriteGprs.log
[WriteGprs] pass in 13.06s
::::::::::::::::::::::::::[ ran 73 tests in 1328s ]:::::::::::::::::::::::::::
10 tests returned not_applicable
63 tests returned pass
```
I will run them a few more times just to see if there are any sporadic failures.

TommyMurphyTM1234, 2024-06-08:
Second run and one failure this time:
```
./gdbserver.py ./targets/RISC-V/spike32-2.py --isolate --print-failures --remotelogfile-enable --server_cmd "openocd -d3" 2>&1 | tee test-spike32-2.log

...

::::::::::::::::::::::::::[ ran 73 tests in 1933s ]:::::::::::::::::::::::::::
10 tests returned not_applicable
62 tests returned pass
1 tests returned fail
   MemorySampleMixed > logs/20240608-155212-spike32_2-MemorySampleMixed.log
```
* `20240608-155212-spike32_2-MemorySampleMixed.log`: [20240608-155212-spike32_2-MemorySampleMixed.log](https://github.com/user-attachments/files/15748609/20240608-155212-spike32_2-MemorySampleMixed.log)

aap-sc, 2024-06-08:
Regarding the MemorySampleMixed. I see the issue. I'll dig to find the root cause. Hopefully it won't take long (no promises, though).

@TommyMurphyTM1234 if you don't mind - can I ask you to run these 10-20 times more and report if there are any failures?

TommyMurphyTM1234, 2024-06-08:
> Regarding the MemorySampleMixed. I see the issue. I'll dig to find the root cause. Hopefully it won't take long (no promises, though).

Great - thanks @aap-sc.

> @TommyMurphyTM1234 if you don't mind - can I ask you to run these 10-20 times more and report if there are more failures?

Will do - I'll do 10 anyway as it take quite a while to run each set of four target tests on the virtual machine.

TommyMurphyTM1234, 2024-06-09:
Here are the results of 10 further runs:

1. No failures
2. [`20240608-195858-spike64-MemorySampleSingle.log`](https://github.com/user-attachments/files/15753214/20240608-195858-spike64-MemorySampleSingle.log)
3. No failures
4. [`20240608-232734-spike64-HwbpManual.log`](https://github.com/user-attachments/files/15753215/20240608-232734-spike64-HwbpManual.log)
5. No failures
6. [`20240609-124049-spike64-WriteGprs.log`](https://github.com/user-attachments/files/15753217/20240609-124049-spike64-WriteGprs.log)
7. [`20240609-142430-spike64-RepeatReadTest.log`](https://github.com/user-attachments/files/15753219/20240609-142430-spike64-RepeatReadTest.log)
8. [`20240609-153957-spike64-JumpHbreak.log`](https://github.com/user-attachments/files/15753221/20240609-153957-spike64-JumpHbreak.log) and [`20240609-162951-spike32_2-MemorySampleMixed.log`](https://github.com/user-attachments/files/15753222/20240609-162951-spike32_2-MemorySampleMixed.log)
9. No failures
10. No failures

aap-sc, 2024-06-09:
ghrrrr. That's way more failures than I expected :(

TommyMurphyTM1234, 2024-06-09:
I'll add further results here:

11. [`20240609-204109-spike64-TriggerExecuteInstant.log`](https://github.com/user-attachments/files/15754186/20240609-204109-spike64-TriggerExecuteInstant.log)
12. TBD

TommyMurphyTM1234, 2024-06-11:
That's the result of 20 runs of these tests - I won't do any more until requested.

aap-sc, 2024-06-11:
> That's the results of 20 runs of these tests - I won't do any more until requested.

@TommyMurphyTM1234 thanks a lot!

I hope it's not that big of a deal to run these... Sorry if it took more effort/time than one could expect :(. 

It will take some time to fix. At least some of these failures are easy to understand (not necessary easy to fix, though).  For example: RepeatReadTest tests fail because test infrastructure expects only output of the underlying monitor command to be present in "monitor <command>" statement. However for whatever abysmall reason OpenOCD adds target warnings to the resulting output (missing keep-alive messages, for example). This results in "invalid literal for int" exceptions to cause test failure. There is no consensus on how to fix this (@en-sc and me have a disagreement at the moment) - hopefully, this should be resolved soon.

Other cases are more tricky and will requite more effort to debug.

TommyMurphyTM1234, 2024-06-11:
> @TommyMurphyTM1234 thanks a lot!
> 
> I hope it's not that big of a deal to run these... Sorry if it took more effort/time than one could expect :(.

No problem. It was a bit tedious on the hardware that I had available for this but not a big deal - just slow! 🙂

> It will take some time to fix. At least some of these failures are easy to understand (not necessary easy to fix, though). For example: RepeatReadTest tests fail because test infrastructure expects only output of the underlying monitor command to be present in "monitor " statement. However for whatever abysmall reason OpenOCD adds target warnings to the resulting output (missing keep-alive messages, for example). This results in "invalid literal for int" exceptions to cause test failure. There is no consensus on how to fix this (@en-sc and me have a disagreement at the moment) - hopefully, this should be resolved soon

That's interesting, thanks. Perhaps some changes to upstream OpenOCD functionality may be required in order to address some of these failures/exceptions? 

> Other cases are more tricky and will requite more effort to debug.

Ok, thanks.

aap-sc, 2024-06-11:
> Perhaps some changes to upstream OpenOCD functionality may be required in order to address some of these failures/exceptions?

yeah. I'm not sure why it was decided to send warnings along with the useful data for "monitor" commands sent by GDB.

TommyMurphyTM1234, 2024-06-11:
> > Perhaps some changes to upstream OpenOCD functionality may be required in order to address some of these failures/exceptions?
> 
> yeah. I'm not sure why it was decided to send warnings along with the useful data for "monitor" commands sent by GDB.

Could it be a `stdout` vs `stderr` mismatch issue perhaps? E.g. "regular" output should go to the former and warning/error output to the latter, maybe?

aap-sc, 2024-06-11:
> Could it be a stdout vs stderr mismatch issue perhaps? E.g. "regular" output should go to the former and warning/error output to the latter, maybe?

Kind of but not exactly. OpenOCD sends data (reply) to GDB via dedicated channel (socket/pipe/whatever) via GDB serial protocol. So the output will go to the same sink anyway. That being said - probably warnings should be sent as "asynchronous notifications" that gdb serial protocol should support. I'm not quite sure what OpenOCD does at the moment.

If it sends these warnings as a notification - we should have facilities to differentiate between these notifications and the actual command output on client side (that is GDB client). If not - this should be fixed. If these warnings are sent along with the actual output - this should be tacked too in my opinion.

The above still requires an investigation - I'm not ready to make a definitive conclusion yet. I'll post a separate notification once I get a more substantial understanding of what is going on.