riscv-collab/riscv-openocd #1051: Clean up/cull stale branches?

TommyMurphyTM1234, 2024-04-22:
As far as I can see...

* [`riscv`](https://github.com/riscv-collab/riscv-openocd) is the effective "master" branch in this repo on which most development occurs and into which development on other branches is merged.
* Only it and these three branches are deemed by GitHub to be active:
  * https://github.com/riscv-collab/riscv-openocd/branches/active
* All 62 other branches are deemed by GitHub to be stale:
  * https://github.com/riscv-collab/riscv-openocd/branches/stale

Perhaps some of these stale branches can be culled in order to avoid [confusion](https://github.com/riscv-collab/riscv-openocd/issues/1047#issuecomment-2068891979) - especially confusion between the [`master`](https://github.com/riscv-collab/riscv-openocd/tree/master) branch and the effective "master" branch [`riscv`](https://github.com/riscv-collab/riscv-openocd)?

TommyMurphyTM1234, 2024-10-15:
Can/should this be done?
Retaining so many branches that are up to 7 years old seems pointless and just confusing to me.
On the other hand maybe some of these still contain some useful changes that might be revived at some stage?
And if the branches are not of interest then they can easily be ignored?

If any/all should be deleted then maybe an admin with rights to delete them via the GitHub interface could do this rather than requiring a PR from a forked repo as it seems a bit convoluted to do it that way?

Feedback welcome...

JanMatCodasip, 2024-10-17:
As for `master` vs. `riscv` - 

I'd recommend to keep the original state - to retain `riscv` as the main branch where the development takes place. The reason is that external users of this repository already count with that (e.g. in their scripts), and I would not like to break that.

As for cleaning up old branches - 

For the time being, I'd rename all the branches to `archive/<orig_name>` (or similar) to clearly mark them as archived, but still leave them available should anyone need them.

TommyMurphyTM1234, 2024-10-17:
Thanks for the feedback @JanMatCodasip.

> As for `master` vs. `riscv` -
> 
> I'd recommend to keep the current state - to retain `riscv` as the main branch where the development takes place. The reason is that external users of this repository already count with that (e.g. in their scripts), and I would not like to break things for them.

I may not have been clear. `riscv` is the "main" branch and I was not suggesting that this should change. If anything I was suggesting that the `master` branch should be eliminated (or, as you suggest, marked in some way as "archived"). I was not suggesting that `master` be made the "main" branch or that `riscv` be renamed to `master`. (Also since the use of the term "master" is discouraged these days for other reasons).

> As for cleaning up old branches -
> 
> I'd also prefer to clean these up. For the time being, I'd recommend to rename all the branches to `archive/<orig_name>` (or similar) to clearly mark them as archived, but still leave them available should anyone need them.

That sounds like a good compromise for now alright. 
Or maybe a branch can actually be archived in GitHub? I'm not sure. Maybe it is only possible to archive a while repo?
Either way, if something like this is decided maybe it can be done via GitHub directly by an admin with suitable privileges rather than requiring a PR from a forked repo as that seems convoluted?

Thanks again for the feedback @JanMatCodasip.

JanMatCodasip, 2024-10-17:
@TommyMurphyTM1234 Thank you for your reply.

> If anything I was suggesting that the master branch should be eliminated (or, as you suggest, marked in some way as "archived").

Sounds good to me. We could mark the "master" as archived: `master -> archive/master`.

> Or maybe a branch can actually be archived in GitHub? I'm not sure.

I don't know of an "archive" feature. We can only rename the branch (or make a tag and then delete it).

> maybe it can be done via GitHub directly by an admin with suitable privileges rather than requiring a PR from a forked repo as that seems convoluted?

If all maintainers agree, I'll be happy to make these changes.