riscv-collab/riscv-openocd #1052: Use "riscv set_ir dtmcs 0x20000" command can't trigger dmihardreset?

Alan-19950616, 2024-04-23:
I'm debugging a RISCV CPU and I noticed that I can't trigger a dmihardreset using OpenOCD's command, is there a problem here?

This is part of the configuration

```
init
riscv set_ir dtmcs 0x20000
reset
halt
```

TommyMurphyTM1234, 2024-04-23:
Are you sure that your target implements it? Seems like many implementations do not or, at least, did not in the past:

* https://github.com/riscv/riscv-debug-spec/issues/470

It may help to capture and attach a verbose `openocd -d3` log for analysis.

Note that `dtmcs[17]` is `dtmcs.dmihardreset` in the 0.13 spec but is now `dtmcs.dtmhardreset` in the latest 1.0.0-rc2.1 frozen draft.

Alan-19950616, 2024-04-23:
I'm sure this feature is implemented on my hardware, the current debug version of the hardware is 0.13

Attaching the log file later

Alan-19950616, 2024-04-23:
Is that my misunderstanding.

How do I write the dtmcs registers correctly? Via configuration file

en-sc, 2024-04-23:
@Alan-19950616, you can use the following:
```
init
poll disable
irscan <tap name> 0x10
drscan <tap name> 32 0x20000
poll enable
reset
halt
```

en-sc, 2024-04-23:
A bit of ellaboration:
* `poll disable/enable` is needed, so OpenOCD does not insert poll operations. These would change IR back to `dmi`.
* If the script above fails, try adding `irscan <tap name> 0x11` before `poll enable` to set IR back to `dmi`. Though this should not be needed and should probably be addressed as a bug.

Alan-19950616, 2024-04-23:
Thanks for the reply, my problem has been solved.

I'll close issue.

closed 2024-04-23 by Alan-19950616