riscv-collab/riscv-openocd #1053: mcontrol6.timing bit is set

JAYANTH-IITM, 2024-04-23:
While trying to implement type 6 trigger (mcontrol6)  with load bit enabled according to the debug spec 1.0 but openocd doesnt seem to support (still using 0.13  version) . tinfo.version == 0  , implies implementation of older debug spec.This issue is persistent also after updating to latest version of openocd and riscv-tests.After the checking the log attached below , dmstaus.version == 2 , which according to latest spec , means version 0.13. Expected to implement in version 1.0

[20240423-115939-spike32-TriggerDmode.log](https://github.com/riscv-collab/riscv-openocd/files/15072645/20240423-115939-spike32-TriggerDmode.log)

en-sc, 2024-04-23:
I don't understand what is the issue with OpenOCD here.

AFAIU, you are using Spike simulator.

* Spike reports `dmstaus.version == 2`
https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/debug_module.cc#L100
* Which is correctly handled by OpenOCD, e.g:
```
Debug: 123 7 riscv-013.c:439 dump_field(): read: dmstatus=0x4c0c02 {version=0_13 anyrunning=1 allrunning=1 anyhavereset=1 allhavereset=1 impebreak=1}
```

* Spike reports `tinfo.version == 0`
https://github.com/riscv-software-src/riscv-isa-sim/blob/3192ee4d31f481e84281a24d55bb6130e3743668/riscv/triggers.cc#L622-L631
* Which is again correctly handled by OpenOCD.
```
Debug: 4516 1777 riscv.c:5267 riscv_get_register(): [riscv.cpu] Read tinfo: 0x807c
Debug: 4517 1777 riscv.c:5501 riscv_enumerate_triggers(): [riscv.cpu] Trigger 0: supported types (mask) = 0x0000807c
```

Could you please elaborate what your expectations are and which OpenOCD behavior does not correspond to these expectations?

JAYANTH-IITM, 2024-04-23:
I want to run with version = 1 (dmstatus.version == 3). Now it is currently running with version 0.13 (dmstatus.version == 2). I have updated openOCD , spike and RISCV-TEST to the latest . I expect to get 100807c in tinfo (tinfo.version == 1). Kindly guide me .

aap-sc, 2024-04-23:
@JAYANTH-IITM dmstatus is reported by spike. @en-sc pointed out the exact line in the source code https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/debug_module.cc#L100 . The version reported by spike is not controlled by OpenOCD.

If you want Spike to support RISC-V Debug Spec 1.0, you need to either make or request appropriate changes there.

JAYANTH-IITM, 2024-04-23:
Thanks for the clarification and support

closed 2024-04-23 by en-sc