riscv-collab/riscv-openocd #1057: The method of burning program to SoC: PB, SBA, AC in external debug

zhangdujiao, 2024-04-24:
For the external debug, before debugging starts, GDB needs to load the program into memory or flash. At this time, is it also done through the three methods mentioned: PB, SBA, AC. However, PB needs to work together with AC to access memory, and AC can only be used when the core enters the debug mode (during program burning, the core should not have entered the debug mode, right?).
So, in this scenario, can the program only be burned through SBA?

aap-sc, 2024-04-24:
@zhangdujiao your question is more about debug spec than OpenOCD. To double-check it's better to ask this question there: https://github.com/riscv/riscv-debug-spec.

My understanding of the situation is. Debug Specification defines several facilities to access memory, namely the spec debug 3 ways to access memory:

1. Program Buffer. Requires the hart to be in Debug Mode.
2.  Abstract Access Memory Command. Optionally, this does not require the hart to be in Debug Mode according to the spec. Quoting the spec:

```
Debug Modules may optionally implement this command and may support read and write access to
memory locations when the selected hart is running or halted. If this command supports memory
accesses while the hart is running, it must also support memory accesses while the hart is halted.
```

3. System Bus Access.

```
A debugger can access memory from a hart’s point of view using a Program Buffer or the Abstract
Access Memory command. (Both these features are optional.) A Debug Module may also include a
System Bus Access block to provide memory access without involving a hart, regardless of whether
Program Buffer is implemented. 
```

Few additional notes:

1. Accessing flash memory is typically done via dedicated driver. I can't comment on the specifics of how this implemented, my guess is that it still uses hart to interact with the controller. Better check documentation or upstream openocd mailing list, since this is not riscv-specific question.
2. There was also a somewhat similar question a while ago https://github.com/riscv-collab/riscv-openocd/issues/1038

zhangdujiao, 2024-04-24:
Thank you very much for your detailed answers and corrections. 
During the initialization process of debug, after loading the configuration file (with reset halt command last) of OpenOCD, the core enters the debug mode automatically, is that correct?

zhangdujiao, 2024-04-24:
many thanks :)

aap-sc, 2024-06-05:
I assume that the question is answered. So closing the issue. Please feel free if you think that we still need to clarify something

closed 2024-06-05 by aap-sc