riscv-collab/riscv-openocd #1059: The OpenOCD service cannot be killed by kill-9

liushuai05, 2024-04-24:
When OpenOCD starts the service, if usb is remotely mounted by usbip, the process of OpenOCD is likely to be "disk sleep" and cannot be killed by kill-9. May I ask someone who has the same problem as me?

- I'm sorry my English is not very good. The information comes from the translation software.

TommyMurphyTM1234, 2024-04-24:
I don't think that this is an OpenOCD specific issue judging by this and the related/duplicate issues linked there: 

* https://github.com/microsoft/WSL/issues/2246

I tried to use usbip with WSL2 but no matter what instructions I followed couldn't get it to work so I stick with physical or virtual machine Linux installations.

liushuai05, 2024-04-24:
> I don't think that this is an OpenOCD specific issue judging by this and the related/duplicate issues linked there:
> 
>     * [kill -9 doesn't work for .exe launched from WSL microsoft/WSL#2246](https://github.com/microsoft/WSL/issues/2246)
> 
> 
> I tried to use usbip with WSL2 but no matter what instructions I followed couldn't get it to work so I stick with physical or virtual machine Linux installations.

Your answer seems to be windows, but my operating system is archlinux, I would like to ask this operating system is the same solution

TommyMurphyTM1234, 2024-04-24:
Sorry - I misread "usbip" as "usbipd" and inferred that you were using WSL and USB passthrough from the host.

Please describe in detail your use case and what exactly happens/goes wrong.

TommyMurphyTM1234, 2024-04-24:
`kill -9`` works fine for me on Zorin OS 17.1 Core:
```
# Run OpenOCD
openocd -f targets/SiFive/HiFive1.cfg
Open On-Chip Debugger 0.12.0+dev-03744-g3991492cc-dirty (2024-04-23-15:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive Inc), part: 0x0e31, ver: 0x1)
Warn : Got exception 0xffffffff when reading tinfo
Warn : Got exception 0xffffffff when reading tinfo
Info : [riscv.cpu] Found 2 triggers
halted at 0x400 due to software breakpoint
Info : Examined RISCV core; XLEN=32, misa=0x40001105
[riscv.cpu] Target successfully examined.
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Found flash device 'issi is25lp128d' (ID 0x0018609d)
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

# In another terminal
ps -a
    PID TTY          TIME CMD
   2207 tty2     00:00:00 gnome-session-b
 431835 pts/4    00:00:00 openocd
 431839 pts/2    00:00:00 ps
kill -9 431835
ps -a
    PID TTY          TIME CMD
   2207 tty2     00:00:00 gnome-session-b
 431840 pts/2    00:00:00 ps

# OpenOCD reports termination
Killed
```
OpenOCD has specific code for handling signals:

* https://github.com/riscv-collab/riscv-openocd/blob/3991492cc101f4289be06ec7dc23c6ad7748a74a/src/server/server.c#L661-L679

liushuai05, 2024-04-24:
> Sorry - I misread "usbip" as "usbipd" and inferred that you were using WSL and [usbipd ](https://learn.microsoft.com/en-us/windows/wsl/connect-usb)USB passthrough from the host.
> 
> Please describe in detail your use case and what exactly happens/goes wrong.

I plan to use wifi to connect my daplink (esp32c, which runs a usbip service), and then download the program through openocd. Of course, usbip is needed as a bridge during this process. Make openocd see daplink in my LAN as a usb device link and transfer the corresponding command through telnet. Everything worked fine when I installed usbip and mounted the daplink device using usbip, but when I opened openocd, the problem occurred. I can't stop this openocd process (when I open it with stlink configuration, everything works fine and can be stopped normally)
Here are the orders I executed:
```
riscv-openocd -f interface/cmsis-dap.cfg -f target/stm32g0x.cfg 
```
Below is the previous usb link stlink command, this is not a problem
```
riscv-openocd -f interface/stlink-v2.cfg -f target/stm32g0x.cfg
```

TommyMurphyTM1234, 2024-04-24:
I think that [@aap-sc's post above](https://github.com/riscv-collab/riscv-openocd/issues/1059#issuecomment-2075298023) explains the issue and why it's not an OpenOCD specific issue.

liushuai05, 2024-04-24:
> > May I ask someone who has the same problem as me?
> 
> This problem is not specific to OpenOCD. This is a known pecularity of linux OS. Here is an example: https://superuser.com/questions/436239/cant-kill-user-process

Looking at the information you provided, I get the gist of what you are saying, linux involves network mounting and if the mount address is wrong, you may never be able to kill the process

liushuai05, 2024-04-24:
> I think that [@aap-sc's post above](https://github.com/riscv-collab/riscv-openocd/issues/1059#issuecomment-2075298023) explains the issue and why it's not an OpenOCD specific issue.

Ok very lucky to receive your solution

liushuai05, 2024-04-24:
> > May I ask someone who has the same problem as me?
> 
> This problem is not specific to OpenOCD. This is a known pecularity of linux OS. Here is an example: https://superuser.com/questions/436239/cant-kill-user-process

Thank you very much for your reply.

TommyMurphyTM1234, 2024-04-26:
I believe that this could/should be closed given that it's not actually an OpenOCD issue?

closed 2024-04-27 by en-sc