riscv-collab/riscv-openocd #1060: [riscv.cpu] Debug Module did not become active. dmcontrol=0x0

l1onog, 2024-04-26:
I implemented RISCV SoC on a Zcu102 fpga. I am trying to use openOCD and jtag-hs2 to download code to the board as instructed in the README. Does anyone know what could potentially cause this error and ways to fix it? @en-sc I would really appreciate it.

Here is what it shows when I run with -d3 flag:
```
Open On-Chip Debugger 0.12.0+dev-gca7d88252 (2024-04-26-14:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3User : 4 2 options.c:52 configuration_output_handler(): 
Debug: 5 2 options.c:233 add_default_dirs(): bindir=/usr/local/bin
Debug: 6 2 options.c:234 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 7 2 options.c:235 add_default_dirs(): exepath=/usr/local/bin
Debug: 8 2 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 2 configuration.c:33 add_script_search_dir(): adding /home/fpga/.config/openocd
Debug: 10 2 configuration.c:33 add_script_search_dir(): adding /home/fpga/.openocd
Debug: 11 2 configuration.c:33 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 12 2 configuration.c:33 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 13 2 command.c:153 script_debug(): command - ocd_find /opt/riscv/pulp/fpga/pulp-zcu102/digilent-hs2.cfg
Debug: 14 2 configuration.c:88 find_file(): found /opt/riscv/pulp/fpga/pulp-zcu102/digilent-hs2.cfg
Debug: 15 2 command.c:153 script_debug(): command - adapter speed 1000
Debug: 16 2 adapter.c:250 adapter_config_khz(): handle adapter khz
Debug: 17 2 adapter.c:214 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 18 2 adapter.c:214 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 19 2 command.c:153 script_debug(): command - adapter driver ftdi
Debug: 20 2 command.c:153 script_debug(): command - ftdi device_desc Digilent USB Device
Debug: 21 2 command.c:153 script_debug(): command - ftdi vid_pid 0x0403 0x6014
Debug: 22 2 command.c:153 script_debug(): command - ftdi channel 0
Debug: 23 2 command.c:153 script_debug(): command - ftdi layout_init 0x00e8 0x60eb
Debug: 24 2 command.c:153 script_debug(): command - ftdi layout_signal SWD_EN -data 0x6000
Debug: 25 2 command.c:153 script_debug(): command - ftdi layout_signal SWDIO_OE -data 0x20
Debug: 26 3 command.c:153 script_debug(): command - transport select
Info : 27 3 transport.c:267 handle_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 28 3 command.c:153 script_debug(): command - transport select
Debug: 29 3 command.c:153 script_debug(): command - jtag newtap riscv unknown0 -irlen 5 -expected-id 0x10102001
Debug: 30 3 tcl.c:401 handle_jtag_newtap_args(): Creating New Tap, Chip: riscv, Tap: unknown0, Dotted: riscv.unknown0, 4 params
Debug: 31 3 core.c:1474 jtag_tap_init(): Created Tap: riscv.unknown0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 32 3 command.c:153 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x249511C3
Debug: 33 3 tcl.c:401 handle_jtag_newtap_args(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 34 3 core.c:1474 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 35 3 command.c:153 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu -coreid 0x3e0
Debug: 36 3 target.c:2136 target_free_all_working_areas_restore(): freeing all working areas
Debug: 37 3 riscv.c:437 riscv_create_target(): [riscv.cpu] riscv_create_target()
Debug: 38 3 command.c:153 script_debug(): command - gdb_report_data_abort enable
Debug: 39 3 command.c:153 script_debug(): command - gdb_report_register_access_error enable
Debug: 40 3 command.c:153 script_debug(): command - scan_chain
Debug: 41 3 command.c:153 script_debug(): command - reset_config none
Debug: 42 3 command.c:153 script_debug(): command - init
Debug: 43 3 command.c:153 script_debug(): command - target init
Debug: 44 3 command.c:153 script_debug(): command - target names
Debug: 45 3 command.c:153 script_debug(): command - riscv.cpu cget -event gdb-flash-erase-start
Debug: 46 3 command.c:153 script_debug(): command - riscv.cpu configure -event gdb-flash-erase-start reset init
Debug: 47 3 command.c:153 script_debug(): command - riscv.cpu cget -event gdb-flash-write-end
Debug: 48 3 command.c:153 script_debug(): command - riscv.cpu configure -event gdb-flash-write-end reset halt
Debug: 49 3 command.c:153 script_debug(): command - riscv.cpu cget -event gdb-attach
Debug: 50 3 command.c:153 script_debug(): command - riscv.cpu configure -event gdb-attach halt 1000
Debug: 51 3 target.c:1594 handle_target_init_command(): Initializing targets...
Debug: 52 3 riscv.c:450 riscv_init_target(): [riscv.cpu] riscv_init_target()
Debug: 53 3 semihosting_common.c:107 semihosting_common_init():  
Debug: 54 3 ftdi.c:732 ftdi_initialize(): ftdi interface using shortest path jtag state transitions
Debug: 55 5 mpsse.c:406 mpsse_purge(): -
Debug: 56 6 mpsse.c:687 mpsse_loopback_config(): off
Debug: 57 6 mpsse.c:732 mpsse_set_frequency(): target 1000000 Hz
Debug: 58 6 mpsse.c:724 mpsse_rtck_config(): off
Debug: 59 6 mpsse.c:713 mpsse_divide_by_5_config(): off
Debug: 60 6 mpsse.c:693 mpsse_set_divisor(): 29
Debug: 61 6 mpsse.c:756 mpsse_set_frequency(): actually 1000000 Hz
Debug: 62 6 adapter.c:214 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 63 6 adapter.c:218 adapter_khz_to_speed(): have adapter set up
Debug: 64 6 mpsse.c:732 mpsse_set_frequency(): target 1000000 Hz
Debug: 65 6 mpsse.c:724 mpsse_rtck_config(): off
[... 132 lines cut ...]

ftdi channel 0
ftdi layout_init 0x00e8 0x60eb
ftdi layout_signal SWD_EN -data 0x6000
ftdi layout_signal SWDIO_OE -data 0x20

set _CHIPNAME riscv

jtag newtap $_CHIPNAME unknown0 -irlen 5 -expected-id 0x10102001
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x3e0

gdb_report_data_abort enable
gdb_report_register_access_error enable

# dump jtag chain
scan_chain
reset_config none

init
halt
echo "Ready for Remote Connections"
```

en-sc, 2024-04-26:
From the log it seems you are using a version from https://github.com/riscv-collab/riscv-openocd/commit/ca7d88252633a5d452ee92e6f8f3eb20cadfb633 (Feb 27).
Please, try to use a more recent version. There were recent changes in DM examination related code. This would help me with diagnosing the issue a lot.
Also, could you please attach a log in a file / zip archive. The conversation will get flooded with logs otherwise.

l1onog, 2024-04-26:
hi, @en-sc , I have used the latest version of openocd, and this is my result. 
[d3.txt](https://github.com/riscv-collab/riscv-openocd/files/15126970/d3.txt)

en-sc, 2024-04-26:
@l1onog, looking at the logs it may be a different issue.
Please, check if #1061 fixes it.

l1onog, 2024-04-27:
hi, @en-sc, I appreciate your suggestion, but it seems like it won’t resolve my issue.  
[new_d3.txt](https://github.com/riscv-collab/riscv-openocd/files/15138429/new_d3.txt)
I adjusted the 'set_command_timeout_sec' from 2 seconds to 240 seconds as suggested by the error message, but it still isn’t working."

l1onog, 2024-05-07:
> @l1onog, looking at the logs it may be a different issue. Please, check if #1061 fixes it.

hi, @en-sc , I tried to replace the new JTAG-HS2, but it still reports an error, this is the exact error report, 
[d3-2024-05-07.txt](https://github.com/riscv-collab/riscv-openocd/files/15231042/d3-2024-05-07.txt)
how do I actually fix it? What is the problem?

en-sc, 2024-05-08:
@l1onog, in the log you have attached it can be observed that:
1. The DMI `dtmcs` is read successfully:
```
Debug: 97 6 riscv.c:1725 riscv_examine(): [riscv.cpu] Starting examination
Debug: 98 6 riscv.c:406 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x1c71
Debug: 99 6 riscv.c:1739 riscv_examine(): [riscv.cpu] dtmcontrol=0x1c71
Debug: 100 6 riscv.c:1741 riscv_examine(): [riscv.cpu] version=0x1
```
2. However, the following read of DM's `dmcontrol` register does not succeed and returns busy for 2 minutes.
```
Debug: 109 7 riscv-013.c:415 riscv_decode_dmi_scan(): 41b r 00000000 @10 -> b 00000000 @10; 0i
Debug: 110 7 riscv-013.c:492 increase_dmi_busy_delay(): [riscv.cpu] dtmcs_idle=1, dmi_busy_delay=1, ac_busy_delay=0
Debug: 111 7 riscv-013.c:481 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Debug: 112 7 riscv-013.c:415 riscv_decode_dmi_scan(): 41b r 00000000 @10 -> b 00000000 @10; 1i
...
Debug: 562 124197 riscv-013.c:415 riscv_decode_dmi_scan(): 41b r 00000000 @10 -> b 00000000 @10; 11019046i
Debug: 563 124197 riscv-013.c:492 increase_dmi_busy_delay(): [riscv.cpu] dtmcs_idle=1, dmi_busy_delay=12120951, ac_busy_delay=0
Debug: 564 124199 riscv-013.c:481 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Error: 565 124199 riscv-013.c:694 dmi_op(): [riscv.cpu] DMI operation didn't complete in 120 seconds. The target is either really slow or broken. You could increase the timeout with riscv set_command_timeout_sec.
```
I can see the following reasons for such behavior:
1. There is no DM at with DMI base address `0x0`. This would indicate an issue with the Debug Module / Debug Message Interface you are using, since the spec requires [3.1. Debug Module Interface (DMI)]:
> The bottom of the address space is used for the first (and usually only) DM
Please, verify this is not the case with your provider of  Debug Module / Debug Message Interface implemetation.
2. In your DM implementation `dmstatus` can not be read while `dmstatus.dmactive` is low. Here I would also need further information about your DM implementation -- which version of  RISC-V Debug Spec it supports (v0.13 or v1.0)? The specification is obscure in what is required from the DM when `dmstatus.dmactive` is low and there is an issue described here: https://github.com/riscv/riscv-debug-spec/issues/1021

l1onog, 2024-05-09:
Hi, @en-sc , Thank you for your kind reply, the version of RISCV debugging I am using is 0.13. Also, if it is my debugging module that is causing the problem, how can I fix it?

TommyMurphyTM1234, 2024-05-09:
> Also, if it is my debugging module that is causing the problem, how can I fix it?

By modifying the HDL code for the FPGA implementation of your RISC-V. But that's moot until such time as the root cause of your issues has been identified.