riscv-collab/riscv-openocd #1065: Improve stability of riscv-tests for targets with low remotetimeout value

aap-sc, 2024-05-02:
After https://github.com/riscv-software-src/riscv-tests/pull/553 was merged in @en-sc expressed the concern that this is a rather radical change and it reduces coverage of the test suite.

During a very heated discussion we've decided that:

1. It would be nice if we could bring smaller timeouts back (for all targets)
2. However, in order to do that we should get rid of the sporadic failures when these timeouts are small. Most likely the presence of these timeout errors is a result of missing keep_alive packets that OpenOCD should send to GDB - this needs to be investigated and fixed (if needed)

@en-sc , @TommyMurphyTM1234 FYI

JanMatCodasip, 2024-05-03:
@aap-sc Please, what is the relationship between the timeouts in the testsuite and the coverage? In what way does the extended timeout reduce the coverage? Thank you for explanation.

TommyMurphyTM1234, 2024-05-03:
I presumed that the reasoning here was that (subject to further investigation/analysis) some of the unexpected test failures/exceptions might point to problems with OpenOCD itself - in particular not sending sufficient "keep alive" packets to GDB to avoid anomalous timeouts?

FWIW at the moment I'm encountering several problems when running tests: 

* `DisconnectTest`: https://github.com/riscv-collab/riscv-openocd/issues/1058 - this one seems to be a non-timeout/keep alive related bug in OpenOCD.
* `MemorySampleSingle` against Spike: https://github.com/riscv-collab/riscv-openocd/issues/1049#issuecomment-2088584628
* Several other sporadic failures against Spike for which I'm currently trying to collect some data in order to see what's up.

JanMatCodasip, 2024-05-03:
@aap-sc @TommyMurphyTM1234 Understood, thank you for explanation!