riscv-collab/riscv-openocd #1067: load_image returns error 'invalid command'

Stycore, 2024-05-06:
I'm trying to load a .elf file onto an Ibex riscv core on a CW305 FPGA using the following shell script [provided by the Ibex developers](https://github.com/lowRISC/ibex-demo-system/blob/main/util/load_demo_system.sh):
```
#!/bin/sh
SCRIPT_DIR="$(dirname "$(readlink -e "$0")")"
TCL_FILE=$SCRIPT_DIR/cw-openocd-cfg.tcl

EXIT_CMD=''

if [ $1 = "run" ]; then
  EXIT_CMD='-c "exit"'
fi

openocd -f $TCL_FILE -c "load_image $2 0x0" \
 -c "verify_image $2 0x0" \
 -c "echo \"Doing reset\"" \
 -c "reset $1" \
 $EXIT_CMD
 ```

This is executed with `script.sh run my-elf.elf` using openocd version `Open On-Chip Debugger 0.12.0+dev-00569-gbc9ca5f4a`
also the cfg file if it is useful (mostly copied from [here](https://github.com/newaetech/chipwhisperer/blob/develop/openocd/cw_openocd.cfg)):
```
adapter driver ftdi
ftdi vid_pid 0x2b3e 0xace2
ftdi channel 1
ftdi layout_init 0x001B 0x001B
ftdi layout_signal nSRST -data 0x0010
ftdi layout_signal SWD_EN -data 0x0010
ftdi layout_signal SWDIO_OE -data 0x0020
adapter speed 500
reset_config srst_push_pull
```

I cannot seem to avoid the error `invalid command name "load_image"` and can also not find any documentation on this, so any help would be appreciated.

aap-sc, 2024-05-06:
@Stycore this is not specifc to riscv-openocd.  You configuration file `cw-openocd-cfg.tcl` is incompatible with the command line you use. The problem is that you don't have `init/halt` statements in the configuration file.  Take a look at the original config from Ibex: https://github.com/lowRISC/ibex-demo-system/blob/main/util/arty-a7-openocd-cfg.tcl last two lines. You can't use `load_image` before `init` completes

Stycore, 2024-05-06:
@aap-sc thanks for the reply. Adding `init` (and `halt`) makes it fail with this error:
```
Error: failed to reset FTDI device: LIBUSB_ERROR_PIPE
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
```
I'm guessing the cfg is still incomplete, but I don't see how

TommyMurphyTM1234, 2024-05-06:
> @aap-sc thanks for the reply. Adding `init` (and `halt`) makes it fail with this error:
>
> ```
> Error: failed to reset FTDI device: LIBUSB_ERROR_PIPE
> Error: unable to open ftdi device with description [...]

What happens if you run OpenOCD with `sudo/root` privileges?
If it works with elevated privileges then you probably didn't install the [OpenOCD `udev` file](https://github.com/riscv-collab/riscv-openocd/blob/riscv/contrib/60-openocd.rules) to allow non-root access to devices such as the FTDI probe for users in the `plugdev` group.

aap-sc, 2024-05-06:
> I'm guessing the cfg is still incomplete, but I don't see how

Again, this is not something specific to riscv-openocd. That is this is a general question. You are unable to properly initialize debug adapter. You should double check that you've specified all the parameters correctly (you should consult your debug adapter documentation) and you should make sure that your user have permissions to access this device (that is you have proper udev rules installed), or as an alternative (though I DO NOT advise this) you could try to run openocd as a root user to see if the issue is with permissions.

That being said, if you still have issues I suggest you to ask a question in the mailing list of https://openocd.org/ project since you have a basic usability issue. Here we mostly track riscv-specific issues.

Note: it does not mean that you won't be able to get any help here, I'm just trying to point out to a more appropriate place for these kinds of questions.

TommyMurphyTM1234, 2024-05-06:
> Error: failed to reset FTDI device: LIBUSB_ERROR_PIPE

Also, if you're using Windows then you may need to install the relevant drivers using [Zadig](https://zadig.akeo.ie/).
As @aap-sc says, none of this is RISC-V OpenOCD specific and is covered in detail elsewhere - for example:

* https://foss-for-synopsys-dwc-arc-processors.github.io/toolchain/ide/how-to-use-openocd-on-windows.html

closed 2024-05-06 by Stycore