riscv-collab/riscv-openocd #1068: Q extension support

en-sc, 2024-05-07:
Currently RISC-V OpenOCD does not support RISC-V Q extension targets.

TommyMurphyTM1234, 2024-05-07:
Can you explain this a bit more?
Is it the case that OpenOCD's RISC-V support has to have awareness of FP registers and copes with F and D but not Q variants?

en-sc, 2024-05-08:
> Can you explain this a bit more? Is it the case that OpenOCD's RISC-V support has to have awareness of FP registers and copes with F and D but not Q variants?

Exactly. OpenOCD will report FPU registers as 64-bit on Q extension targets.

> Also - do targets exist that implement Q support? In hardware or just simulators?

AFAIU, Spike does support Q extension.

TommyMurphyTM1234, 2024-05-08:
> Exactly. OpenOCD will report FPU registers as 64-bit on Q extension targets.

Thanks, that makes sense. 👍 

> AFAIU, Spike does support Q extension.

Yes, that seems to be the case alright. 

* https://github.com/riscv-software-src/riscv-isa-sim

> Spike supports the following RISC-V ISA features:
> 
> * ...
> * Q extension, v2.2
> * ...