riscv-collab/riscv-openocd #1069: how to read v register if vslide1down instruction not present

fanghuaqi, 2024-05-13:
Hi, I am wondering if a risc-v vector don't have vslide1down instruction, how to read the vector register

https://github.com/riscv-collab/riscv-openocd/blob/ae7ffa424e1b686559ae833ce2d384ded47ba4c4/src/target/riscv/riscv-013.c#L2522-L2546

Is there any other way to read vector register in openocd?

Thanks

TommyMurphyTM1234, 2024-05-13:
I find the way that specifications are published very confusing.

If I go here:

* https://riscv.org/technical/specifications/

It says:

> **ISA Specifications (Ratified)**
>
> The specifications shown below represent the current, ratified and published releases.
> * Volume 1, Unprivileged Specification version 20240411  [[PDF](https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view?usp=drive_link)][[GitHub](https://github.com/riscv/riscv-isa-manual/releases/tag/20240411)]

But if I download that PDF then it says:

> **31.1. Introduction**
> 
> This document is version 1.1-draft of the RISC-V vector extension.

Which contradicts the earlier comment that the docs linked are "the current, ratified and published releases".

I'm trying to see what version of the RISC-V V vector extension the "vector slide" instructions belong to but I can't seem to ascertain that.

fanghuaqi, 2024-05-13:
`vslide1down` list as below

[image]

aap-sc, 2024-05-13:
@fanghuaqi regardless of the version of the spec. Current implementation assumes that vslide1down is present. If this instruction is not support by your target - you won't be able to read vector registers. Patches or issues are welcome.

TommyMurphyTM1234, 2024-05-13:
What is the nature of the RISC-V target that you're using that does not support `vslide1down`?
In particular what version of the RISC-V V vector extension does it implement?

fanghuaqi, 2024-05-13:
> @fanghuaqi regardless of the version of the spec. Current implementation assumes that vslide1down is present. If this instruction is not support by your target - you won't be able to read vector registers. Patches or issues are welcome.

Hi @aap-sc , is there any other way to implement read vector register in openocd

TommyMurphyTM1234, 2024-05-13:
Is this of any relevance here perhaps?

* https://github.com/riscv-collab/riscv-openocd/pull/505

> Accessing registers on targets that implement 0.9 or earlier will no
longer work. 

If not then it might be worth reviewing other PRs related to vector register accesses - i.e. some or all of the following:

* https://github.com/riscv-collab/riscv-openocd/pulls?q=is%3Apr+vector+registers

fanghuaqi, 2024-05-13:
Hi @aap-sc ,  thanks for your explaination, I will see how to implement it using the way you suggest.

aap-sc, 2024-05-13:
> But that's my confusion here - what official specification (and version)? 

https://github.com/riscv/riscv-v-spec/releases/download/v1.0/riscv-v-spec-1.0.pdf (1.0, not ratified, FROZEN, public review)
https://github.com/riscv/riscv-v-spec/releases/download/zvfh/zvfh-public-review-complete.pdf (1.1, not ratified, FROZEN, public review COMPLETE)

> I presume you mean a ratified version of the V vector extension? But what version? Isn't 1.0 the only ratified version so far?

There is no **ratified** version currently. HOWEVER, the versions presented above are FROZEN. This means that no major changes are expected before the formal ratification.

>  I can't seem to find an authoritative V extension 1.0 spec and the unprivileged ISA spec linked from the RISC-V International page is supposed to be ratified but still contains draft extension narrative.

I've provided the links above. While these are marked as "draft" - these are frozen. Vector Slidedown/Slideup Instructions seems to be mandatory in these specifications.

TommyMurphyTM1234, 2024-05-13:
> There is no **ratified** version currently. HOWEVER, the versions presented above are FROZEN. This means that no major changes are expected before the formal ratification.

As I said - it's confusing...

* https://wiki.riscv.org/display/HOME/Ratified+Extensions

[image]

aap-sc, 2024-05-13:
> As I said - it's confusing...

O_o. Yeah... Quite unexpected.

TommyMurphyTM1234, 2024-05-13:
> > As I said - it's confusing...
> 
> O_o. Yeah... Quite unexpected.

Yes - and also strange that the vector extension ostensibly ratified in 2021 was only officially announced this year:

* https://riscv.org/announcements/2024/04/risc-v-international-achieves-milestone-with-ratification-of-40-specifications-in-two-years/

aap-sc, 2024-05-13:
@TommyMurphyTM1234 , however. If we look here:

https://riscv.org/technical/specifications/ , there is this document "Volume 1, Unprivileged Specification version 20240411 " which seems to be ratified. Vector isa is present there. 
https://drive.google.com/file/d/1uviu1nH-tScFfgrovvFCrj7Omv8tFtkp/view?usp=drive_link

TommyMurphyTM1234, 2024-05-13:
> https://riscv.org/technical/specifications/ , there is this document "Volume 1, Unprivileged Specification version 20240411 " which seems to be ratified. 

That page says that it's ratified but it still contains draft info.
That's what I was saying earlier above:

* https://github.com/riscv-collab/riscv-openocd/issues/1069#issuecomment-2107298197

and here:

* https://github.com/riscv/riscv-isa-manual/issues/1400

aap-sc, 2024-05-13:
@TommyMurphyTM1234 yeah. https://github.com/riscv/riscv-isa-manual/issues/1400 - this good thing we have this issue. I hope that the situation will be clarifed soon.

aap-sc, 2024-06-05:
@fanghuaqi could you please clarify if your question is answered? If you have plans to submit MR we can leave this issue open (it would not hurt to rename then), otherwise I suggest to close the issue.

fanghuaqi, 2024-06-06:
Yes, you can close it, we implement read csr not using vslidedown instruction, but require working memory.

Thanks

closed 2024-06-06 by fanghuaqi