riscv-collab/riscv-openocd #1071: Issues Flashing Baremetal Applications Directly with OpenOCD on MPFS

abelixrev, 2024-05-27:
Hi everyone,

I'm trying to flash baremetal applications directly using OpenOCD for the Microsemi RISC-V board. Here's the command I'm using:

```bash
openocd -c 'set DEVICE MPFS' -f 'board/microsemi-riscv.cfg' -c 'program mpfs-emmc-sd-write-read.elf'
```
```

Here's the content of `board/microsemi-riscv.cfg`:

```cfg
#------------------------------------------------------------------------------
# Microsemi RISC-V board
#------------------------------------------------------------------------------
 
# FlashPro
source [find interface/microsemi-flashpro.cfg]
 
# Device
source [find target/microsemi-riscv.cfg]
 
# Board specific initialization
proc do_board_reset_init {} {
}
```

However, I'm encountering the following error:

```
** Programming Started **
flash
 flash bank bank_id driver_name base_address size_bytes chip_width_bytes
      bus_width_bytes target [driver_options ...]
 flash banks
 flash init
 flash list
...
```

When I checked `target/microsemi-riscv.cfg`, I noticed the following lines were commented out:

```cfg
device_family # MPFS      PolarFireSoc
device_envm_size # MPFS      0x20000
envm_base # PolarFireSoC  0x20200000
envm_driver # PolarFireSoC  microsemi-polarfire-soc-envm
```

I tried uncommenting these lines but received the error:

```
Error: flash driver 'microsemi-polarfire-soc-envm' not found
```

I then attempted to use another driver (`microsemi-smartfusion2-envm`) and got:

```
** Programming Started **
Error: invalid ELF file, only 32bits files are supported
...

```

I also tried to flash the `mpfs-emmc-sd-write-read.hex` instead of `mpfs-emmc-sd-write-read.elf`:

```bash
openocd -c 'set DEVICE MPFS' -f 'board/microsemi-riscv.cfg' -c 'program mpfs-emmc-sd-write-read.hex 0x08000000 verify reset exit'
```

and got:

```
** Programming Started **
Warn : no flash bank found for address 0x08000000
** Programming Finished **
```

Could anyone please guide me on the correct OpenOCD command to use for flashing the baremetal application `app_baremetal.elf`? Any help would be greatly appreciated.

Thank you!

TommyMurphyTM1234, 2024-05-27:
Microchip (who acquired Microsemi) use their own custom version of OpenOCD with specific support for their FPGA SoC eNVMs, FlashPro programmers etc. I don't believe that these customisations have been upstreamed to the master OpenOCD or this `riscv-openocd` projects to date. You'll have to contact them about finding the right OpenOCD/sources to use with their devices. 

Disclaimer: I used to work for Actel/Microsemi/Microchip on SoftConsole etc. from 2005 until late 2021.

TommyMurphyTM1234, 2024-05-27:
Closing this as it is not an issue with the `riscv-openocd` fork of OpenOCD.

closed 2024-05-27 by TommyMurphyTM1234