riscv-collab/riscv-openocd #1078: How can I build openocd with static libs?

gushi84, 2024-05-31:
I want to run openocd on an old centos server, but the glibc library version on the server is too low, so I want to compile a fully statically linked openocd.
   When I use " ./configure --enable-static --disable-shared LDFLAGS="-static" --enable-ftdi",。After the compilation is complete, it is still  dynamically linked ：
ldd src/openocd
        linux-vdso.so.1 (0x00007ffce6d5c000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007ff669678000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff669591000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff669368000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007ff66933e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007ff669b11000)

Can anyone give me some advice?

TommyMurphyTM1234, 2024-05-31:
This looks like a general query about (building) OpenOCD and not anything specific to the `riscv-openocd` fork.

At least one person previously asked the same thing on the OpenOCD mailing lists but unfortunately there are no replies:

* https://sourceforge.net/p/openocd/mailman/message/37229360/

Maybe there are other threads with useful advice?

* https://sourceforge.net/p/openocd/mailman/search/?q=static+build

As far as I recall @ilg-ul with his [xPack projects](https://xpack.github.io/) (including xPack OpenOCD) builds everything as far as possible statically so it may be worth studying what he does there?

* https://xpack.github.io/dev-tools/openocd/

Maybe if he had time he might have some general advice on this issue?

ilg-ul, 2024-05-31:
> I want to run openocd on an old centos server

You'll probably need to build openocd on that old machine.

gushi84, 2024-06-01:
> This looks like a general query about (building) OpenOCD and not anything specific to the `riscv-openocd` fork.
> 
> At least one person previously asked the same thing on the OpenOCD mailing lists but unfortunately there are no replies:
> 
> * https://sourceforge.net/p/openocd/mailman/message/37229360/
> 
> Maybe there are other threads with useful advice?
> 
> * https://sourceforge.net/p/openocd/mailman/search/?q=static+build
> 
> As far as I recall @ilg-ul with his [xPack projects](https://xpack.github.io/) (including xPack OpenOCD) builds everything as far as possible statically so it may be worth studying what he does there?
> 
> * https://xpack.github.io/dev-tools/openocd/
> 
> Maybe if he had time he might have some general advice on this issue?
> 
> > I want to run openocd on an old centos server, but the glibc library version on the server is too low, so I want to compile a fully statically linked openocd.
> 
> In case it's relevant here (I'm not completely clear on what exactly you want statically link), my recollection is that you cannot (easily or at all?) statically link GLIBC and that will usually be dynamic even if some or all other libs are statically linked. Which might mean that you need to build OpenOCD on your "old" distro in order for it to be able to run there. I believe that this is basically what @ilg-ul does with the xPack projects - i.e. builds on an "older" Linux distro dynamically linking to the older GLIBC on that platform for maximum compatibility with distros from that version onwards. Even if most of all other libraries are statically linked.

Thank you very much for your suggestion. Maybe I will try  xPack projects on my old machine.

TommyMurphyTM1234, 2024-06-01:
Closing this since it's not really an issue specific to `riscv-openocd` but a general question about building OpenOCD.

closed 2024-06-01 by TommyMurphyTM1234

ilg-ul, 2024-06-01:
> Maybe I will try xPack projects on my old machine.

If your machine has a GLIBC 2.27 or higher, the xPack binaries are expected to work.

TommyMurphyTM1234, 2024-06-01:
> > Maybe I will try xPack projects on my old machine.
> 
> If your machine has a GLIBC 2.27 or higher, the xPack binaries are expected to work.

For reference purposes this dependency is documented here:

* https://xpack.github.io/xbb/end-of-support/#current-base-version

ilg-ul, 2024-06-01:
>  this dependency is documented here ...

That's correct, thank you Tommy for providing the link.

This requirement is also mentioned in each release page, for example:

- <https://xpack.github.io/blog/2024/04/02/openocd-v0-12-0-3-released/#prerequisites>

ilg-ul, 2024-06-02:
Another tip: in case you decide to build static executables, please note that RedHat derived distributions do not pack the static libraries in the usual configurations, you have to explicitly install `glibc-static libstdc++-static`. Debian derived distributions are less strict and the static libraries are easier available.

gushi84, 2024-06-03:
> Another tip: in case you decide to build static executables, please note that RedHat derived distributions do not pack the static libraries in the usual configurations, you have to explicitly [...]

Yes, I can not  get all the static libraries that openocd depends on, such as udev. I tried to compile libudev.a from source and link it statically, but a "Segment fault" would occur when I run openocd.

TommyMurphyTM1234, 2024-06-03:
> Yes, I can not get all the static libraries that openocd depends on, such as udev. I tried to compile libudev.a from source and link it statically, but a "Segment fault" would occur when I run [...]

Why don't you just try the latest binary build of the xPack OpenOCD or do you specifically need a build of the very latest sources from this `riscv-openocd` repo?

If you need to build it yourself statically then you might want to study how @ilg-ul builds things with the xPack XBB approach or maybe see if this approach can be leveraged to build from this repo rather than the xPack source tarballs.

gushi84, 2024-06-03:
I cannot use the  latest binary build of the xPack OpenOCD because I have some  customizations for openocd. And then I will study  xPack XBB approach and find a way that works. Thank you both very much for your enthusiastic response. @ilg-ul @TommyMurphyTM1234

TommyMurphyTM1234, 2024-06-03:
You could also maybe ask about building OpenOCD statically on the OpenOCD mailing lists as mentioned earlier: 

* https://github.com/riscv-collab/riscv-openocd/issues/1078#issuecomment-2142001033

ilg-ul, 2024-06-03:
> I will study xPack XBB approach and find a way that works

If your machine has GLIBC 2.27 or later, the xPack XBB approach might be an answer; with some minimum customisation you can select your custom git repository; Tommy tried such builds some years ago.