riscv-collab/riscv-openocd #1106: openocd only supports Debug Module version 2 (0.13) and 3 (1.0), not 0 (dmstatus=0x0). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed

UmeshChandargi, 2024-07-22:
Hello, 
I am trying openocd with neorv32 riscv32 softcore processor.
openocd_neorv32.cfg
```
# NEORV32 on-chip debugger openOCD configuration file

# ----------------------------------------------
# Physical interface configuration
# -> ADJUST THIS FOR YOUR ADAPTER / SETUP
# ----------------------------------------------
adapter driver jlink
#adapter driver ftdi
#ftdi vid_pid 0x0403 0x6010
#ftdi channel 0
#ftdi layout_init 0x0038 0x003b

adapter speed 4000
transport select jtag

# ----------------------------------------------
# Target configuration
# ----------------------------------------------
set _CHIPNAME neorv32
jtag newtap $_CHIPNAME cpu -irlen 5
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME

# expose NEORV32-specific CSRs
riscv expose_csrs 2048=cfureg0
riscv expose_csrs 2049=cfureg1
riscv expose_csrs 2050=cfureg2
riscv expose_csrs 2051=cfureg3
riscv expose_csrs 4032=mxisa

# enable memory access error reports
gdb_report_data_abort enable

# ----------------------------------------------
# Start session
# ----------------------------------------------
init
halt

echo "Target HALTED. Ready for remote connections."
```

I initally connected to the openocd many times and it worked fine.
But then i got this error "openocd only supports Debug Module version 2 (0.13) and 3 (1.0), not 0 (dmstatus=0x0). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed."

And then onwards I am unable to connect to openocd.

Below the image where it was working initally and then stopped connecting.
[image]

I rechecked the pin connections and the pins are pulled up in the FPGA.

Could you please help me with this?

TommyMurphyTM1234, 2024-07-23:
Same issue raised here: 

- https://github.com/stnolting/neorv32/issues/961

UmeshChandargi, 2024-07-23:
Thank you for your reply.  I have added the openocd log with -d3 in the attachments. 
[openocd_log.txt](https://github.com/user-attachments/files/16347432/openocd_log.txt)

I apologise for the miscommunication earlier. I first detected the issue when I enabled the bootloader, but the issue is persisting when I enable or disable the bootloader now. 
As you have suggested, I am requesting an investigation in the neorv32 community.

TommyMurphyTM1234, 2024-07-23:
> Thank you for your reply. I have added the openocd log with -d3 in the attachments. [openocd_log.txt](https://github.com/user-attachments/files/16347432/openocd_log.txt)

You seem to have added the log for the failing case but not for the working case?

It's also not clear what the sequence of events/usage is that leads to the failure to connect. 

Is it something like: 

1. First connect works? 
2. Second and subsequent connections fail? 
3. Target reset allows connection to work again? 

Or something else?

UmeshChandargi, 2024-07-23:
I did not save the verbose logs when the openocd connection was working. 
All I have is this 
```
ml013:~> openocd -f /home.filer/chandargi/GIT_repo/neorv32-setups/neorv32/sw/openocd/openocd_neorv32.cfg
Open On-Chip Debugger 0.12.0+dev-00634-g23c33e1d3 (2024-07-12-12:09)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : J-Link ARM V8 compiled Sep 22 2014 23:26:43
Info : Hardware version: 8.00
Info : VTarget = 3.332 V
Info : clock speed 1000 kHz
Info : JTAG tap: neorv32.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=1 progbufsize=2
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40801104
Info : [neorv32.cpu.0] Examination succeed
Info : [neorv32.cpu.0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Target HALTED. Ready for remote connections.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
So the sequence of events are as follows:

1. I successfully established connection with openocd and started gdb. But I could not `load` the elf file as I was using the IMEM as ROM in my neorv32 riscv softcore processor. 
2. So as per the documentation of the target, I enabled the internal bootloader and re-uploaded the bitstream onto the FPGA. The connection failed. 
Here is the first instance where it failed. 
[image]

If you notice the first part, the connection is working fine. But when I uploaded the new bitstream it records an "JTAG signal issue". So i invoked shutdown and tried to reconnect. It fails now to connect in the second part of the screenshot.

3. I reversed the changes and tried to connect, it has been failing ever since.
4. I removed the `jtag_trst_i` connection in my block design and connected a constant of 1 so that the reset never happens. 
[image]
It is failing regardless of all these changes.

vicencb, 2024-07-23:
Hi,
i am facing a similar issue:
```
Error: OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x1).
This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
```
openocd reports the register `dmstatus` is `0x1`.
But if i read that same register with low-level jtag commands:
```
> irscan pspl.tap 0x923 ; drscan pspl.tap 41 0x4400000001
00440100320c
> irscan pspl.tap 0x923 ; drscan pspl.tap 41 0x4400000000
00440100320c
```
`00440100320c` decodes as `ADDR:0x11 DATA:0x00400C83`
where `0x11` is the address of the `dmstatus` register and the contents are not `0x1` but `0x00400C83`.
The value `0x00400C83` matches the expected value set in the `neorv32`.
This proves the data read from the FPGA is correct.

TommyMurphyTM1234, 2024-07-23:
> Hi, i am facing a similar issue:
> 
> ```
> Error: OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x1).
> This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
> ```
> 
> openocd reports the register `dmstatus` is `0x1`. But if i read that same register with low-level jtag commands:
> 
> ```
> > irscan pspl.tap 0x923 ; drscan pspl.tap 41 0x4400000001
> 00440100320c
> > irscan pspl.tap 0x923 ; drscan pspl.tap 41 0x4400000000
> 00440100320c
> ```
> 
> `00440100320c` decodes as `ADDR:0x11 DATA:0x00400C83` where `0x11` is the address of the `dmstatus` register and the contents are not `0x1` but `0x00400C83`. The value `0x00400C83` matches the expected value set in the `neorv32`. This proves the data read from the FPGA is correct.
> 
> Edit: forgot to mention my configuration:
> 
> ```
> adapter driver ftdi
> adapter speed 200
> ftdi vid_pid 0x0403 0x6014
> ftdi channel 0
> ftdi layout_init 0x00E8 0x60EB
> reset_config none
> transport select jtag
> jtag newtap arm  dap -irlen 4  -expected-id 0x5BA00477
> jtag newtap pspl tap -irlen 12 -expected-id 0x24738093
> 
> target create pspl.tap riscv -chain-position pspl.tap
> riscv set_ir idcode 0x903
> riscv set_ir dtmcs  0x922
> riscv set_ir dmi    0x923
> ```

Verbose logs for all tests including the manual read?

vicencb, 2024-07-23:
Here is the log: [oocd.log](https://github.com/user-attachments/files/16352163/oocd.log)
I've changed the commands to:
```
irscan pspl.tap 0x923 ; drscan pspl.tap 2 1 32 0 7 0x11
irscan pspl.tap 0x923 ; drscan pspl.tap 2 0 32 0 7 0x11
```
to avoid >32 register size.
Also, i've executed `reset halt` and `shutdown` at the end:
```
TAP arm.dap does not have valid IDCODE (idcode=0x48e70126)
JTAG tap: pspl.tap tap/device found: 0x24738093 (mfg: 0x049 (Xilinx), part: 0x4738, ver: 0x2)
Debugger is not authenticated to target Debug Module. (dmstatus=0x3). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.

[pspl.tap] Hart unexpectedly reset!
OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x1). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
[pspl.tap] Couldn't determine state. dmstatus=0x1
[pspl.tap] datacount=0 progbufsize=0
[pspl.tap] We won't be able to execute fence instructions on this target. Memory may not always appear consistent. (progbufsize=0, impebreak=1)
OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x2000a01). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
[pspl.tap] Vector support with vlenb=33556993
[pspl.tap] S?aia detected with IMSIC
[pspl.tap] Examined RISC-V core
[pspl.tap]  XLEN=64, misa=0x2008a0102000a01
[pspl.tap] Target successfully examined.
[pspl.tap] Examination succeed
OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x2000a01). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
> shutdown
shutdown command invoked
```

UmeshChandargi, 2024-07-24:
Hello,
I solved it. I checked with the oscilloscope for tck, tdi, tdo on the PMOD. I got no readings so I checked the IO report. None of the JTAG signals appeared in the report. After analysing a little I saw that vivado had duplicated the wrapper file from the block design (I do no know why it suddenly happened now). All the JTAG signals were updated in the duplicated wrapper.vhd. I cleaned it and now the openocd works fine.
It's a bit embarrassing, I apologise for taking your time.
Thank you for all your help.

en-sc, 2024-07-24:
@vicencb, from the logs it seems like a comletely different isuue:
* E.g. you can see `dmstatus` being read twice in a row, however, the first read fails, while the second one succeeds:
```
Debug: 171 119 batch.c:157 riscv_batch_run_from(): [pspl.tap] Running batch of scans [0, 2)
Debug: 172 121 riscv-013.c:398 riscv_log_dmi_scan(): 41b r 00000000 @11 -> + 00000001 @00; 0i
Debug: 173 121 riscv-013.c:411 riscv_log_dmi_scan(): read:
Debug: 174 121 riscv-013.c:398 riscv_log_dmi_scan(): 41b - 00000000 @00 -> + 00000001 @11; 0i
Debug: 175 121 riscv-013.c:411 riscv_log_dmi_scan(): read: dmstatus=1 {version=0_11}
Error: 176 121 riscv-013.c:602 dmstatus_read(): OpenOCD only supports Debug Module version 2 (0.13) and 3 (1.0), not 1 (dmstatus=0x1). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed.
Debug: 177 121 riscv-013.c:1977 examine_dm(): [pspl.tap] Detected 1 harts.
Debug: 178 121 riscv-013.c:5309 riscv013_invalidate_cached_progbuf(): [pspl.tap] Invalidating progbuf cache
Debug: 179 121 batch.c:157 riscv_batch_run_from(): [pspl.tap] Running batch of scans [0, 2)
Debug: 180 123 riscv-013.c:398 riscv_log_dmi_scan(): 41b r 00000000 @11 -> + 00430c83 @00; 0i
Debug: 181 123 riscv-013.c:411 riscv_log_dmi_scan(): read:
Debug: 182 123 riscv-013.c:398 riscv_log_dmi_scan(): 41b - 00000000 @00 -> + 00430c83 @11; 0i
Debug: 183 123 riscv-013.c:411 riscv_log_dmi_scan(): read: dmstatus=0x430c83 {version=1_0 anyrunning=1 allrunning=1 anyresumeack=1 allresumeack=1 impebreak=1 authenticated=true}
Debug: 184 123 riscv-013.c:2040 examine(): [pspl.tap] dmstatus:  0x00430c83
```

I'd like to dig further, but `-d3` logs are unfortunately not enough.

Could you please:
* Open a new issue. You configuration seems to be completely different, and I'd like to close this one, since the original problem was solved.
* Attach `-d4` logs

vicencb, 2024-07-24:
At first it looked like the same issue, but it could not be.
So, from my side, feel free to close it.

I am also currently investigating the problem from the RTL side.
If i find it is a SW issue, i'll attach `-d4` logs in a new issue.

Thanks for helping!