riscv-collab/riscv-openocd #1110: How to program multiple cores using Openocd ?

rtspk, 2024-08-07:
Hi Guys, 

I am in the process of implementing a multicore processing system based on Pulpissimo on FPGA. In this regard, I'm facing a few challenges. After weeks of trying many methods, I thought of writing it here to get some assistance so that we can narrow down the problems. 

Context:
Single-core Pulpissimo works on FPGA. We can program the single-core Pulpissimo using openocd and GDB. The configuration file for the openocd is attached for reference. 

•	We have extended Single-core Pulpissimo to four riscv cores. 
•	The four core IDs are 0x3e0, 0x3e1, 0x3e2,  0x3e3.
•	All four cores share the same L2 memory. 
•	Each core has been allotted a separate memory segment (kindly have a look at the GDB output after loading the program).
•	Core 0 prints messages on the UART terminal 
•	Core 1, 2, and 3 prints incrementing counter on the LCD (we have connected an LCD on the FPGA board).  

In theory, all cores should work in parallel and I should see the core0 printing the messages in the terminal and core 1, 2, and 3 continuously increasing the counter value on the LCD screen. 

The problem is, at one time, only one core runs on FPGA. I am not able to run all the cores in parallel. i.e., if core 0 prints messages on the UART terminal, then core 1, 2, and 3 won't increase the counter value on the LCD. 

I am also not sure how to modify the opened configuration file to support four cores. As of now, create separate configuration files for each core. The only difference in this configuration file is the value of -coreid.

Is there a specific way in which I should program the multicore pulpissimo design?  Could you please help me narrow down this problem for me ? 

Thanks a lot and best regards
Rizwan

[quad_core_gdb_github.txt](https://github.com/user-attachments/files/16523024/quad_core_gdb_github.txt)
[olimex_sep_core0.txt](https://github.com/user-attachments/files/16523072/olimex_sep_core0.txt)
[olimex_sep_core1.txt](https://github.com/user-attachments/files/16523075/olimex_sep_core1.txt)
[olimex_sep_core2.txt](https://github.com/user-attachments/files/16523076/olimex_sep_core2.txt)
[olimex_sep_core3.txt](https://github.com/user-attachments/files/16523077/olimex_sep_core3.txt)

en-sc, 2024-08-07:
> The four core IDs are 0x3e0, 0x3e1, 0x3e2, 0x3e3.

What do you mean by `core ID`? Please, check out [this comment](https://github.com/riscv-collab/riscv-openocd/issues/278#issuecomment-2151940404) for details.

From the logs you have provided, I do not understand how you resume the cores. Each core is halted (by `halt` in OpenOCD config), but never resumed.

en-sc, 2024-08-07:
Please, take a look at the documentation on config files:
https://openocd.org/doc/html/Config-File-Guidelines.html

To write a correct config file one would also need a more detailed understanding on how exactly you are joining debug interfaces of the cores:
* Is there a separate JTAG TAP for each core?
* Is there a single JTAG TAP, but each core has it's own RISC-V Debug Module?
* Is there a single RISC-V Debug Module all the cores are connected to,  behind a single JTAG TAP?

You can specify all your targets in one file. This will greatly simplify the process. Also consider making joining the cores into one SMP group, so that they halt and resume simultaneously: https://openocd.org/doc/html/GDB-and-OpenOCD.html#usingopenocdsmpwithgdb.

rtspk, 2024-08-12:
Hi @en-sc 

Thanks a lot for the detailed answer. 

For your first comment: 
Yes, this configuration file has a `halt `command in the end. 
I also added the resume and exit at the end of the configuration file. In this case, the core is free running (using the `resume `command),  but when I use the resume command, I face some other problems. 
After I used the resume command for one core, I was not able to connect the other cores using openocd. 

For your second comment, our design is based on your third bullet point. i.e. we have only one JTAG TAP and debug model in the quad-core system.

Thank you very much for the suggestions on core ID and using SMP. I will look into this and will update you on the progress. 

Best regards
Rizwan

en-sc, 2024-09-06:
@rtspk, what is the status of the issue? Were you able to connect?

closed 2024-09-06 by en-sc