riscv-collab/riscv-openocd #1117: Autoconf version not recognized

pcotret, 2024-08-21:
- Ubuntu 22.04.4 LTS
```bash
$ git clone https://github.com/riscv-collab/riscv-openocd
$ cd riscv-openocd
$ git submodule update --init --recursive
$ ./bootstrap
./bootstrap 
+ aclocal --warnings=all
configure.ac:3: error: Autoconf version 2.69 or higher is required
configure.ac:3: the top level
```
However:
```bash
$ autoconf --version
autoconf (GNU Autoconf) 2.71 # The only autoconf version I have on my OS
$ libtoolize --version
libtoolize (GNU libtool) 2.4.6
```

TommyMurphyTM1234, 2024-08-21:
Might be useful to attach the `bootstrap` and `configure` logs?

TommyMurphyTM1234, 2024-08-21:
Are you absolutely sure that only version 2.71 of `autoconf` is installed?
```
apt list --installed | grep autoconf

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

autoconf/jammy,now 2.71-2 all [installed]
```

TommyMurphyTM1234, 2024-08-21:
Also - what is your version of `autom4te`?
```
autom4te --version
autom4te (GNU Autoconf) 2.71
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<https://gnu.org/licenses/gpl.html>, <https://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Akim Demaille.
```

pcotret, 2024-08-21:
> Also - what is your version of `autom4te`?
>
> ```
> autom4te --version
> autom4te (GNU Autoconf) 2.71
> Copyright (C) 2021 Free Software Foundation, Inc.
> License GPLv3+/Autoconf: GNU GPL version 3 or later [...]

I'm closing my issue.
I had two `autom4te`: default was 2.64 while autoconf was 2.71...

Thanks @TommyMurphyTM1234 !

closed 2024-08-21 by pcotret