riscv-collab/riscv-openocd #1119: How can a config support two types of tap configuration

lz-bro, 2024-08-30:
prijtag.cfg
```
set _CHIPNAME riscv

jtag newtap pre unknown -irlen 1 -expected-id 0x00000000

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21

jtag newtap post unknown -irlen 9 -expected-id 0x08501C0D

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME
```
secjtag.cfg
```
set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME
```

jtag.cfg
```
set _CHIPNAME riscv
if { [info exists SECJTAG] } {
    jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21
} else {
    jtag newtap pre unknown -irlen 1 -expected-id 0x00000000
    
    jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x10000E21

    jtag newtap post unknown -irlen 9 -expected-id 0x08501C0D
}
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME
```
I can use `secjtag` by add `-c 'set CWTAP 1'`, and Is there a better way to turn it into automation？

TommyMurphyTM1234, 2024-08-30:
> Is there a better way to turn it into automation？

What exactly do you mean? If you mean is there any way to dynamically and automatically select the correct scripting then I can't see any obvious way given that it's presumably not possible to do any sort of querying of the target this early on to find out what class of JTAG connection it needs?

aap-sc, 2024-08-30:
If I undestand your question correctly....

It depends on you implementation
Here is what you can try to do.

- Once OpenOCD starts it can initiate blind interrogation of the scan chain. If this procedure succeeds  - you can try and guess your configuration from the number of taps detected. You can also try to execute some raw jtag commands to figure things out.

Here is an example:

```
openocd \
    -f "$DebugInterfaceConfig" \
    -c "transport select jtag" \
    -c "adapter speed $ScanSpeed" \
    -c "adapter serial $AdapterSerial" \
    -c "noinit" \
    -c "jtag init" \
    -c "echo \"detected number of taps: \[llength \[jtag names\]\]\"" \
```

Once you have jtag scan chain right - you can try to figure out IDCODES of devices you have.

- If above succeeds you can organize your scripting as follows:
1. run auto-detection script, print output to stdout (or file)
2. run openocd the second time with appropriate parameters corresponding to step1.

aap-sc, 2024-08-30:
If you don't need auto detection facilities, then

> -c 'set SECJTAG 1'

if quite reasonable approach.

closed 2024-09-02 by lz-bro