riscv-collab/riscv-openocd #1138: How to enable sba read and write access instead of using program buffer

DecodeTheEncoded, 2024-09-25:
My teams is deploying our Debug Module with system bus access(currently, we comply with 0.13spec), but it seems that the openocd always tried to use program buffer to access the memory space(i use -d flag to trace the log to confirm this), I wonder if I can forecefully ask openocd to use system bus access feature instead of program buffer, We want to test whether our SBA feature DM is ok. But it seems that there is no easy way for this. 
Any help will be appreciated. Thanks

TommyMurphyTM1234, 2024-09-25:
`riscv set_mem_access sysbus`?

- https://openocd.org/doc/html/Architecture-and-Core-Commands.html#index-riscv-set_005fmem_005faccess
- https://github.com/riscv-collab/riscv-openocd/blob/3bed4c8015b319b97e6651a383db9ae001f11aa9/src/target/riscv/riscv.c#L5115-L5122

DecodeTheEncoded, 2024-09-25:
Hi, Thanks for your help, I guess this will work. I will try this right now.

  *   https://openocd.org/doc/html/Architecture-and-Core-Commands.html#index-riscv-set_005fmem_005faccess
  *   https://github.com/riscv-collab/riscv-openocd/blob/3bed4c8015b319b97e6651a383db9ae001f11aa9/src/target/riscv/riscv.c#L5115-L5122

DecodeTheEncoded, 2024-09-25:
> `riscv set_mem_access sysbus`?
> 
> - https://openocd.org/doc/html/Architecture-and-Core-Commands.html#index-riscv-set_005fmem_005faccess
> - https://github.com/riscv-collab/riscv-openocd/blob/3bed4c8015b319b97e6651a383db9ae001f11aa9/src/target/riscv/riscv.c#L5115-L5122

This works, Thanks. I will close this. 😊

closed 2024-09-25 by DecodeTheEncoded