riscv-collab/riscv-openocd #1143: Unable to halt Hart 992

RenatoBelmonte, 2024-10-04:
After I added a module to pulp_soc and synthesized PULPissimo for zcu102 I'm struggling with openocd, this is the result of the debug view. I've installed the patched version of openocd through pulp-sdk and I'm using the last version of PULPissimo

```
Error: 11368 1888 riscv-013.c:3090 riscv013_halt_go(): unable to halt hart 992
Error: 11369 1888 riscv-013.c:3091 riscv013_halt_go():   dmcontrol=0x83e00001
Error: 11370 1888 riscv-013.c:3092 riscv013_halt_go():   dmstatus =0x00000c82
Error: 11371 1888 riscv-013.c:1594 examine(): Fatal: Hart 992 failed to halt during examine()
Debug: 11372 1888 openocd.c:161 handle_init_command(): target examination failed
Debug: 11373 1888 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_flash init
Debug: 11374 1888 command.c:143 script_debug(): command - ocd_flash ocd_flash init
Debug: 11376 1888 tcl.c:1222 handle_flash_init_command(): Initializing flash devices...
Debug: 11377 1888 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_mflash init
Debug: 11378 1888 command.c:143 script_debug(): command - ocd_mflash ocd_mflash init
Debug: 11380 1888 mflash.c:1377 handle_mflash_init_command(): Initializing mflash devices...
Debug: 11381 1888 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_nand init
Debug: 11382 1888 command.c:143 script_debug(): command - ocd_nand ocd_nand init
Debug: 11384 1888 tcl.c:497 handle_nand_init_command(): Initializing NAND devices...
Debug: 11385 1888 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_pld init
Debug: 11386 1888 command.c:143 script_debug(): command - ocd_pld ocd_pld init
Debug: 11388 1888 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 11389 1888 gdb_server.c:3425 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 11390 1888 server.c:310 add_service(): Listening on port 3333 for gdb connections
Debug: 11391 1889 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_halt
Debug: 11392 1889 command.c:143 script_debug(): command - halt ocd_halt
Debug: 11394 1889 target.c:3059 handle_halt_command(): -
Error: 11395 1889 target.c:573 target_halt(): Target not examined yet
```

en-sc, 2024-10-04:
Please, attach a verbose OpenOCD log.
It can be obtained by adding `-d3` in the command line or by specifying `debug_level 3` in your config file (or via Telnet).
The log can be redirected to a file via `-l` or `log_output` respectively.

RenatoBelmonte, 2024-10-04:
[openocd_d3.log](https://github.com/user-attachments/files/17259266/openocd_d3.log)

RenatoBelmonte, 2024-10-04:
https://github.com/pulp-platform/riscv-openocd

I think that the only change is the maximum number of harts

en-sc, 2024-10-04:
Maybe you could attempt using RISCV OpenOCD?
04d8cfc48cf669400eb9c344c846c571f1fa5c91 adds support for up to 2048 harts (the same as the patch from the fork).

The version you are using is out of date. There are quite a few commits in RISC-V OpenOCD that could help with your issue (e.g. 418fcf1ceac981973eb898a03a3ea8136bc3db9c makes DM examination compliant with RISC-V Debug Spec v1.0).

RenatoBelmonte, 2024-10-04:
Thank you for your reply. I'll try and let you know