riscv-collab/riscv-openocd #1145: openOCD issue : failed to read mis

abhiram-padikkat, 2024-10-09:
While debugging a SoC with a 64-bit RISC-V processor and a Debug Module (Spec 1.00), we're encountering connectivity issues with OpenOCD. The error indicates a failure to read the MISA register, suggesting the target hasn't been examined yet. Upon reviewing the OpenOCD source code, we suspect that accessing the progbuffer might be the underlying issue. However, we're uncertain about the progbuffer's location - is it within the Debug Master Module or the Debug Slave attached to the AXI bus? Could the integration of these components or any other peripheral in bus be the root cause of the issue, or are there other potential reasons for this failure?

TommyMurphyTM1234, 2024-10-09:
As always, the first step is to collect and upload a verbose `openocd -d3` log which might shed some light on the behaviour.

abhiram-padikkat, 2024-10-09:
> As always, the first step is to collect and upload a verbose `openocd -d3` log which might shed some light on the behaviour.

[output.log](https://github.com/user-attachments/files/17317736/output.log)
this is our log report got by setting debug level as 4 in cfg file. i cant understant the exact issue from this

TommyMurphyTM1234, 2024-10-09:
You're using a very old version of OpenOCD:
>```
> Open On-Chip Debugger 0.10.0+dev-01252-g675259382 (2020-09-22-12:07)
> ```
You should probably try a build of the latest sources.

abhiram-padikkat, 2024-10-09:
> You're using a very old version of OpenOCD:
> 
> > ```
> > Open On-Chip Debugger 0.10.0+dev-01252-g675259382 (2020-09-22-12:07)
> > ```
> 
> You should probably try a build of the latest sources.

but we have successfully debugged our reference soc system with same version. and when we try to make new soc , by mimicing the integration and adding new peripheral, this issue comes

TommyMurphyTM1234, 2024-10-09:
I would still recommend using a build of the latest OpenOCD as there have been so many changes to the RISC-V (and general) OpenOCD support since the version that you're using. In any case it's going to be difficult to find people prepared to investigate issues with such an old version.

abhiram-padikkat, 2024-10-09:
sure. defnitely do that

abhiram-padikkat, 2024-10-09:
btw are you familiar with debug module struction. in the latest spec, there is a debug slave which we attach on system bus,what is its purpose? and where is exactly the program buffer,in dm master or dm slave?

TommyMurphyTM1234, 2024-10-09:
> btw are you familiar with debug module struction. in the latest spec, there is a debug slave which we attach on system bus,what is its purpose? and where is exactly the program buffer,in dm master or dm slave?

Sorry - I'm not really up to date on the latest specs.
You might get better info on the spec and implementation issues on the debug spec repo or mailing list?

- https://github.com/riscv/riscv-debug-spec
- https://lists.riscv.org/g/tech-debug

abhiram-padikkat, 2024-10-09:
okey. thank you

abhiram-padikkat, 2024-10-09:
> okey. thank you so much!!