riscv-collab/riscv-openocd #1151: Support for adding data watchpoints

RaamyPi, 2024-10-16:
According to the latest debug spec (dated May 15, 2024), the select field in mcontrol and mcontrol6 provides an option to configure a watchpoint to break on address/data but openocd doesn't have any reference to this field.

TommyMurphyTM1234, 2024-10-16:
Doesn't `riscv.c` reference/use these?

- https://github.com/riscv-collab/riscv-openocd/blob/riscv/src/target/riscv/riscv.c

TommyMurphyTM1234, 2024-10-16:
> @RaamyPi, AFAIK neither GDB, nor OpenOCD provides means to set a watchpoint on data values (i.e. there is no "select" in OpenOCD's `wp` or GDB's `watch/awatch`.

Hi @en-sc, I think you're correct about GDB:

- https://sourceware.org/gdb/current/onlinedocs/gdb.html/Set-Watchpoints.html

but doesn't OpenOCD support this via the `value` argument? 

-  https://openocd.org/doc/html/General-Commands.html

> Command: `wp [address length [(r|w|a) [value [mask]]]]`
>
> With no parameters, lists all active watchpoints. Else sets a data watchpoint on data from `address` for `length` bytes. The watch point is an "access" watchpoint unless the `r` or `w` parameter is provided, defining it as respectively a read or write watchpoint. If a `value` is provided, that value is used when determining if the watchpoint should trigger. The value may be first be masked using `mask` to mark “don’t care” fields.

en-sc, 2024-10-16:
@TommyMurphyTM1234, thanks! I've missed it.

Then it's a bit easier. RISC-V targets can provide support for OpenOCD watchpoints with `value` specified by chaining a watchpoint with `select = address` and a watchpoint with `select = data`.

TommyMurphyTM1234, 2024-10-16:
FWIW this code seems to explicitly check for and reject watchpoints that specify a data value to match:

- https://github.com/riscv-collab/riscv-openocd/blob/a4020f1a0207dc218812ee262ce861d81aa7962f/src/target/riscv/riscv.c#L1528-L1533

> Then it's a bit easier. RISC-V targets can provide support for OpenOCD watchpoints with value specified by chaining a watchpoint with select = address and a watchpoint with select = data.

It's certainly feasible but I'm not sure about "easy"? At the moment the code seems to be predicated on using one trigger per watchpoint (and hardware breakpoint?) whereas  this change would require additional trigger resource management and availability checking in order to deal with the situation where two triggers are required for a data watchpoint with a value match?

TommyMurphyTM1234, 2024-10-16:
> > At the moment the code seems to be predicated on using one trigger per watchpoint (and hardware breakpoint?)...
> 
> Not exactly. RISC-V OpenOCD already uses chained triggers, e.g. to set an unaligned watchpoint:
> 
> https://github.com/riscv-collab/riscv-openocd/blob/a4020f1a0207dc218812ee262ce861d81aa7962f/src/target/riscv/riscv.c#L1067

Ah - OK - thanks. I wasn't aware of that. 👍

RaamyPi, 2024-10-17:
> If you wish for this feature to be supported (on the level of wp command) consider posting this question to mainline OpenOCD: https://openocd.org/pages/mailing-lists.html

@en-sc thank you, I shall follow this up.

> In the meantime nothing prevents you from setting the watchpoint on data values manually, by reserving the trigger (riscv reserve_trigger command) and programming the trigger with accesses to tselect, tdata1, tdata2 etc.

@en-sc you're right and I have already tried this as well but this is impractical as gdb is not aware of these artifical watchpoints.

RaamyPi, 2024-10-17:
> Even if/when OpenOCD is extended to support value option on data watchpoints this may still be the case given that GDB doesn't have such an option. Unless it will have **some** awareness of watchpoints set using something like `monitor wp 0x80004000 w 0xdeadbeef`? But I don't think that it will? If you want full control and awareness of the value option in GDB then GDB itself would probably also need to be extended?

GDB definitely has to be extended as well,  I wanted to take up the openocd side of things before that.

TommyMurphyTM1234, 2025-07-08:
> Hello [@TommyMurphyTM1234](https://github.com/TommyMurphyTM1234) [@RaamyPi](https://github.com/RaamyPi) , gdb can watch on address with expression checked, like below, I just did a example as below in x86 like this `watch *0x555555558014 if *0x555555558014=0x1234`

Yes - but the mechanism used to implement that is most likely repeated reading of the target memory to check if the watch conditions have been met and while this may be pragmatic on a native host platform it is most likely not so in a remote embedded target situation where communitation is over a limited bandwidth/speed connection such as JTAG.

TommyMurphyTM1234, 2025-07-08:
> Yes, [@TommyMurphyTM1234](https://github.com/TommyMurphyTM1234) the gdb `Z-packet` as described in https://sourceware.org/gdb/current/onlinedocs/gdb.html/Packets.html#Packets seems not to support watchpoint with condition, do you know whether there is a feature request in gdb to support watchpoint with condition?

No - you'd need to check the GDB mailing lists/bugbase/community.