riscv-collab/riscv-openocd #1153: "riscv set_dm_baseaddr" command is not shown in User's Guide

zhangdujiao, 2024-10-22:
I didn't see the "riscv set_dm_baseaddr" command, witch is used for direct the base address of DM in the configuration files of OpenOCD.
Is that command really exist? or works?
Is that means the base address of the DM is totally defined by designer rather than fixed at the bottom of the address space, from "The bottom of the address space is used for the first (and usually only) DM.", from <RISC-V Debug Specification>.

Many Thanks!

TommyMurphyTM1234, 2024-10-22:
No such command exists as far as I can see:
```
git clone https://github.com/riscv-collab/riscv-openocd
cd riscv-openocd/src/target/riscv
grep set_dm_baseaddr *
# No results
```
Where are you getting this command from?

TommyMurphyTM1234, 2024-10-22:
Any use?

- https://github.com/riscv-collab/riscv-openocd/pull/875

zhangdujiao, 2024-10-22:
> Where are you getting this command from?

from GPT... while I didn't find this command from any other link...

My most concern here is whether the base address of the DMs can be defined by our-self rather than in the bottom of the address space.

en-sc, 2024-10-22:
> My most concern here is whether the base address of the DMs can be defined by our-self rather than in the bottom of the address space.

AFAIR, you should use `-dbgbase <DM base address>` target configuration parameter (in `target create` or `<targetname> configure`) in the configuration script (before the call to `init`).

TommyMurphyTM1234, 2024-10-22:
> > Where are you getting this command from?
> 
> from GPT... 

That's strange because I can find no mention of `set_dm_baseaddr` in any open or closed Issue or PR in this repo so I don't know where GPT got it from. (And, having used GPT a bit lately, I have found that it very regularly gives incorrect or misleading information and conflicting information when challenged!).

zhangdujiao, 2024-10-22:
> I have found that it very regularly gives incorrect or misleading information and conflicting information when challenged!

It seems that they becomes smart enough to trick humans :)

zhangdujiao, 2024-10-22:
> `- <DM base address>` target configuration parameter

I didn't see this command in  16.11.3 RISC-V Debug Configuration Commands, User's Guide,
Can -dbgbase be used for both ARM and RISC-V arch. ?
If the DM address can be directed in configuration file, I think we can define the address of each DMs by ourself.

TommyMurphyTM1234, 2024-10-22:
> I didn't see this command in 16.11.3 RISC-V Debug Configuration Commands, User's Guide, Can -dbgbase be used for both ARM and RISC-V arch. ? If the DM address can be directed in configuration file, [...]

The only place that it's mentioned in the OpenOCD documentation is in passing here - it seems to be overlooked for proper separate documentation as far as I can see. This is an upstream issue.

- https://openocd.org/doc-release/html/Config-File-Guidelines.html

en-sc, 2024-10-22:
> Can -dbgbase be used for both ARM and RISC-V arch. ?

It can. Though there is a different meaning for RISC-V. For RISC-V `-dbgbase` specifies the DM base address. Please note, this is not stable interface and will probably change in the near future -- I'd kike to introduce RISC-V specific parameter instead.

zhangdujiao, 2024-10-23:
> For RISC-V `-dbgbase` specifies the DM base address.

Thanks, really appreciated!
The Users can configure the RISCV DMs base address by `-dbgbase`. So, the SoC designer can arrange the DMs address arbitrarily.

closed 2024-10-23 by zhangdujiao