riscv-collab/riscv-openocd #1159: Issues for write memory progbuf handle busy

lz-bro, 2024-11-05:
When encountering abstract command busy response while writing block, `write_memory_progbuf_handle_busy` will be called.
https://github.com/riscv-collab/riscv-openocd/blob/f9a129271642febac27f93673fd96481aff503ca/src/target/riscv/riscv-013.c#L4728-L4747
It seems that only the final address of the failed write (s0) is being processed here，actually, the entire memory block failed to write.
https://github.com/riscv-collab/riscv-openocd/blob/f9a129271642febac27f93673fd96481aff503ca/src/target/riscv/riscv-013.c#L4650-L4671
Is this a bug, or did I misunderstand.

lz-bro, 2024-11-05:
> > It seems that only the final address of the failed write (s0) is being processed here，actually, the entire memory block failed to write.
> 
> While writing memory via Program Buffer something similar to the following program is used:
> 
> ```
> sw s1 0(s0)
> addi s0 s0 4
> ebreak
> ```
> 
> https://github.com/riscv-collab/riscv-openocd/blob/f9a129271642febac27f93673fd96481aff503ca/src/target/riscv/riscv-013.c#L4799-L4807
> 
> Therefore write at `s0 - <write size>` was successful (otherwise the increment wouldn't have been executed).

```
Debug: 12223 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e816eb81 @05 -> ?; 5i
Debug: 12224 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12225 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 01130001 @04 -> ?; 5i
Debug: 12226 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12227 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 1273f416 @05 -> ?; 5i
Debug: 12228 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12229 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 04023283 @04 -> ?; 5i
Debug: 12230 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12231 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e2163410 @05 -> ?; 5i
Debug: 12232 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12233 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 22f33402 @04 -> ?; 5i
Debug: 12234 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12235 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e002e616 @05 -> ?; 5i
Debug: 12236 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12237 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 300022f3 @04 -> ?; 5i
Debug: 12238 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12239 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f81af012 @05 -> ?; 5i
Debug: 12240 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12241 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w ec0ee406 @04 -> ?; 5i
Debug: 12242 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12243 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e8aae4a6 @05 -> ?; 5i
Debug: 12244 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12245 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e0a2fc1e @04 -> ?; 5i
Debug: 12246 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12247 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f8baf4b6 @05 -> ?; 5i
Debug: 12248 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12249 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f0b2ecae @04 -> ?; 5i
Debug: 12250 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12251 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e94ae546 @05 -> ?; 5i
Debug: 12252 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12253 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e142fcbe @04 -> ?; 5i
Debug: 12254 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12255 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f95af556 @05 -> ?; 5i
Debug: 12256 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12257 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f152ed4e @04 -> ?; 5i
Debug: 12258 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12259 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e9eae5e6 @05 -> ?; 5i
Debug: 12260 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12261 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w e1e2fd5e @04 -> ?; 5i
Debug: 12262 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12263 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f9faf5f6 @05 -> ?; 5i
Debug: 12264 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12265 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w f1f2edee @04 -> ?; 5i
Debug: 12266 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12267 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 25f30001 @05 -> ?; 5i
Debug: 12268 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12269 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 0533fdfe @04 -> ?; 5i
Debug: 12270 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12271 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 80e70000 @05 -> ?; 5i
Debug: 12272 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12273 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 70973400 @04 -> ?; 5i
Debug: 12274 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12275 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 720261e2 @05 -> ?; 5i
Debug: 12276 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12277 22001 riscv-013.c:431 riscv_decode_dmi_scan(): 41b w 60a2bae0 @04 -> ?; 5i
[... 101 lines cut ...]
Debug: 12559 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: command=0x331008 {control=0x331008}
Debug: 12569 22001 riscv-013.c:415 riscv_decode_dmi_scan(): 41b r 00000000 @16 -> + 00331008 @17; 3i
Debug: 12579 22001 riscv-013.c:415 riscv_decode_dmi_scan(): 41b - 00000000 @16 -> + 08000002 @16; 3i
Debug: 12580 22001 riscv-013.c:428 riscv_decode_dmi_scan(): read: abstractcs=0x8000002 {datacount=2 progbufsize=8}
Debug: 12581 22001 riscv-013.c:1754 register_write_direct(): [riscv.cpu.0] fp <- 0x40000500
Debug: 12591 22001 riscv-013.c:415 riscv_decode_dmi_scan(): 41b w 00000000 @05 -> + 00000000 @16; 3i
[... cut ...]
Debug: 12668 22001 riscv-013.c:438 riscv_decode_dmi_scan(): write: 
Debug: 12678 22001 riscv-013.c:415 riscv_decode_dmi_scan(): 41b - 00000000 @18 -> + 00000001 @18; 3i
Debug: 12679 22017 server.c:608 sig_handler(): Terminating on Signal 22
```
From my log，write at s0 seems unsuccessful, ultimately leading to overflow，write memory in batch cannot guarantee that the address before s0 was successfully written.
https://github.com/riscv-collab/riscv-openocd/blob/f9a129271642febac27f93673fd96481aff503ca/src/target/riscv/riscv-013.c#L4825-L4838

en-sc, 2024-11-05:
@lz-bro, is the reproduction stable?
If so, please, check whether the memory image you were writing was loaded completely.
AFAIK, the easiest way to do it is:
1. Run `load_image <args...>` -- leads to OpenOCD crashing
2. Second run (without `reset` in-between) `verify_image <same args...>`

My assumption is: the last write in the batch results in `abstractcs.cmderr == busy`. Therefore `write_memory_progbuf_startup()` is called with `address_p` equal to the address of an element one past the last one (`0x40000500`). And there is an extra erroneous write:
```
Debug: 12657 22001 riscv-013.c:3198 log_memory_access64(): M[0x40000500] writes 0x0000000000008082
```

lz-bro, 2024-11-06:
> @lz-bro, is the reproduction stable? If so, please, check whether the memory image you were writing was loaded completely. AFAIK, the easiest way to do so is:
> 
> 1. Run `load_image <args...>` -- leads to OpenOCD crashing
> 2. Second run (without `reset` in-between) `verify_image <same args...>`
> 
> My assumption is: the last write in the batch results in `abstractcs.cmderr == busy`. Therefore `write_memory_progbuf_startup()` is called with `address_p` equal to the address of an element one past the last one (`0x40000500`). And there is an extra erroneous write:
> 
> ```
> Debug: 12657 22001 riscv-013.c:3198 log_memory_access64(): M[0x40000500] writes 0x0000000000008082
> ```

It not look like that the last write in the batch results in abstractcs.cmderr == busy. 
[image]

```
Debug: 11578 21970 riscv-013.c:3198 log_memory_access64(): M[0x40000400] writes 0x0032c2930032f293
Debug: 11579 21970 riscv-013.c:3198 log_memory_access64(): M[0x40000408] writes 0x000102b300028763

Debug: 11868 21986 riscv-013.c:4829 write_memory_progbuf_run_batch(): [riscv.cpu.0] Encountered abstract command busy response while writing block M[0x40000210.. 0x40000410)

Debug: 12009 21986 riscv-013.c:1783 register_read_direct(): [riscv.cpu.0] fp = 0x40000410
Debug: 12010 21986 riscv-013.c:4759 write_memory_progbuf_handle_busy(): [riscv.cpu.0] Restarting from 0x40000210

Debug: 12163 21986 riscv-013.c:3198 log_memory_access64(): M[0x40000410] writes 0x02b3a029eb820113

------------------------------------------------------------------------------------------------------------------------

Debug: 12212 21986 riscv-013.c:3198 log_memory_access64(): M[0x400004f0] writes 0x00018082b1450513
Debug: 12213 21986 riscv-013.c:3198 log_memory_access64(): M[0x400004f8] writes 0xb085051301801517

Debug: 12362 22001 riscv-013.c:4829 write_memory_progbuf_run_batch(): [riscv.cpu.0] Encountered abstract command busy response while writing block M[0x40000418.. 0x40000500)

Debug: 12503 22001 riscv-013.c:1783 register_read_direct(): [riscv.cpu.0] fp = 0x40000500
Debug: 12504 22001 riscv-013.c:4759 write_memory_progbuf_handle_busy(): [riscv.cpu.0] Restarting from 0x40000418

Debug: 12657 22001 riscv-013.c:3198 log_memory_access64(): M[0x40000500] writes 0x0000000000008082

```

en-sc, 2024-11-06:
@lz-bro:
1. What is the version of OpenOCD you are using to reproduce this issue?
2. Please, post a complete `-d3` log of both memory write and read. The memory image would also be very helpful (to understand what are the expected values).

lz-bro, 2024-11-06:
> @lz-bro:
> 
> 1. What is the version of OpenOCD you are using to reproduce this issue?
> 2. Please, post a complete `-d3` log of both memory write and read. The memory image would also be very helpful (to understand what are the expected values).
commit was 687f00c06055026331fa34250dcb05c9f652c361, and OpenOCD will be no crush when using the latest version.
It seems that this  #1087  has eliminated this issue.
[ocd.log](https://github.com/user-attachments/files/17647057/ocd.log)

lz-bro, 2024-11-08:
> > @lz-bro:
> > 
> > 1. What is the version of OpenOCD you are using to reproduce this issue?
> > 2. Please, post a complete `-d3` log of both memory write and read. The memory image would also be very helpful (to understand what are the expected values).
> 
> commit was [687f00c](https://github.com/riscv-collab/riscv-openocd/commit/687f00c06055026331fa34250dcb05c9f652c361), and OpenOCD will be no crush when using the latest version. It seems that this #1087 has eliminated this issue. [ocd.log](https://github.com/user-attachments/files/17647057/ocd.log) [ocd-read.log](https://github.com/user-attachments/files/17656823/ocd-read.log)

@en-sc The latest version of OpenOCD no longer crashes when loading the memory image, but instead reports a "MIS-MATCHED" in the memory block that it encounters an abstract command busy response. 
Openocd report encountered abstract command busy response while writing block M[0x40000418.. 0x40000500)，but  write_memory_progbuf_startup() is called with address_p equal to the address of an element one past the last one (0x40000500). 
Should we restarting from 0x40000418 to handle abstract command busy ？

https://github.com/riscv-collab/riscv-openocd/blob/f9a129271642febac27f93673fd96481aff503ca/src/target/riscv/riscv-013.c#L4650-L4671

en-sc, 2024-11-08:
> The content of s0 seems different between abstract access busy and DMI busy：
1.the content of s0 for abstract access busy -- the address of an element one past the last one.
2.the content of s0 for DMI busy -- the address of the failed write.

I don't think you are right.

The hart is executing the following in the Program Buffer:
1. store using the address from `s0`
2. increment `s0`
3. ebreak
DMI busy can happen at any moment and does not abort the execution of the Program Buffer. Therefore after the last DMI operation finishes, `s0` is the same as it would have been without the DMI busy error occurring. Now remember that the DMI write that caused DMI busy was the request to write the next element. Therefore the address in `s0` is the address of the failed write (that have not started due to DMI busy).

`abstractcs.cmderr == busy` (shortened to AC busy further) is different. It will abort the execution of Program Buffer (the request to start the next write arrived too early).
There are 3 cases:
1.1 AC busy happens before the store retires -- the store is not finished, `s0` contains the address of the write that failed.
1.2 AC busy happens after the store retires but before the increment retires -- the store is fine, `s0` contains the address of the last successful write.
1.3. AC busy happens after the increment retires --  the store is fine, `s0` contains the address of the next write.

Therefore in cases 1.1 and 1.3 the address in `s0` is the address of the failed write (these are the elements that were not yet written).
The case 1.2 is different. Due to the fact that the store has succeeded but the increment fails `s0` contains the address of the last successful write. Therefore, restarting from this address writes the memory there the second time.
However, in none of the cases the address in `s0` is incremented but the store is not successful.

This makes me believe this is a HW issue.

lz-bro, 2024-11-08:
> > The content of s0 seems different between abstract access busy and DMI busy：抽象访问忙和DMI忙s0的内容似乎不同：
> > 1.the content of s0 for abstract access busy -- the address of an element one past the last one.1.用于抽象访问的s0的内容busy——最后一个元素之后的一个元素的地址。
> > 2.the content of s0 for DMI busy -- the address of the failed write.2.DMI busy s0 的内容——写入失败的地址。
> 
> I don't think you are right.我不认为你是对的。
> 
> The hart is executing the following in the Program Buffer:hart 在程序缓冲区中执行以下操作：
> 
> 1. store using the address from `s0`使用`s0`中的地址进行存储
> 2. increment `s0`增量`s0`
> 3. ebreak 希伯来人
>    DMI busy can happen at any moment and does not abort the execution of the Program Buffer. Therefore after the last DMI operation finishes, `s0` is the same as it would have been without the DMI busy error occurring. Now remember that the DMI write that caused DMI busy was the request to write the next element. Therefore the address in `s0` is the address of the failed write (that have not started due to DMI busy).DMI 忙可能随时发生，并且不会中止程序缓冲区的执行。因此，在最后一个 DMI 操作完成后， `s0`与没有发生 DMI 繁忙错误时的情况相同。现在请记住，导致 DMI 繁忙的 DMI 写入是写入下一个元素的请求。因此， `s0`中的地址是写入失败的地址（由于 DMI 忙而尚未开始）。
> 
> `abstractcs.cmderr == busy` (shortened to AC busy further) is different. It will abort the execution of Program Buffer (the request to start the next write arrived too early).`abstractcs.cmderr == busy` （进一步缩短为AC busy）是不同的。它将中止程序缓冲区的执行（开始下一次写入的请求到达得太早）。 There are 3 cases: 有3种情况： 1.1 AC busy happens before the store retires -- the store is not finished, `s0` contains the address of the write that failed.1.1 AC 忙发生在存储退出之前——存储尚未完成， `s0`包含失败写入的地址。 1.2 AC busy happens after the store retires but before the increment retires -- the store is fine, `s0` contains the address of the last successful write.1.2 AC 忙发生在存储退出之后但增量退出之前 - 存储很好， `s0`包含最后一次成功写入的地址。 1.3. AC busy happens after the increment retires -- the store is fine, `s0` contains the address of the next write.1.3. AC 繁忙发生在增量退出之后——存储正常， `s0`包含下一次写入的地址。
> 
> Therefore in cases 1.1 and 1.3 the address in `s0` is the address of the failed write (these are the elements that were not yet written).因此，在情况 1.1 和 1.3 中， `s0`中的地址是写入失败的地址（这些是尚未写入的元素）。 The case 1.2 is different. Due to the fact that the store has succeeded but the increment fails `s0` contains the address of the last successful write. Therefore, restarting from this address writes the memory there the second time.情况1.2有所不同。由于存储已成功但增量失败， `s0`包含最后一次成功写入的地址。因此，从该地址重新启动会第二次向该地址写入内存。 However, in none of the cases the address in `s0` is incremented when the store is not successful.然而，在任何情况下，当存储不成功时， `s0`中的地址都不会递增。
> 
> This makes me believe this is a HW issue.这让我相信这是一个硬件问题。

@en-sc Thank you, I will discuss it with my hardware colleagues

lz-bro, 2024-11-11:
@en-sc It seems to indicate that `abstractcs.cmderr == busy` will not abort the execution of Program Buffer from debug spec-v1.0.0. 
1. any of these operations fail, cmderr is set and none of the remaining steps are executed. (Fail is not equivalent to busy)
[image]
2. A new command still gets to run to completion while busy is set.
[image]

en-sc, 2024-11-11:
@lz-bro, you are right. `abstractcs.cmderr == busy` does not abort the execution.

However, the previous comment:
> 1.the content of s0 for abstract access busy -- the address of an element one past the last one.

Is still incorrect. The next run of program buffer should not start if `abstractcs.cmderr` is not clear [3.14.6. Abstract Control and Status (abstractcs, at 0x16), cmderr field description]:
> No abstract command is started until the value is reset to 0.

Therefore `s0` contains the address from which we need to restart the writes (since the status of the write is unknown) in case of `cmderr == busy`.

lz-bro, 2024-11-12:
Thanks. I will close this.

closed 2024-11-12 by lz-bro